grbl/grbl #1087: 2 axis move at the same time while i command only for one

mohammadmnsh, 2016-09-21:
i use arduino uno and cnc shield ,grbl controller 3.6.1 and not change any setting of grbl

so when i send G01 X0.1 for example the both y and x move

the same for G01 Y0.1

just for x-y and not for z
please help what the solution ???

jahnj0584, 2016-09-21:
You probably have the jumper set wrong to slave the axis

mohammadmnsh, 2016-09-21:
@jahnj0584  i didn't set any of jumber

jaketeater, 2016-09-21:
Can you post a picture of your setup?  It sounds like a hardware issue.

allyerkan, 2020-12-18:
i have the same problem with y-z. if you find a solution,please can you explain how to do it?

mkinney, 2021-03-14:
You probably need to set 2 jumpers. If you want the A to mirror the Y, then put a jumper over on the Y blues and yellows.

Take a look at the "Electronics" part of this instrucable: https://www.instructables.com/DIY-3D-Printed-Dremel-CNC/ There is a picture showing this exact thing.