grbl/grbl #1088: Issues with TB6600 drivers

jahnj0584, 2016-09-21:
I won't double post so here is my google+ conversation of it 

https://plus.google.com/_/notifications/emlink?emr=08000575080696691449&emid=CPChx62bn88CFQHqjAod9dwDxQ&path=%2F107374903637228753974%2Fposts%2FCgkCWuaTX5S&dt=1474418076171&ub=STREAM_COMMENT_NEW

109JB, 2016-09-21:
First of all, You should not daisy chain the wiring from one driver to the next.  For example, the power supply wires should run from each driver direct to the power supply.  Same for the logic and ground wires from the arduino to the drivers. They should go straight from each driver to the arduino, not jumping from one to the next.

The ground on the arduino should connect to the EN-, PUL-, and the DIR- pins on the driver.  Separate the grounds between the 2 ground pins. Then the appropriate pins connected to each of the EN+, PUL+ and DIR+ pins.

I use large drivers like that on several machines and that is how I have them wired.

jahnj0584, 2016-09-22:
I had it sorted before you posted, but thank you for that. That was what I ended up wiring it as. Pictures are on the Google thread

closed 2016-09-22 by jahnj0584