grbl/grbl #1090: Homing did not set X_AXIS  position to near 0

Appulingam, 2016-09-24:
Homing X Axis, moved in the right (left to right) direction, hit the limit switch and backed a little  and stopped. After that X_Axis goes from 0 to 900 and not -900 to o.  When G00 cmd is issued, get Alarm. 

Also I see in defaults.h there is a check on shield types but where is shield type defined

chamnit, 2016-09-24:
@Appulingam : Please read this [FAQ](https://github.com/grbl/grbl/wiki/Frequently-Asked-Questions#why-is-grbl-in-all-negative-coordinates-after-homing-or-it-so-annoying-and-not-what-im-used-to).

closed 2016-09-24 by chamnit

109JB, 2016-09-24:
> Homing X Axis, moved in the right (left to right) direction,

Left to right of the table, spindle, something else?  This is very vague. We don't know what kind of machine. Is it a gantry style router, a milling machine, something else? We also don't know if "moving to the right" is in the positive direction or negative direction? 

> hit the limit switch and backed a little and stopped. 

Sounds right for hoing

> After that X_Axis goes from 0 to 900 and not -900 to o. 

Verify that the above refers to machine coordinates

> When G00 cmd is issued, get Alarm.

G00 what?  G0 command needs to be followed by an axis command, eithe X, Y, or Z.  Also when you invoke the G0 command are you in G90 or G91?

> Also I see in defaults.h there is a check on shield types but where is shield type defined

The defaults.h file is for machine default settings, like steps/mm, mm/min, mm/sec/sec, etc. It is not for shield types