grbl/grbl #1091: chuckm

64chuckm, 2016-09-26:
has anyone tried generating g code from a manual cnc lathe  part to file for future part use xyz?

109JB, 2016-09-26:
I have absolutely no idea what you are trying to say, or how it relates to GRBL.

64chuckm, 2016-09-30:
what it was called teach mode for making a part on cnc like recording all movements to convert to gcode without cad/cam it into a machine or saves it for future use it has been done in past sorry you dont understand maybe it dont relate to this site.

X3msnake, 2016-09-30:
GRBL does not have a Teach Mode

It only acts as the brain for the Machine, If you want to have a teach mode
it must be implemented on the software that sends the moves to GRBL

closed 2016-10-01 by chamnit