grbl/grbl #1092: $1 - Step idle delay, msec   how to enforce

PhilippeRubbers, 2016-09-28:
I use $1=255 most of the time, but find the motors are not powered when I switch on.
My stepper motor driver is slow, and it therefore misses the 1st 0.5seconds after every switch on.

Is there a G-code to enable the drivers?
Conversely, if reset is pressed, I would like to disable motors (for manual override).
If I set $1=100 after a reset is pressed, it does nothing, until the next move..... i.e. the motors are still energized. 
Is there a G-code to disable the drivers?
I imagined that a work around would be to make a nul move..... but grbl appears to be too clever, and ignores such a command. Any suggestions?

chamnit, 2016-09-28:
@PhilippeRubbers : No there is no override command in v0.9. The imminent Grbl v1.x release will include this.

closed 2016-09-28 by chamnit

109JB, 2016-09-29:
You could simply wire a good old fashion switch to the enable pins of the drivers.