grbl/grbl #1094: should GRBL wait for a G0 to finish before executing subsequent G0's?

PeterGrace, 2016-09-29:
Hi,

I am using GRBL with my Mostly-Printed CNC.  My x and y axis are rather fast, but my Z axis is much slower than the X and Y.

In my gcode for my program, I have the following:

```
G00 Z5.0000

(No. 2: Cutout 3)
G00 X66.1518 Y2.0071 Z5.0000
```

The problem I'm having, at least it appears to be such, is the X/Y move in the second G00 is beginning before my G00 Z5 has a chance to execute.  As a result, my bit is snapping as it catches on the side of the workpiece.  The bit doesn't have enough time to clear the workpiece before the x/y travel begins.

Should I be changing my G0 Z5 to a G1 Z5 to cause grbl to pause execution until the first move is complete?  Is there a way to make grbl wait for a G00 move to complete before moving to the next line?

I'm using grbl 0.9j.  Any suggestions would be appreciated.

chamnit, 2016-09-29:
@PeterGrace : Grbl shouldn't be pausing or skipping a line, unless the line returns a parsing 'error:' rather than an 'ok'. If your problem is on the retract, it's possible that you are losing steps when the spindle is raised. Try reducing your z acceleration by half to start with.

PeterGrace, 2016-09-29:
OK, so, in either a G00 or a G01 move, the move should complete synchronously before the next line of code is executed?  I was wondering if maybe the acceleration planner saw two G00's and said "eh, travel moves, lets do them together" or something.

chamnit, 2016-09-29:
> OK, so, in either a G00 or a G01 move, the move should complete synchronously before the next line of code is executed? 

Yes. This is how CNC control should work.

> I was wondering if maybe the acceleration planner saw two G00's and said "eh, travel moves, lets do them together" or something.

No. Grbl should never skip any valid commands that it receives. If the position Grbl is reporting is correct, but your machine doesn't move this way, it's a problem with the machine. Grbl reports position based on the actual step count it pulses to your stepper drivers.

doppelhub, 2016-09-29:
For testing, change your G0 commands to G1 and then set the feedrate to some slow value... that way you can really see what the machine is doing.   Alternatively, you can limit the machines maximum velocity in the configuration settings.  Either, way, I suspect that making the machine move slower will show that grbl is in fact moving step by step.

doppelhub, 2016-09-29:
Also, make sure you understand the difference between an absolute (G90) and relative (G91) move command.  It's possible you're issuing an absolute command and are already at absolute position Z=+5

splitn2, 2016-09-29:
What stepper driver are you using? I'm very curious about the $1 stepper idle delay, used it on a couple of small mills with steppers and set $1=255 to keep the stepper current enabled, this was the easiest way to stop the Z dropping under gravity pull. Other options include using pulleys and air springs etc to provide a neutral support for the z to stop it drifting, Whilst this may not be your issue loosing your z position its worth thinking about. The down side is it keeps the stepper current running all the time and hence heating effects. the wee DRV8825 drivers get hot quick so we have good heat sinks and fan forced cooling as well as deliberately making sure our stepper current is set to a safe 1A approx instead of maxing out at say 1.8A. @chamnit I'm very interested in what you are planning with GRBL 1.x  and stepper current control - I saw a post somewhere that there was something planned? Keep up the good work! cheers Rich

chamnit, 2016-09-29:
@splitn2 : Grbl on the Arduino Uno will be at the end of its planned development after this v1.1 release. There simply isn't any more flash or memory to do anything useful, but dont discount it though. Grbl v1.1 on the 328p is a complete and simple solution for 3-axis machines, especially with the new real-time (zero lag) overrides fulfilling the last core functionality needed in a CNC controller.

Afterwards, I'll be investing all of my time into continuing the development of the ARM version, where I'll add everything and the kitchen sink. If people want it, I can install things like 4th-axis, backlash comp, SPI stepper driver config, and maybe lathe control into the Grbl-Mega branch as a stop-gap, but it's really doesn't make much sense to waste a lot of time on it, since the ARM version will also have it.

splitn2, 2016-09-29:
@chamnit  yeah I understand the challenge and need to move to a platform with more resources, I can't wait to try it, for now the Arduino powered GRBL is awesome I love it! I know a guy who would be very interested in lathe control who is a very innovative machinist I'll try get him to start using GRBL too!

PeterGrace, 2016-09-30:
I haven't had time to retest since I'm waiting on more bits to arrive, but I will definitely give it a shot when I have an opportunity this weekend.   I'm not near the machine to pull actual P/Ns from my drivers but I think they're knockoff Pololu's that I ordered from ebay some time ago.

Thanks for the feedback!

splitn2, 2016-09-30:
I just finished setting up a wee desktop router and I am once again amazed by GRBL. In very short time we had it doing some nice 3D profiling and teaching the owner to do CNC , he is stoked. Thanks agaibn @chamnit and co who develop this neat package, every child should have access to one of these!

PeterGrace, 2016-10-17:
Hi,

To close the loop -- my motor drivers were set too high, and the controllers were malfunctioning due to the heat.  Once I got my drivers dialed down right to about 2V off the pot, everything started working MUCH better.

closed 2016-10-17 by PeterGrace