grbl/grbl #1095: Change probe architecture from polled to interrupt

doppelhub, 2016-09-29:
The probe architecture is polled in the existing code.  The way the stepper ISR works, the slower your feedrate, the less frequently the probe position is polled (because the probe position is simply part of the stepper ISR).  For example, with F40, the ISR period is ~1ms once the velocity is constant.  I'm completely glossing over the response rate during the acceleration trapezoidal ramp for now, but it's even more of an issue (i.e. the period is always longer while we're accelerating).  In short, the probe poll rate is a function of the feedrate and quite slow.

Why is this an issue?  If the probe hardware is actually a tool touching a metal surface, then the probed signal is actually quite noisy.  Even though the probe signal is guaranteed to drop low as the probing tool touching the work piece, it might still bounce back up to VCC (particularly if the tool is spinning and thus cutting), and thus when we poll there's an indeterminacy condition between when the signal is both low AND polled simultaneously.  If grbl used an interrupt instead, then the first time the pin was pulled below VCC*0.3V (i.e. Vil(max)), then the hardware would generate an ISR which would immediately cease further stepper motion.  The result is a more accurate probe and no race condition.

Our empirical observations show that the probe can over travel 0.04" from the scoped time of first contact. 

Implementation specifics:
The Probe pin should be configured as an interrupt via PCINT13 (since the probe is connected to Arduino Uno pin A5).  PCINT13 is handled by interrupt vector PCI1.  PCI1 can be configured to interrupt via multiple pins, so you'll need to make sure to set the PCMSK1 mask so that only PCINT13 generates an interrupt (i.e. 0b00100000).  You're calling PCMSK1 "CONTROL_PCMSK" in cpu_map.h, then ORing it with "CONTROL_MASK" in system.c.  However, there's no reason I can see to OR "PCMSK1" with "CONTROL_MASK" but maybe I'm missing something there.

So what we do is each time a G38.x command falls into the stream, we set PCMSK1(PCINT13), then we tell the steppers to do their thing, then we wait for the probe to trip.  Finally, after the probe is done we clear PCMSK1(PCINT13) so we're not getting false interrupts.

Now, inside the PCI1 ISR, we should just halt stepper motion (i.e. "bit_true(sys_rt_exec_state, EXEC_MOTION_CANCEL);").  I think grbl can handle everything else later on.

...

Thoughts?  I'm sure there's a reason you aren't doing this already?

chamnit, 2016-09-29:
@doppelhub : Sorry TLDR. But I think I understand what you are asking. Yes, probing is polled. But it's synced with the stepper ISR which also controls motion. As long as it is polled every ISR, then it will record the probe position within the resolution of the steppers. 

The reason you are seeing an over travel is because Grbl has to decelerate the machine appropriately. It can't halt it immediately upon detecting the probe without risking losing steps. Instead, it'll record probe position and immediately invoke a feed hold to come to a stop. The probe position reported is almost always different from the stop point.

Also, because of this deceleration issue, you can't really move a probe very fast anyway. You are limited by the deceleration distance and allowable travel of your probe. 

I tend to err on the side of using the minimum amount of interrupts necessary. Try not to have two or more, along with the main program, trying to interact with each other. If you do, it becomes a huge mess of tracking states, volatile variables, and debugging race conditions.

doppelhub, 2016-09-29:
Howdy.  I'll summarize with pictures:
https://postimg.org/image/u33wf76qt/#
https://postimg.org/image/j4smx0i5h/#

In short, our probe routine uses a moving tool, which touches the metal surface intermittently due to the cutting blades on said tool.  We want grbl to interrupt and stop the probe routine the first time the probe voltage (pin A5) drops below Vil(max) (1.5 volts@ VCC=5V).

chamnit, 2016-09-29:
@doppelhub : Ok so it's a noise issue. The easiest solution is to get rid of the noise with a different probing method, like a switch. 

However, I suppose an interrupt can just set a state variable, such that when it's polled by the stepper ISR, it'll check this state variable instead. I don't have any time to work on anything like this right now (busy with v1.1 release). It's fairly low priority and specific. But, I'm open to looking at a pull request on your proposal.

doppelhub, 2016-09-29:
We can't change to a switch*, but an interrupt is better even with a switch, as it will behave immediately (like it should) while the switch debounces.  I'm working on a clean solution right now that uses a state variable that essentially replaces probe_get_state's internal logic.

*In our application, we probe the part while cutting:  Instead of rapid G0 moves, we use G38.3 to make sure that if we hit anything we'll stop sooner rather than later; our host has a decision tree based on the expected versus actual probe results.  Of course it's an error if we hit anything when we're not expecting it, but at least the machine will stop without user interaction.  Anywho, I'll post the pull request once I'm satisfied it's working well.

chamnit, 2016-09-29:
@doppelhub : Interesting. Looking forward to what you come up with. FWIW, a switch would work with a weak low-pass filter. You'd have a little lag, but it'd be consistent.

doppelhub, 2016-09-29:
OK, got it working with interrupts.  I'm sure you can roll it into your code more elegantly, but I leave that to you.  Posting change request now.

Harvie, 2018-09-21:
Is the deceleration really so important as acceleration? I can imagine that machine would be able to decelerate bit faster than it accelerates. Because if you accelerate you have to overcome machine friction, OTOH when you decelerate, the friction helps you!. That's why i think you can decelerate faster than you accelerate. It would make sense to have GRBL decelerate faster. Or maybe add option to disable deceleration completely when probing...

Harvie, 2018-09-21:
Interrupt approach would have another upside: Sometimes the probe is activated just for tiny second and then slips a bit over the material and gets deactivated again until next step. Interrupt would allow you to catch that bounce. (the idea is exact opposite of debouncing).

I put LED to my probe and found when i go downwards by 0.1mm steps, it sometimes just blinks as probe is lowered one step before the step which makes it to trigger permanently.

chamnit, 2018-09-22:
If you don’t decelerate after a probe, you run the risk of losing position, which will cause a lot more problems than solve.

Harvie, 2018-09-22:
@chamnit i think that inertia is eliminated in leadscrew setup.

chamnit, 2018-09-22:
No inertial forces don’t just disappear. It translates to torque on the leadscrew. There is also rotational inertia of the leadscrew and the motor rotor to deal with too. These are not negligible.

Harvie, 2018-09-23:
That might be in case of ballscrew with higher pitch. Ballscrew is capable of translating linear motion to torque. Leadscrews with low pitch have too much friction to do so.

chamnit, 2018-09-23:
Yes friction does play into this but inertial forces become quite huge when you are trying to make the acceleration infinite.

Harvie, 2018-09-24:
I think there's no way that linear motion of Z will translate to rotation in my eshapeoko machine. Also the Z axis is painfully slow due to leadscrew pitch, but i am glad i didn't go with cheaper threaded rod option :-)

langwadt, 2018-09-24:
you don't just have linear inertia, you also have rotational inertia. think of the motor and leadscrew as a flywheel

truglodite, 2021-02-26:
I have an understanding of the physics involved here and that the probe will pass the contact point. According to this post:
https://github.com/grbl/grbl/issues/1436#issuecomment-423701117

GRBL "records" the contact point, which would make decel a non-issue, if that stored point can be used in zeroing the machine. However, I am having trouble clearly understanding how to use this recorded contact point. Take for example the following z probe script:

```
G91
G38.2 Z-20 F100
G91 G0 Z5
G10 P0 L20 Z5
G90
```
 It seems what would happen is the probe will contact, and move past contact (lets assume 0.5mm past). At this point, MCS is not at the contact point, so what does GRBL do to use this... offset WCS the distance of decel? So then, WCS might be say, -0.5 correct? So then the probing macro would set z=0 without the decel error, right?

...or is there something else happening. When I use this macro, my PCB's are etched consistently about 0.3mm too deep. So it seems like something else. It would be great if there were some documentation to clarify just how the decel is handled with the typical probing script in this regard. The linuxcnc gcode manual has more info than most, but it does not elaborate on how the coordinate registers are used during G38.X decel.