grbl/grbl #1097: Steppers not working, everything else seems to work

Combatir, 2016-09-29:
Dear Grbl experts,

When I use G code sender or Chillipeppr to let my stepper motors move it does not work. But on screen it seems to work so I dont know what I am doing wrong hehe.

For example, I use a Arduino UNO with a CNC shield V3, I have the drivers connected on the shield and the shield connected to the arduino. Connected to my computer with usb and the power supply connected for the steppers. 
I flashed GRBL to the arduino and that seems to work, it says Grbl 0.8c ['$' for help] when connecting. (or 0.9 when testing it with that version).

When I use the machine control it says this:>>> G91 G0  X-1
ok
etc, so that seems to work, however the motors don't move a bit.
When I try to run with chillipeppr they also seem to work on screen but they do not.

Did I do something wrong with the wiring?
Would love to hear your ideas!

Combatire

jahnj0584, 2016-09-29:
Do the motors make noise (a hum) when you turn the power supply on?

Combatir, 2016-09-29:
Dear Jahnj,
They do not, they make no sound or anything.
Btw I use nema 17 stepper motors.

jahnj0584, 2016-09-29:
Wired wrong. Find out which is a coil pair and look at the diagram again
from protneer

Combatir, 2016-09-29:
Thanks Protneer I will try to rewire and post the results.

Combatir, 2016-09-29:
I have found a diagram for how to connect the wires to the stepper motor and those are connected like they should. I also found the pairs, for example the gray and green are a pair and yellow/red are one. I try to look it up but cant find it for sure so maybe you know. Do you just put the pairs together in the pin and then connect to the shield?

Combatir, 2016-09-29:
[image]
[image]

Encarnacao, 2016-09-29:
from what you discribe it seems that you have the wiring wrong or that your adjustment of the vref trim pot of drivers might not be correctly adjust

Combatir, 2016-09-29:
Yes I think so too, tried to change the wires but did not work. By the way I saw that when I open it up on Gcode sender the UNO begins blinking a red light, dont know what that means. I will continue tomorrow. If anyone has anything that might help me much appreciated.

Encarnacao, 2016-09-29:
did you check with a multimeter to see if you got voltage on the VMOT and gnd pins?

Combatir, 2016-09-29:
Encarnacao, I did not check that I do not have a multimeter quite a noob I am lol.

Encarnacao, 2016-09-29:
you see the blue conectors on the cnc shield, they arent powered

Combatir, 2016-09-29:
Yes but I do have the black cable plugged in from a power source, is that okay or do I still need to power the blue connectors?

Encarnacao, 2016-09-29:
all you doing is powering the arduino, you need to power the motors externaly and yes on the blue connectors, thats why isnt working

Encarnacao, 2016-09-29:
also remeber that you using the CNC shield V3 that was made for the grbl 8, and that if you use the grbl 9 you will need to swap the pins 11 and 12 , go on the grbl github and you will see, you can do that by soldering 1 wire in the pin 11 and 1 wire in pin 12 on the back of the arduino and then bend those pins on the shield and swap them if you gonna use the grbl 9, but this isnt the reason why isnt working, just to let you know how to proper conect the grbl9 with the shield you have

Combatir, 2016-09-29:
Thanks for your advice, tomorrow I will continue to solve my problem. I will need to get a power supply then hahaha

Encarnacao, 2016-09-29:
you can always get a 12v power supply from a old comp just to test it out things :)

Combatir, 2016-09-29:
So Just to be sure,
http://www.elecrow.com/12v2a-acdc-power-adapter-with-cable-p-683.html Will only Power the arduino itself but cannot Power the steppermotors Right ?
Thanks for the Help!

Encarnacao, 2016-09-29:
to plug the motors you would need to cut the end of the 12v adapter to connect to the shield or get some adapter, now that 12v adapter is only 2 amps so assuming you will use 3 motors  and assuming  they will spend 1 Amp eatch that would be 3 Amps, if you could find some old computer you could remove the power supply and use the yelow ans black wire to supply you with 12v and alot of amps, just a idea lol

Encarnacao, 2016-09-29:
also one thing from picture i see you have there 3 drivers and only one motor, always remove the drivers that dont have the motor on (load) you shouldnt never plug the shield with drivers with out a load on it, that will damage the drivers

Combatir, 2016-09-30:
Thanks for all the advice, when I come home I will look for such a power supply. 
I think I maybe already damaged all the drivers. If I attach only one driver with one motor applied, connected with usb and the 2 amp plug, it won't work, so I think that means the shield are damaged. 
I will later try with a bigger power supply.

electrokean, 2016-09-30:
Usually the drivers are only damaged if you plug or unplug the motors while they are powered on, so they're probably still OK if you didn't have any power connected to the shield.  But it is probably a good idea to leave off any drivers that aren't in use till you get the basics working.

Combatir, 2016-09-30:
Yes I will do that, but suppose I have the 2 amp power supply and only one motor and driver connected it should work right?

electrokean, 2016-09-30:
Yes, it should.
Your issue isn't really with Grbl, so you might want to read/use the CNC Shield forums at http://forum.protoneer.co.nz/

Combatir, 2016-09-30:
Will do thanks for the help.

matiasgonzalez90, 2017-01-19:
HI I'm having the same problem as you. Did you solve it?? Could you tell me how? 

Thanks!

closed 2017-01-26 by chamnit

madbuster, 2018-04-14:
I have made a  cnc router with grbl arduino and drv8825 ...some days ago it was working well then first problem come that all axis move in same direction for +ve and -ve command ...then second problem was y-axis stopped working at all ...and now whole electronic part stopped working...

Rob2029, 2018-04-15:
did you check everything with a multimeter??

mikypolice, 2018-09-19:
Hey guys, you have an idea for this?
https://www.dropbox.com/s/t4up3aelpswlmlg/20180915_150257.mp4?dl=0
I changed the board with Arduino uno clone and the driver with 3 tb6560 with same result!!!! I've tried 0.9j and 1.1 with my clone but the result is the same!
I used a 12v 20A power supply with my wiring, the tb6560 are setup to 1.2A for nema 17 (1.33A). Is only Y axis...
It's possible to be a feed rate setting? $111?

mikypolice, 2018-09-20:
solved! it was the wiring from the motor Y to driver!

Pappagoran, 2018-10-31:
Can I ask how you wired it? I seams to have the same problem.

mikypolice, 2018-10-31:
The controller have 4pin jst plug for each stepper motor, but each motor have 6pin jst plug. The wire are only 4. So you check wire by wire the position and the color if are plugged corrected into jst plug.
Motor side (6pin jst plug) the correct sequence of wire are:
black---not connected---green---blue---not connected---red.
In my case the problem was in that sequence:
black---green---blue---not connected---not connected---red...
So check the wire of jst!!
Sorry for my bad english..

Pappagoran, 2018-10-31:
Thank you. Was this Only för the Y?

Goran

mikypolice, 2018-10-31:
Yes, because the other axis worked fine.
I discovered the problem only by doing cross-tests between engines and drivers. in the end I isolated the problem on the cable, comparing it with the others, it was badly wired.

Pappagoran, 2018-10-31:
Ok, can I just  change order on the wire?
[image: 20181031_165124-1 copy.jpg]

mikypolice, 2018-10-31:
Yes, but the order must ne like this...
https://www.dropbox.com/s/viwy9utdk8vuqwf/SmartSelect_20181031-193642_Samsung%20Internet.jpg?dl=0
and side controller like this
https://www.dropbox.com/s/a8oclk4hj1ok54s/woodpeckercnc.png?dl=0

Pappagoran, 2018-10-31:
> Yes, but the order must be like this...
>
> https://www.dropbox.com/s/viwy9utdk8vuqwf/SmartSelect_20181031-193642_Samsung%20Internet.jpg?dl=0
> More attention at position NOT connected!
>
> Side controller like this
> https://www.dropbox.com/s/a8oclk4hj1ok54s/woodpeckercnc.png?dl=0

I see you are using a woodpeeker. So do I with this wiring (see pictures)
and it works great. The problem whit that card is it loosing connection to
the port all the time. Have you expired the same problem?

I had no success trying rewiring the Arduino. Must be something els.
[image: 20181031_204732.jpg]
[image: 20181031_204732-1.jpg]

mikypolice, 2018-10-31:
have you tried to load grbl 1.1j with xloader? that controller if it shows the writing rev 0.9 is also compatible with the fw grbl 1.1j

I've uploaded that using xloader.

The speed of the USB port to be set is 115200.

Try changing usb cable.

Pappagoran, 2018-11-05:
After turning the wifi of it seams to work. Also it seams to be very
important to click the reset before every new work/tool.

mikypolice, 2018-11-06:
So you use USG to control your cnc? And you turning ON (switch from usb to wifi) or turning OFF (switch from wifi to usb) the wifi?

Shahid-lateef, 2020-07-25:
> Dear Grbl experts,
>
> When I use G code sender or Chillipeppr to let my stepper motors move it does not work. But on screen it seems to work so I dont know what I am doing wrong hehe.
>
> For example, I [...]

I have a same problem tell me how you can solve it

MeJasonT, 2020-07-25:
Hi Shahid-lateef
Lateef ? (is that your first name, i know some countries the surname is first)
It sounds like you stepper drivers have an enable signal and nothing is connected, or the enable signal may need inverted in the config.ini.
It is probably best if you ask your question here
[https://github.com/gnea/grbl/issues](url)  

 As this Git branch is no longer being supported, you will most certainly have more luck there.
I understand how frustrating it is when someone re directs you, hopefully this does not offend but its been a long time since members used this site and i would relly like to see you get help with your question.