grbl/grbl #1099: Homming Issue

kardinal7, 2016-10-01:
I've been searching for the issues about something like mine but I couldn't find. That's why I'm creating a new topic.
So I have CNC3020 and I've successfully converted it to GRBL interfacing it with an arduino by the parallel port. but one or 2 things are missing. One is homming that's my issue. So I've wired 5 microswitch in NO mode. Made a pull-up resistor and cap to GND (100uF -> a little to big for my taste. but I have 52ms of rise time).
All the switches work on their on, if i'm scrolling the CNC hand and in file mode.
But when I hit the homming sequence the Z axis crashes in the machine, doesn't detect the switch. Finally one thing i think it's awkward is that after the switch is activated and the machine doesn't move, normally I software reset the arduino and pull the machine from the activated switch and when this **switch goes to the off state the arduino enters in alarm mode again (ALARM: Hard limit).** I think this should happen.

I'm using the variable spindle.

My version of grbl is this one:
>>> $i
[0.9j.20160317:]

I'm going to post my settings here:
>>> $$
$0=10 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=0 (dir port invert mask:00000000)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.010 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=1 (hard limits, bool)
$22=1 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
$24=20.000 (homing feed, mm/min)
$25=100.000 (homing seek, mm/min)
$26=20 (homing debounce, msec)
$27=10.000 (homing pull-off, mm)
$100=400.000 (x, step/mm)$I
$101=400.000 (y, step/mm)
$102=400.000 (z, step/mm)
$110=1200.000 (x max rate, mm/min)
$111=1200.000 (y max rate, mm/min)
$112=300.000 (z max rate, mm/min)
$120=20.000 (x accel, mm/sec^2)
$121=20.000 (y accel, mm/sec^2)
$122=20.000 (z accel, mm/sec^2)
$130=200.000 (x max travel, mm)
$131=300.000 (y max travel, mm)
$132=60.000 (z max travel, mm)

chamnit, 2016-10-01:
It's very likely you've miswired the zlimit switch. It's on pin 12, rather than pin 11. It switched with v0.9 when variable spindle was enabled to access the hardware PWM there. You can check if the pins are reading correctly if you set your status report to show pin state.

kardinal7, 2016-10-01:
I'm quite sure i've wired correctly,  cause if I play around by hand the arduino detects the switch. only not in homming sequence

chamnit, 2016-10-01:
@kardinal7 : Please double check each axis switch. Grbl will report which switch it senses as being triggered in the status report. If its possible that its wired on the wrong axis.

kardinal7, 2016-10-05:
Just confirmed i was wrong I had the switched in the wrong position

Z switch -> on pin 10 
X switch -> on pin 12
Y switch ->on pin 9 

Now it's working so now it's 

Z switch -> on pin 12 
X switch -> on pin 9
Y switch ->on pin 10

Like it should

Thanks, chamnit!

closed 2016-10-05 by kardinal7