grbl/grbl #1100: Grbl download

81zapper, 2016-10-03:
When I go to down load grbl. Any version all that comes up is a whole lot of numbers. .????

What am I missing here ?

Encarnacao, 2016-10-03:
thats Hex you got use Xloader to compile to the arduino or you can go to
https://github.com/grbl/grbl and dowload the zip and put inside your arduino libary folders and then go to exemples and should be there

gerritv, 2016-10-03:
Mnn, no. The file you are downloading is hex format, use Save As to a .hex or .txt file. Then use Xloader to flash to the Arduino. Usually you right click to get the Save As option depending on browser.

Encarnacao, 2016-10-03:
thanks for point out forgot to mention that got to save the files as a .hex or .txt

81zapper, 2016-10-03:
Why would it be that easy.  I just tried it and of corse there is no save as when I left click the mouse 
I also tried the way Encarnacao mentioned and that didn't work either. Any other ideas  or why it is going into hex format when trying to down load.

X3msnake, 2016-10-03:
It's a Github thing Master

You must go to the file and click raw

the browser adress will change from this

https://github.com/grbl/grbl-builds/blob/master/builds/grbl_0_51_atmega168_16mhz_9600.hex

to this

https://raw.githubusercontent.com/grbl/grbl-builds/master/builds/grbl_0_51_atmega168_16mhz_9600.hex

here you can right click and download the raw file as hex

then you load it in Xloader  > http://russemotto.com/xloader/

And upload it to your connected arduino

here's a stepbystep video, sorry didn't find a better one in native english
only subtitled :\
https://www.youtube.com/watch?v=rX8nLEuN2nE

81zapper, 2016-10-03:
Thanks for all your help so far I feel like such an idiot  any way  I figured out the raw way and got everything done but when I go to tools and serial monitor it is still al jibberish. I will check out the vid tomorrow
regards

X3msnake, 2016-10-03:
That's probably because of a wrong the Baud Rate. I believe you must be
connected at 115200 not 9600 or vice versa

Best Wishes Mate ;)