grbl/grbl #1101: homing understanding

Encarnacao, 2016-10-03:
why when i press homing only the Z axis move 5 steps and x and y axis doesnt move anywhere

pitzerwm, 2016-10-03:
Trying not to hijack your thread, but I have a "not going to home" problem too and maybe both of our problems can be fixed with one solution.

My G-code is essentially the same as yours. But nothing moves when I hit home.

When I turn it on, there is not an alarm showing.  It doesn't matter where the x,y,z positions are.  When I hit Home, the Alarm hard limit pops up.  I can reset and unlock to solve that.  I happen to have $23=0.

I'm pretty certain that the limit switches are wired right and are not picked.  I can hit each one with my hand and the Alarm hard limit turns on as expected.

chamnit, 2016-10-03:
@Encarnacao @kardinal7 @pitzerwm : Ok. It seems like there truly is a problem somewhere. Multiple people with the same problem would indicate so. Not exactly sure why this issue would suddenly come up like this, when homing has been working for a long time and there hasn't been any significant code changes recently.

Can you all tell me how you are compiling Grbl and flashing it? Can you also download and use only the current firmware `[v0.9j.20160726]`? And don't enable any compile-time options unless you have to. If you do, tell me which ones. 

FWIW, I did notice that the Arduino IDE has updated their compiler to a new avr-gcc version v4.9.2. They haven't changed in several years. Grbl compiles to be much larger than before and can't say for sure if this has an effect on how it runs.

pitzerwm, 2016-10-03:
In my case, I have a Phoenix controller, and 9i.  I am not smart enough to flash anything.  I notice that some people have $23=3, I have mine at 0, which I expect it to go to the back left corner.

There is noise suppression built into the card.  I have 6 NO switches wired in parallel for each axis  If I manually run the gantry into a limit switch it gives me the hard limit alarm..

Is there a command that would "read" the switches?

Encarnacao, 2016-10-03:
i have version 0.9J.20160317 i have uploaded using arduino ide i can try to wipe EEprom and flash using Xloader to see if any diference
@chamnit  can you point me link for the version v0.9j.20160726 i see on github but dont see where is version v0.9j.20160726

thanks

gerritv, 2016-10-03:
Set $10 to 31, then see what the Lim: part of response message is. These are the 3 limit pins.

pitzerwm, 2016-10-03:
Thanks gerritc, now how do I read the status?  I did $# and G21 is in the list, but nothing with it, just a string of G commands with ok after.  I also found how to read the build i. 20150329

gerritv, 2016-10-03:
You send a ? to get responses that include Lim:xxx message. Or use a gui that does these things for you :-) Reading this should help also: https://github.com/grbl/grbl/wiki/Interfacing-with-Grbl

pitzerwm, 2016-10-03:
Thanks gerrity, When I send a ? I get a string of G commands with G21 in the list.  then an ok on the next line.

Encarnacao, 2016-10-03:
ok so far im having a hard time for a long time to understand how homing really works and what to exepect from it, i have read alot and seen some videos but nothing that i do works
for what i understand when you do $H the machine is supose to start runing bump into a switch then goes to next and then to next and stops?
now when i do homing $H it will bump into Z switch and doesnt stop will bump into machine and forces to get out so i have change to
#define HOMING_CYCLE_0 ((1<<X_AXIS)|(1<<Y_AXIS)) 
#define HOMING_CYCLE_1 (1<<Z_AXIS)
for goes first the X and Y first and same thing the switches will not make it stop
but if i run a normal code like X200 machine will move and hits switch and stops like expected
before i had problems with noise issues on the machine that made it trigger the switches so i have made some changes i have put all cables shielded conected mesh to ground for dissipate noise and i have put NC switches with externall pullup mantaining the internall pull ups and changed to $5=1 , sense then i never had problems with triggers i have test machine runing for 2  hours and all goes smooth as expected
i have also switch to grbl [v0.9j.20160726] like @chamnit  sugested, and i still have same problem i just cant make it to home, at this point i dont know what to expext from homing sense i never got it to work, any help to figuer out in whats going on would be apreciate

gerritv, 2016-10-04:
@pitzerwm I am assuming you are using a terminal to talk to Grbl? Send several ?'s, you will eventually see a response like <Idle:......> . Alternatively use a gui that is known to work, e.g. GrblPanel (Windows only) or UCS (many platforms) and save yourself all the experimenting. That said there may or may not be an actual problem in Grbl. I unfortunately do not have a machine to try with at the moment so can't help much more.

pitzerwm, 2016-10-04:
I'm using Bcnc which I assume is the GUI that you refer to.  Under "Control" there is a status line that says idle/run/connected/Alarm.  But nothing about the limit switches unless I trip one and then I get Alarm Hard Limit.

I was just trying to verify that I have the limit switches connected right.  I checked them and I'm pretty sure that they are right.  I can't get HOME to do anything but pop up the Alarm Hard Limit, the problem is that none of the switches are tripped and the gantry doesn't move.

Thanks again for all help.

gerritv, 2016-10-04:
@pitzerwm I would look in the gui for a display of the data coming back from Grbl. That will show the Lim: as part of the usual response to regular ? polling. Or ask on Bcnc github how to see that information.

pitzerwm, 2016-10-04:
Thanks, I thought that is where I was:o)  I'll get out a map.

swarfer, 2016-10-04:
no has asked what the config.h settings are, nor what the various invert masks are.

pitzerwm, 2016-10-04:
Thankfully, I got it working.  In my case it was a wiring issue, one axis I had the hot & grd reversed.  It worked manually, but GRBL wasn't buying it.  Thanks for all of the help.

chamnit, 2016-10-04:
@pitzerwm : Thanks for the update. This problem is almost always an issue with mis-wiring and the misunderstanding of how hard limits work inside Grbl. It trips upon ANY pin change, not just on trigger. There are good reasons for this (which I will not elaborate on), but it can cause some confusion when wiring up your homing switches, if you are not aware of it.

Encarnacao, 2016-10-04:
homing now is working i still dont understand why the only change i did was  from  $23=0 to
$23=7 (homing dir invert mask:00000111)

now there is something happening that i dont understand machine goes to x switch y switch then off sets and then goes again to x switch y switches then offsets and then goes z down and up then again z down and up and stops

it seems that is repeating same actions 2 times any idea what is happening

chamnit, 2016-10-04:
@Encarnacao : That changes the direction Grbl searches for the limit switches. By default it moves in the positive direction on all axes. So, tool should move right, back, and up when $23=0. If any of these axes are not moving in these directions, then you have your coordinate system incorrect and need to alter your step direction mask to fix it.

lalo-uy, 2016-10-04:
It goes fast, backup a little and do it again slow in order to be por
precise

Encarnacao, 2016-10-04:
@chamnit  i still dont understnad correc tly how really works homing and coordinate system i have changed to  $23=7 cause i wanted to have machine on front and left side cause usualy when i do some works i always design them on positive side of y and x never on negative sides, so how i think that homing works is that u click $H and machine goes to a 0 position then i would send a comand to go for exemple to X100 and y 100 and then set to x0 y0 and z 0 and start my work, however you migh be correct that should go to positive sides and that would make machine spindle been on the back when not in use.

so from what @lalo-uy  sayed it should be normal that machine repeats 2 times same move, first time goes fast and 2 time goes slow to be more precise on position

please bare with me, im still trying to understand how this really works, plz correct me if anything wrong

chamnit, 2016-10-04:
@Encarnacao : Grbl sets it's coordinate origin in the back-right-top corner of the machine, such that the machine coordinates exist in negative space. This is different from 3d printers. [This is why.] (https://github.com/grbl/grbl/wiki/Frequently-Asked-Questions#why-is-grbl-in-all-negative-coordinates-after-homing-or-it-so-annoying-and-not-what-im-used-to) Also, please note that you can't change this behavior unless you alter the config.h to do so. (See config.h for clear explanations on what each option does and how to use them.)

Next, please take the time to read the Wiki documentation on how to configure your machine. In particular, the section about the [homing setting](https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.9#22---homing-cycle-bool). It describes how Grbl expects to move and how the $23 setting changes the direction of homing. 

Lastly, there is no reason to touch off 3 times during homing. I'm aware that 3d printers do this, but all you need is a quick search and a slower locating. That's it.

closed 2016-10-04 by chamnit