grbl/grbl #1103: Tool Change in latest version

DOUG888, 2016-10-05:
Couldn't find an answer to this elsewhere so:  
In the yet to be released GRBL will M6 be supported, the reason I ask is that I run a Bench Mill with a Diode Laser attached (this is becoming more and more popular) and to cut at different intensities a M6 tool change is ideal, also most CadCam software operates with a M6 for general Milling.

To my way and I am sure many others the inclusion of M6 would be a benefit to many users.

I realize that space is a challenge in the UNO but tool change is a widely used feature.
I read in the forums of the use of a card reader or memory chip to bolster memory, if they ran via SDA/SCL that would solve that issue.

Thanks again for all your hard work, it is appreciated.
Doug

chamnit, 2016-10-05:
@DOUG888 : First, you can do M6 tool changes already, if you have a GUI that supports it. Grbl provides the basic commands to do this, namely a tool length offset command to alter Z-changes in tools. All a GUI needs to do is intercept an M6 command and insert a (customizable) macro program to do the tool change. There are a few GUIs that do this. bCNC is one that I know does this. Some OEM GUIs do too for certain products they sell.

Implementing M6 directly into Grbl itself is a bit more involved that you may think. You have to store and manage tool data, plus the type of tool and how they are used. Because tools are very specific to the CNC machine they are used on, it's very difficult for something like Grbl to support every scenario, store it internally, and ensure any machine specific motions for a tool change are performed correctly. It would require lots of firmware level configurations that would be too complicated for a typical user. 

Lastly, I'm failing to see how an M6 tool change would make your laser cutter work any different. Most laser cutters use the spindle speed output to alter intensities/power. Not M6. 

Anyhow, please remember that the Grbl project is not focused on doing everything for everyone. It tries to provide all the core functionalities needed in a CNC machine and not anymore than it has to. This is primarily so I can concentrate on solving things like real-time overrides without unnecessary distractions or problems. The next Grbl project on an ARM processor will do a complete CNC system and aim to be a Mach3 replacement. So don't worry, just be patient.

closed 2016-10-05 by chamnit

DOUG888, 2016-10-05:
Sonny

Thanks for the prompt reply. I wasn’t aware that it could be actioned in that way, I will post my, now informed, query on UGS forum and see where that leads.

The only reason I was looking at M6 was that the Tools have spindle speeds built into them and CadCam software can use various tools.

Thanks for all your hard work.

The journey continues.

Doug

 

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

109JB, 2016-10-06:
@DOUG888 

CAD/CAM does not assign speed to tools. The CAM output has tool changes and then spindle speed commands but they are separate. They are not one in the same. For example, a CAM, or hand programmed tool change would look as follows in a production type CNC machine:

M6 T2  (M6 means Change tool to T2 tool numberr 2)
G43 H2  (Assign the tool offsets for tool 1 in the tool table
M3 S3000 (Start spindle clockwise at 3000 rpm)

As you can see the tool change has nothing to do with speed. Take the above code, the next line could easily be:

M3 S2000 and then the spindle would slow from 3000 to 2000 rpm.

DOUG888, 2016-10-06:
Yep….You are correct, I was getting feed rates and spindle speeds mixed up…..more sleep needed  J

My CadCam Software, in nesting, only allows for one S setting, I will look at individual component programming and see what is available to me there.

I am still learning the ins and outs of the software.

Thanks for your help and patience.

Doug

 

https://www.avast.com/antivirus