grbl/grbl #1106: Constant Feed

luni64, 2016-10-12:
I need to implement a constant feed setting on my lathe. I.e., I want the z-Motor to move at a constant speed until I stop it. To do so I want to set relative mode (G91), Feedrate and then continuously send "G1 Z100" commands. 
Does this overflow internal counters if running very long? If so, can I periodically reset the counters without having the motor stop?

Thanks
Lutz

luni64, 2016-10-13:
I meanwhile glanced at the code and it looks like you are using int32 for the step counters. 

At 1µm per step this would give a maximum movement of   2^32  µm = 4294m. Assuming a feedrate of 20mm/s this would give a maxiumum "time to overflow" of 4294m / 20mm/s = 59.6 hours. I.e. I could run my leadscrew for about 2.5 days without risking an overflow (which would be sufficent of course) 

Question: is this correct or do I miss something in my calculation.

chamnit, 2016-10-13:
@luni64 : You are correct. There is a limit on how far you can move in one direction. Grbl was built on the assumption of a finite machine volume. Not continuous motion. If I recall correctly, I think the counters should be good to tens of meters in the worst case. Tens of meters is much larger than any CNC machine I've ever seen.

luni64, 2016-10-13:
Definitely, but the mode I'm after would be semiautomatic, i.e. I would have the leadscrew moving at constant speed and would couple the support in/out mechanically. Pretty much as you do in a conventional lathe. This means that the motor might be moving much longer than the equivalent of a few meters.

I'll do some experiments this evening. Would grbl issue an error if overflowing or would it just hang somehow?

chamnit, 2016-10-13:
@luni64 : Not sure. Like I said, Grbl was designed for a finite volume. If you need something that continuously running and don't care about multi-axis synchronized motions, then I suggest that you just implement a simple single-axis step controller to do what you need. That is simple to do.

closed 2016-10-13 by chamnit