grbl/grbl #1107: Grbl v1.1 Public Beta Release!

chamnit, 2016-10-13:
#### ALL: 
This has been a long time coming and it's finally here! Grbl v1.1 is available for public beta testing at this new [project site](https://github.com/gnea/grbl/). The new version has some huge features:

- Real-time overrides for rapids, feeds, spindle-speed, spindle-stop, and toggle coolant states! All of these alter the running state within tens of milliseconds for instantaneous feedback. It feels totally different than how others do it, where you have to wait several seconds to a minute for a change to go through. It's awesome. You can have your CAM program generate a conservative feeds and speeds and then alter them on the fly at the machine to optimize your job. You can also feed hold and stop the spindle and coolant (vacuum) mid-job to inspect how your part is coming along or to clean out swarf!

- Laser mode! Grbl now officially supports laser cutting and it works with spindle speed overrides (aka laser power) too! It's just a simple `$` setting to change and Grbl will then move continuously through consecutive G1/G2/G3 motions when an `S` spindle speed word is present. 

- Jogging mode! This new jogging command is different than feeding Grbl g-code commands, mainly because you can cancel it and Grbl will automatically feed hold and flush the buffers. It also doesn't alter the g-code modal state, so you no longer have to track them. This helps reduce inadvertent crashes when you forget and start you job.

There's a bunch more features, but those are the big ones. Keep in mind that this is a _**BETA**_ release. There most certainly are some bugs here and there. So, please use this if you'd like to help out with the testing. Please report issues at the [NEW SITE](https://github.com/gnea/grbl/issues), rather than here. 

#### IMPORTANT NOTE: Grbl v1.1 altered some of its interface to cram more data into the status reports and make it easier for developers to maintain their GUIs. Grbl v1.1 only works with GUIs that support it. So far, this includes (in order of most v1.1 support): 
- [Grbl-Panel](https://github.com/gerritv/Grbl-Panel): Just about full compatibility.
- [UGS Platform Nightly Build](http://winder.github.io/ugs_website/download/): Click the bottom-most link. The overrides pane is enabled via a menu item. A few new features not yet implemented.
- [bCNC](https://github.com/vlachoudis/bCNC/tree/V1): Operational, but no override controls yet.
- [PicSender](http://www.picengrave.com/PicSender.htm): Status unknown, but they are working hard at it!

The Wiki is not yet up and operational, but there is a lot of documentation that I've written in the [/docs/markdown](https://github.com/gnea/grbl/tree/edge/doc/markdown)  folder. You can navigate to it through the Github website or click the link. Github will show it in markdown form.

Thanks everyone for being so patient! It's been a tough couple of years for me and my family, but we are finally seemingly in the clear. That and with this beta release finally out, it's a huge weight off my shoulders. Cheers!

splitn2, 2016-10-13:
you're a legend! I will be testing the new release asap. Is this Uno or Mega ? cheers

PicEngraver, 2016-10-13:
Thank you Sonny, and congratulations on a job well done.

John

tklus, 2016-10-13:
Sweet! Very excited about the new features! Hoping to start testing this weekend.

Thanks for all you have done Sonny!

Tim

winder, 2016-10-13:
Congratulations Sonny!

jahnj0584, 2016-10-13:
Amazing. I'll be testing feed over ride tomorrow. I have some 0.5mm micro
bits and wanna see when exactly they break, since 500mm/min is pretty slow
right now but the only thing that works.

I also donated today :)

DOUG888, 2016-10-13:
Sonny

Thanks for ALL your work and time that has been put into GRBL, without your dedication this project would have disappeared long ago.

Be proud of what you have achieved.

Thanks

Doug

 

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

X3msnake, 2016-10-13:
> <javascript:_e(%7B%7D,'cvml','notifications@github.com');>]
> website/download/> : Click the bottom-most link. The overrides pane is
> WHDULs5yTQtOks5qzY5_gaJpZM4KVZnG> . <https://github.com/
> notifications/beacon/AT7z7sJzXcZ4rcibnPI0zEf60zfxKa

Cool Master,

Thank you for all your effort and commitment to the Project. :)

Encarnacao, 2016-10-13:
congratz
i have went to site https://github.com/gnea/grbl/ to get the grbl 1.1 i have uploaded but i dont see the new functions and when i open still says Grbl 0.9j ['$' for help]

swarfer, 2016-10-13:
I think you have done something incorrectly.
I get
   **** Connected to COM11 @ 115200 baud ****
   Grbl 1.1c ['$' for help]
after loading the new code to my UNO

Encarnacao, 2016-10-13:
i have fix no idea what happen i had to remove all hold grbl folders from libary, was some conflict with the old ones eavean know i changed names

jahnj0584, 2016-10-13:
You have to reflash your arduino too, right?

Encarnacao, 2016-10-13:
yes i had to reflash after removed the files

Encarnacao, 2016-10-13:
now shows the lines with out info
$0=10
$1=25
$2=0
$3=0
$4=0

doesnt show the  extra info in what does eatch line like before
ex:
$0=10 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=0 (dir port invert mask:00000000)
$4=0 (step enable invert, bool)

electrokean, 2016-10-13:
Yes, you have to delete or move any old library folders.  The Arduino environment can't handle multple library folders containing files of the same name.

electrokean, 2016-10-13:
I presume the descriptions for each setting had to be removed to make space for other functionality.

gerritv, 2016-10-13:
@Encarnacao  there are no text descriptions anymore. I highly recommend reading the files in doc directory before asking questions. There are many changes to the protocol as Sonny stated int he message announcing 1.1

chamnit, 2016-10-13:
Thanks everyone for the kind words! 

Please keep an eye out on stability. The addition of overrides eats up a quite of bit of memory, so the planner buffer had to be reduced in size a little. Right now, it's down just one, but it's very close to the memory limit. You'll see "low memory" warnings when compiling. I've checked how the stack looked in the worst conditions and we're still a few dozen bytes below the 2K RAM limit. So, it should run ok. If it doesn't, try reducing the planner buffer size by one.

Also, the Arduino IDE has recently updated the avr-gcc compiler from v4.8.1 to v4.9.2. For some reason, it compiles the code about 0.4KB to 1KB larger than before. Even with this, the default build should still fit on Arduino Duemilanove/Nano. I've made sure of that. If compile-time options are added, you may need to use an Arduino Uno, burn the Uno boot loader on your Nano to free up 1KB of additional usable flash, or use an older Arduino IDE with v4.8.1 to get it to fit.

jahnj0584, 2016-10-13:
Where is the hex file to flash?

jahnj0584, 2016-10-13:
Nevermind. Had to look around and delete the old grbl library (wasn't
accepting the new ones). Didn't show it as an error when i had the IDE as a
1/4 screen.

gerritv, 2016-10-13:
@jahnj0584 I use platformio with visual studio to avoid the Arduino idea mess.  Check out platformio.

tklus, 2016-10-14:
Loving the new firmware! This is going to be amazing!

Did a quick test of the feed override on my almost completed ox! Works awesome!

http://youtu.be/6v5JRbw2-EU

chamnit, 2016-10-14:
@tklus : Nice! Thanks for sharing.

Lob0426, 2016-10-15:
Thank You for your work. I have been running it for about a week and it is working well on our Chinese blue Laser machines. "Laser Mode" is awesome. Running very quietly as the stops are gone, very smooth!

Thank you for the advice the other day. I moved the buffers back to default and no problems so far at all! Single pin Homing is working just fine!

tklus, 2016-10-17:
Can we put a link on the front page of this grbl repository that links to the 1.1 release project page. For those who are not following the original grbl, it may be hard to find. The only link I know of is in an issue log that states 1.1 is released.

Thanks
Tim

chamnit, 2016-10-18:
@tklus : That was intentional. I did not want everyone to start using it until its closer to a stable release. I have a couple more tweaks to do. Thankful, it seems pretty stable, since I haven't heard anything from anyone. So, inviting more people to test may come sooner than later.

tklus, 2016-10-18:
That makes sense, I was running mine tonight and paused the program, used the spindle over ride to turn the spindle on and off several times and then hit cycle start to continue the job. It continued to run for a little bit and then alarmed out. I am using grbl panel. My guess is that it used up the moves that were buffered then alarmed out from the spindle over ride... I haven't tried to recreate the problem yet though... thoughts?

chamnit, 2016-10-18:
@tklus : Hard to say whether the GUI or Grbl is causing it. I did find a few minor bugs related to spindle overrides but nothing that would cause an alarm. If you can produce steps to recreate it, please post a new issue at the other site.

tklus, 2016-10-18:
Sure thing, thanks

jsb1, 2016-10-20:
Damned! I just implemented real time spindle updates for lasers in v0.9: [https://github.com/jsb1/grbl](https://github.com/jsb1/grbl). Forget it.
I'll be Your beta tester...

jsb1, 2016-10-20:
Well, there is a little difference: What I do is recalculating spingle pwm in every line segment while accelerating. Would that be a userful feature for v1.1?

109JB, 2016-10-20:
Great work.  I tried it out using GRBL Panel to give it a trial run.  I am working on modifying my GUI to work with it when I have time.

I have a new machine build in the works and it will use GRBL exclusively. In that vein, the machine will have a small amount of backlash (0.004") and I'm wondering if backlash compensation made it into this version?  I didn't see any reference and a quick look in the config.h I didn't see it.  Not a complaint as the other features were needed and seem to work great.  For milling, the backlash comp would make the feature list pretty complete but I can live with 0.004" backlash for a bit.

jahnj0584, 2016-10-20:
You should either remove that backlash or program it away in CAD and not do it controller side. No-one i know uses jt, even with mach3

109JB, 2016-10-20:
@jahnj0584 
You must not know many people using Mach3, or LinuxCNC, because the use of backlash compensation is quite widespread in both cases.  The people that don't use it either have negligable backlash, or they just don't require the tight tolerances.  

Removing backlash is ideal, but many times easier said than done. I have eliminated as much as I can the way the machine is now and it is the best it can be without drastic modifications that I am not willing to take at this venture. Backlash compensation could make it so I never have to.

"Programming it away in CAD is something that quite frankly is not standard nor desirable. Why would I want to make circles with offsets at every quadrant, or make my model longer or shorter than the actual desired measurement. That is what would be required in CAD. Also, it is dependent on the direction of travel. If no reversal of direction is made then no compensation is needed, so applying it in CAD is fraught with problems. 

In any case it was not a question of whether backlash compensation should or should not be used, but whether it is implemented or planned.

chamnit, 2016-10-20:
@109JB : FWIW, backlash comp just missed the cut in v1.1. I was pretty disappointed too, but I simply ran out of flash and memory. Although, I'm not entirely sure how much it'll take. It might be small, but I need all of the remaining space to address any bugs or implementation oversights. 

When v1.1 goes to master, I'll add backlash comp to the Mega branch when I integrate all of v1.1 into it. I'll be able to see how much room it'll take to implement it. If I can cram it into the 328p version, I'll probably do so as a compile-time option only, since most Grbl-based machines are belt-driven (zero backlash). And, it'll likely only fit on an Arduino Uno (or a hacked FTDI-based Arduino with the Uno's 1KB smaller bootloader.)

109JB, 2016-10-20:
@chamnit : Thanks.  I knew it was a planned feature and understand the limitations on the UNO platform.  I hope you can cram it in there somehow as I have several UNO's with the 16U2 available and would like to use one for this new machine.  Not a big deal if you can't, just have to find a MEGA with the 16U2 if you can't.

Thanks again for your hard work.

John B.

McNugget6750, 2016-10-23:
@chamnit Thank for the great work on grbl! I now converted over to v1.1 but I still have some issues that I had before and maybe you have the answer to these issues:
* I'm using grbl on an Arduino Nano on my K40 laser.
* I used grbl 0.9j with some extensions to be able to laser engrave at a decent speed using planned S commands.
* I have issues cutting arcs as can be seen in the pictures below:

This is the dxf file I'm trying to cut. A small quad copter design. This is the import within DXF2GCODE
[http://www.microforge.de/grblproblem/DXFImport.PNG](http://www.microforge.de/grblproblem/DXFImport.PNG)

This is the interpretation of the generated gCode loaded into grbl Controller
[http://www.microforge.de/grblproblem/DXF-GCodeInterpretationBy%20grblController.PNG](http://www.microforge.de/grblproblem/DXF-GCodeInterpretationBy%20grblController.PNG)

And this is the cut that comes out of grbl. Please note that some parts are cut correctly but some larger parts are cut completely asymmetrically, the holes are in the wrong locations, the complete shape of the quad base place is asymmetrical and the motor mounts at the arm extensions are entirely unusable. 
[http://www.microforge.de/grblproblem/LaserCutPartsByGRBLv1.1.JPG](http://www.microforge.de/grblproblem/LaserCutPartsByGRBLv1.1.JPG)

Even though it looks like a step error, I'm very certain it is not a mechanical issue. I cut several parts before with no issues and only more complex files seem to produce these errors. I remember that I have seen this before a couple of years ago when I tried to implement a mill with this toolchain...

Any ideas?

[image]

[image]

[image]

jahnj0584, 2016-10-25:
I would like to know which linear system you are using that can't be fixed.
I run ballscrews and have an acceptable tolerance of 2 thou when cutting
for clients.

109JB, 2016-10-26:
@jahnj0584 : I didn't say it couldn't be fixed, but that I am not willing to invest the time or money to fix it at this point.  Having said that, the machine is an X-Y table that uses acme screws and the machine is old and has a lot of use. It has plain bronze thrust washers on the screws and the lead nuts are not adjustable on this machine. All of this adds up.  Sure I could buy some ball screws, some thrust bearings, spend hours making mounts and brackets, but I would have more into that stuff than the rest of the machine.

Regardless, backlash compensation is something that is even included in production machining centers and it is used in those machines. It is not by any means an unused feature. Fanuc, Fadal, Cincinnati, and other production controls have backlash compensation.  Here is an example I found for Fadal in a simple Google search. https://www.fadalcnc.com/media/pdf/Backlash_Adjustment_Procedure.pdf

109JB, 2016-10-26:
@McNugget6750 : I see that you said you use an Arduino Nano to run GRBL. I also used to use a Nano on my machine.  The one I used was a clone that used the CH340G USB to serial converter chip. The genuine Arduino Nano uses a  FTDI FT232RL USB-serial chip, and the Genuine UNO uses a Atmel 16U2 or 8U2 chip.  The reason I mention this is because I found that the CH340G chip would randomly cause transmission errors between the GUI and GRBL. The nature of the errors is such that some could be missed and some could result in an error response from GRBL. If the GUI is not set to stop on errors, then GRBL will just skip the errant lines and move on. Now I have no idea if this is what is happening to you, but is a possibility. These errors can also happen on the 16U2/8U2 chips. I don't think anyone has tested the FTDI FT232RL, so that one is unknown. The 16U2/8U2 firmware has been fixed and can be flashed to the chip to remove this possibility. The CH340G can't be fixed at this time.  Since there is no fix for the CH340G, I decided to abandon it in favor of an UNO with a re-flashed 16U2 for my machine. The 16U2 generally doesn't have the problem even with the stock firmware at 115200 baud. You can read more about this in "Known Bugs" part of the WIKI with links to the original issue and a link to the new firmware.

https://github.com/grbl/grbl/wiki/Known-Bugs

McNugget6750, 2016-10-26:
@109JB Thank you for this detailed response! I got my specific problem solved here: [https://github.com/gnea/grbl/issues/20#event-834899833](https://github.com/gnea/grbl/issues/20#event-834899833)
In my case, it was the motors shutting down after every command because $1=25 was set to the default value that caused the motors to turn off after a random command, sometimes even in between successive commands. I have no answer to the WHY this should ever happen because there are always more moves available in the command buffer - checked and tested.

However, in the other thread the problem with the CH380 / CH340G was mentioned and it got me thinking:
IF there is a communication error with a bit-flip (I assume this will be the symptom), the command can be executed as long as a letter is not changed (G, X, Y, ...) . In case of absolute coordinates, this will almost always result in properly finished runs with single failures somewhere throughout the run.

In case an ERROR gets reported back, I now have the GUI set up to stop and not continue. I also plan on activating the ECHO setting in GRBL and run a large file through it to test the communication. So far I had no errors at all. All my problems were related to the motors shutting down. But I won't make the mistake and disregard the fact that there has been numerous people reporting faulty CH340G/CH380.

I will keep my eyes open and if in doubt swap the Nano to a genuine Nano. I also have a couple of FTDI adapters lying around that I can solder to the Nano in order to test this known bug and rule it out/debug it.

109JB, 2016-10-26:
@McNugget6750 :  Be aware that the transmission errors won't necessarily result in GRBL detecting an error. For example, the following 2 commands:

G0 X1.234 Y5**.678 Z9.012
G1 X9.876 Y5.432** Z1.098

could get transmitted as 

G0 X1.234 Y5Z1.098

where everything in bold gets dropped.  GRBL would see that as a legitimate command, not generate an error, and continue.

The FTDI chip has not been tested for this, and may be OK, but for me I decided to go with what has been fixed.

Glad you got the other issue resolved.

McNugget6750, 2016-10-26:
@109JB Yes, this is a possibility and would look terrible in the final result. I agree that an engraved iPhone would probably loose a bit of value with an error pronouncing the dumb decision to use a $2 part as a main controller... ;)
For now, I'm only engraving and cutting rather worthless wood. I will let you guys know if as soon as I encounter an error like this! I find it curious though that no one tested the FTDI extensively.

chamnit, 2016-10-26:
@McNugget6750 : I hajve only heard of issues with the CH380 chip. I think it would be safe to assume the FTDI FT232RL chip is ok since all other FTDI chips have never had a problem and there have been no reports. 

As for pausing, your gcode program has a lot of arcs. It's possible that Grbl ran out of queued motions by the time it finished one arc and started the next. Arcs are pretty heavy in calculations for its setup, so since you are running relatively fast, this wouldn't surprise me. It also wouldn't surprise me that the CH380 has a larger latency that is adding to this. If you run you gcode program with converted arcs to G1's, the pausing should go away.

LCPINAGE, 2016-10-28:
Hi @chamnit !!! Great job, excelente!!

I am testing here in my laser engraver core XY and works nicely the laser mode!

But I am having trouble in homming. It homes x axis normally, and for Y axis, it hits the first time but then go to the other direction instead of going back  again...

I changed the configurations correctly in config.h, basically activated #DEFINE COREXY and
#define HOMING_CYCLE_0 (1<<X_AXIS)
#define HOMING_CYCLE_1 (1<<Y_AXIS)

Is anybody trying it on a core XY machine also?

Thanks

chamnit, 2016-10-28:
@LCPINAGE : Thanks! I don't have a CoreXY machine to test on so I have to depend on user feedback to find any issues with it. The CoreXY implementation in v1.1 should be exactly the same as in Grbl v0.9j, so if your machine behaves the same with v0.9j, then there might be an issue.

I do know that some CoreXY machines are not configured according to the theory. For example, motors are positioned in the wrong spots or move in the wrong direction. Check your machine against the CoreXY theory page. Make sure that the motors are rotating the correct way for a given motion according to the equations.

LCPINAGE, 2016-10-28:
@chamnit works nice and neat with v0.9j!

Same mechanical and parameters configuration... Any ideias? It is just the last Y movement that goes to the other way...

chamnit, 2016-10-28:
@LCPINAGE : Ok. Thanks. I'll open a new issue at the new site for this and I'll do a diff to see what is different. Maybe it's a typo.

mkeyno, 2016-11-07:
thanks to all GRBL contribute but I was wondering why not made big enhance  to GBRL by moving from AVR 328p to more powerful  board  like MEGA 256 or STM32F103 ? I think not really necessary to work hard push back AVR328p limitation

chamnit, 2016-11-07:
@mkeyno : For the record, the Mega2560 is not anymore powerful than the 328p. It just has more flash and RAM. And to re-iterate for the hundredth time, Grbl is moving on to ARM after v1.1's release.

The intent of the Grbl project has always been to make CNCing accessible to everyone. Given that everyone has a (or a dozen) 328p-based Arduino, it's very easy for anyone to flash Grbl and see what CNCing is all about.

Grbl v1.1 represents just about everything I think is necessary to have good basic machine control. I originally designed the current master v0.9 with all this planned out, so I knew long-ago that all of the remaining features in v1.1 would (barely) fit. When v1.1 is released to master, I think it's safe to say 'mission-accomplished' with the Grbl project.

Also, I did not want all of the current Grbl users to feel left out from the awesomeness of real-time overrides. I don't think most people understand what it means to have this until they have it and then they don't. It's night and day.

Finally, because of the space restrictions on the 328p, this made Grbl very efficient and compact, which is a very good thing. It will make the jump to ARM that much more impressive, especially with all of the things I have planned for it.

mkeyno, 2016-11-08:
thanks @chamnit for information , I should dig more about couple of branch that already developed before paste my note , however I drop GBRL long ago because it hadn't  enough pin and peripheral for my CNC and  I switched to repitier and marlin  but after long time playing , have noticed how important to have hardware pin to control real time overrides , that's why I come back to GBRL, specially when I add the ESP transparent serial bridge  to my board , and eagerly waiting for your final 1.1 release

quangvi, 2016-11-24:
thank @chamnit  i'm waiting for your final ver 1.1 release , hi @109JB , i write the Backlash Compensation in Grbl v 0.9j is so good, but i use the STM32F4 Discovery and eclipse C , if  @chamnit final ver 1.1 release , i will write Backlash Compensation, i will sent my project to you, it's maybe waiting a long time, after ver 1.1 release, i had use grbl ver 0.9j mod to cnc machine 5 axis, it's have 1 or 2 problem but the information in ver 1.1 may be solve my problem ,one more thank @chamnit,

quangvi, 2016-11-29:
Dear @chamnit   in line 298 printPgmString(PSTR(" (homing dir invert mask\r\n$24="));  bug fix 
printPgmString(PSTR(" (homing dir invert mask)\r\n$24=")); it's had problem to setting in Grbl  3.6.1

quangvi, 2016-11-29:
Dear @chamnit in file defaults.h line 50 : can change if use classic mode
#ifdef USE_CLASSIC_GRBL_INTERFACE
    #define DEFAULT_STATUS_REPORT_MASK ((BITFLAG_RT_STATUS_MACHINE_POSITION)|(BITFLAG_RT_STATUS_WORK_POSITION))//
  #else
    #define DEFAULT_STATUS_REPORT_MASK 1 // MPos enabled
  #endif

chamnit, 2016-11-29:
@quangvi : Thanks for the bug fixes. However, I'm going to completely remove the CLASSIC reports option from the config.h file. This was intended to be a bridge for users and GUI developers to update their interface for the new one in Grbl v1.1. Instead, what's been happening is GUIs are not updating and you have people using interfaces for v0.9j, v1.1, and v1.1 classic. This is not what I intended. 

Instead, please try to get your GUI updated for the new v1.1 instead. The documentation of all of the changes are in the /doc folder of the v1.1 repo.

Finally, please post all issues related to v1.1 at the v1.1 website. Not here. Thanks.

arkypita, 2016-12-04:
I have published source of my LaserGRBL GUI, a Laser optimized GUI.
https://github.com/arkypita/LaserGRBL

It is a young software, but stable and usable. I think my GUI is compatible with v1.1 cause i have developed it targeting to 0.9 but testing on 1.1 too (and some feature are developed explicit for 1.1, like error/alarm/config number decoding).

nsiatras, 2016-12-04:
You did a really good job with the v1.1. I just released the new version of my GCode sender which is compatible with the new version. Fill free to try it https://github.com/nsiatras/sourcerabbit-gcode-sender

quangvi, 2016-12-05:
Thank You, every one

chamnit, 2016-12-10:
Ok all. Grbl v1.1e is officially release to master! It's available [here](https://github.com/gnea/grbl/releases). The main grbl/grbl site will be shelved and just retain prior versions. v1.1 will be maintained at the new site. Enjoy!

closed 2016-12-10 by chamnit

Codymarchant, 2016-12-22:
Hi,

I am having an issue when using the IDE to compile the new version 1.1. It seems to fail every time. I have tried it on to computers now and neither works. Errors are;

#required homing cycle not defined
#wco refresh must be greater than one
#override refresh must be greater than zero.

multiple libraries were found for config.h

Any help would be great as i would really like to start using version 1.1.

109JB, 2016-12-22:
You have to delete the 0.9 Grbl library from the IDE folder and replace it with the 1.1 library folder.

[29 more replies]