grbl/grbl #1108: Question about limits

109JB, 2016-10-13:
I have my machine wired in such a way that GRBL does not have control of stepper driver enable/disable, so I can't check it directly.  So, my question is, when hard limits are active, and the machine is driven to a point where a limit switch is activated, does GRBL disable the drivers (If they are hooked up that way.) ?  

Or, does GRBL just halt sending step pulses, leaving the driver enabled?

The WIKI only says "When the switch triggers, it will immediately halt all motion, shutdown the coolant and spindle (if connected), and go into alarm mode, which forces you to check your machine and reset everything."

It is not clear if the drivers are disabled at this point.

Thank You,

John B.

chamnit, 2016-10-13:
@109JB : Good question. It's supposed to disable the stepper drivers, but I think I have the order of operations in the hard limit interrupt wrong. The alarm flag should be set before the mc_reset() call to shut things down. 

Right now, I think Grbl will keep the drivers enabled. If I switch the order, I'll disable them immediately upon hard limit. It'll apply the step idle delay before it does though to help with stopping the machine briefly. 

IMHO, it can go either way.

109JB, 2016-10-13:
I have another question/suggestion, at least for the UNO GRBL versions that are limited in the number of available pins.  

The default configuration for the pins in GRBL is to have 3 pins for limits, X, Y, and Z.  Freeing up 2 pins is possible by combining all limits to one pin.  For example, using normally closed switches in series, or normally open switches in parallel, they can all be wired to one pin, freeing up the other 2 pins for things such as another axis, or other uses.

This could also be set to work for homing as well. If one pin is used for the switches, the homing sequence would require homing only one axis at a time.

All that would be lost is the ability to see which limit tripped.

This is an option that LinuxCNC allows, but I think it would be extremely beneficial for the UNO based GRBL installs in order to free up a couple pins for other possible uses.

jahnj0584, 2016-10-14:
What else do we need pins for?

chamnit, 2016-10-14:
@109JB : The pin issue isn't the limiting factor now with v1.1. It's entirely flash and RAM. That said, back when I implemented limit pins and homing, I thought about combining them into one pin to free up two. I ended up deciding against doing this because it just adds more more thing you have to explain and support. 

Technically, you can already do this in Grbl. At least, I wrote in a way that you can , but it may have a couple of minor issues. You need to set the limit pins all to the same pin in the cpu_map, alter the homing cycle to do one axis at a time, enable the two limits per pin option, and I think that's it.

In the ARM version, I do plan on fully expanding the limit pin configuration and homing options. I'll have the resources to do so and support it well.

109JB, 2016-10-14:
@jahnj0584 : For whatever is desired? 2 more pins means another axis, or you could add another probe input for a fixed tool setter in addition to a probe, or a second coolant output for mist or flood coolant (i have both), or ...

@chamnit : Thanks.  I understand that the space requirements are pretty much maxed out, especially in V1.1, but being open source it could be possible for a user to sacrifice some functionality in favor of other.  I am looking into this for my GUI and a new machine I m building. Will have to learn C-programming if I go down that road.

Thanks.

closed 2016-10-15 by 109JB