grbl/grbl #1109: Probe question

pitzerwm, 2016-10-16:
I made a plate and wired it in no issue.  Go to Probe tab and after positioning Z near plate, hit probe and it lowers until it touches the plate and stops.  So far so good.  Right now I'm just doing Z.

I read everything on Probe but I don't find the info for filling in any blanks or adding the thickness of the plate.  It jumps to Autolevel.

Can someone steer to where I input that?  For that matter, I could use all of the blanks that I might need on the Probe "page".

Thanks,

jahnj0584, 2016-10-16:
https://youtu.be/f4ybGiXNN24

pitzerwm, 2016-10-16:
Thanks Jahnj, I have watched that video.  I'm using Bcnc and it appears that it will automatically do all of that for you,  I just don't know what to put in the blanks to let it know the plate thickness.  Would tool offset do the trick?

chamnit, 2016-10-16:
@pitzerwm : I suggest learning about work coordinate systems and how to set them. Tool length offsets will work, but they aren't for the purpose of setting your part origin. This isn't a Grbl issue, but rather a bCNC one. So, I'll be closing this issue.

closed 2016-10-16 by chamnit

jahnj0584, 2016-10-16:
The G92 Z_ is where you put the plate thickness. I'll need a picture of
your options if you need more help.

109JB, 2016-10-16:
G92 is not a work coordinate and is misused very very often. G92 is global coordinate system offset.  If you only ever use one work offset (G54-G59), then it isn't a big deal. However, if you use multiple work offsets, such as running parts in different fixtures at the same time, applying G92 offsets all of them the same amount. This can be useful, but needs to be considered.  Many people start off by using G92 as if they are using a work offset and then later when their CNC skills get more advanced, they want to use G54-G59 to run multiple parts or operations.  At this point they have to re-learn how G92 and G54-G59 really work.  My advice would be to learn G54-G59 work offsets from the beginning.  G92 is seldom needed and in industry is seldom used.

As an example of how these might be properly used (This isn't the only right use), consider this fictitious example.  Let's say you are making overgrown dice 2 inches on a side, like used with board games. The cubes are sized already, but you have to perform 6 operations that put the dimples on each face and since you have thousands to make, you decide you can put 6 blocks in 6 fixtures and do each different face in one run.  You can use G54 for the face with one dimple, G55 for the face with 2 dimples, G56 for the face with 3, and so on. You set the 0, 0, 0 point of each of those faces to the bottom left corner using the appropriate G54-G59 offset.  That way when you program the dimples you do it all as if you are working on a single part. You just have to remember to switch G54-G59 as appropriate for each face.  The center of face 1 is then G54 X1 Y1, The center of face 2 is at G55 X1 Y1, and so on.  So programming is trivial.

Now you run the first batch of parts and for some reason you find that the dimples on all 6 faces are off   by the same amount.  Maybe you forgot to take into account the diameter of your edge finder when setting all the work offsets and everything is shifted -0.100" in x and y.  You have 2 options.  First option is to go through all 6 work offsets (G54-G59) and reset them correctly. That's 12 settings, and multiple steps to reset each one.  The second option is to use G92, which is applied globally across all work offsets. All you need to do is jog to the correct X=0, Y=0 position in ANY one of the G54-G59 work offsets and invoke G92X0Y0. It will offset all of the other work offsets the same amount and you can run the parts.

In the above example lets imaging you set everything up the same, but when you run the first batch you find that only face 1 (G54) is off, so you jog to where G54 x=0 Y=0 should be and you invoke G92X0Y0.  You have fixed face 1, but since G92 is global, you have now screwed up faces 2-6 by offsetting them as well. In this case you should only reset the G54 offset and leave the others alone, and do not use G92.

jahnj0584, 2016-10-16:
I appreciate the big example but i would cut the fixture with one
operation, then all 6 faces with another ;)

If offsets were easier to use on UGS, i think they would be used properly.
The other thing is 99% of the users are hobbyists who are cutting one thing
at a time so i understand why the time hasn't been put into them. I
personally think over rides and jogs should be fixed, then the offset widgit

chamnit, 2016-10-16:
@jahnj0584 : Please don't dismiss the importance of using G54-G59 work coordinates. If you have homing switches, it's highly recommended that you use them instead of G92, because they are saved in EEPROM. G92 is lost upon a reset or power-cycle. That said, G92 is fine, if you don't have homing switches.

Work coordinates are particularly important if you have to cancel your job with something like an e-stop, which would kill Grbl, and wipe your G92 offsets. With work coordinates, you don't lose your part origin (starting point) in this very common scenario. It's also common for your original reference point to be machined away at the beginning and you'll likely not have any edge to reference against anymore, if your trying to set back up with a G92. With homing and work coordinates, you just re-home, if needed, and restart the job.

jahnj0584, 2016-10-16:
Oh Im totally not! I thought this was in the UGS hub actually. I slowly want to use them (like, cutting 2 programs and doing centre lines. I CAN just jog over 225mm and run the engrave path again, or use offsets and then do it all in software with the programs work 0) but they arent very well implemented in UGS as it stands. I have limits in the mail (not using the printer style ones) and simply use hard limits for now. Hit the limit, reconnect the machine, and go from there. Its suprisingly accurate to well under 0.5mm.

I have ran into the g92 issue often actually, but Im just good at writing coords down before in case as well as making a mark somewhere. Is there a way I can just change my macros from G92 to G___ to use the G54 offset for everything instead?

109JB, 2016-10-16:
@jahnj0584 :  The example I gave is fictitious and is just an example to explain how the G92 and G54-G59 offsets work. It wasn't a guide on the best or most efficient way to make dice. 

If you want to keep using G92 the wrong way then I'm not going to stop you, but everyone who uses G92 and tells people to use G92 as a work offset is simply spreading the misuse. When I say wrong, I actually mean wrong from an industry standpoint.  I worked as a CNC programmer/operator for most of the 1990's. running vertical machining centers with 10 hp spindles, and tables the size of a kitchen table. In all that time I can't remember a single instance where I used G92 in actual industrial production.

As I understand it, back when GRBL was originally written, it only had G92, and didn't have G54-G59, so the only way to set any kind of offset back then was G92.  People had to use G92 because they had no other option, and this unfortunate practice has carried forward. Now we do have G54-G59, and although it doesn't affect me one iota, I am trying to get people to use offsets the way they are supposed to is all.

gerritv, 2016-10-16:
I think the GUI you choose to use must have offset configuration. AFAIK There are only a few that do this at present. When you can't see or modify these offsets then using them will remain a dark art to those who are in a hurry to cut material. They are a nuisance to set using just MDI. 
Using offsets you can create and re-use fixtures on your mill/router table, a big time save for repetitive parts.

chamnit, 2016-10-16:
@109JB : I agree, but I'd like to mention again that work coordinates are ineffective when you don't have a homing reference point. In this case, you're mostly ok using G92. 

I'd also like to add that at my day job, I work closely with two machinists who have about 70 years of combined experience. They also never use G92 because they have personally witnessed very bad  crashes when an operator resets the machine, clearing their G92 offsets, and then destroying the part, tools, and damaging their very expensive CNC machine and halting production for days to weeks. 

For hobbyists, the monetary consequences are much less, but you can never rule out personal injury caused by an easy mistake like this.

jahnj0584, 2016-10-16:
UGS currently doesnt. You have to send something like G54 L0 X_ Y_ Z_ P1 and then send the offset command in console again to see what their offsets are. Im hoping Will will be able to "copy" some of the tools from GRBLPanel which has a better cord, setting, and offset layout and function.

When I do jigs, I write their machine cords on them. What are you to do when you have more then 5 jig locations or are only ever using one? 

Again, Im all for using offsets but G92 is my only option until it gets implemented properly. Im a problem solver by nature so I've been able to deal with every downside of the 92 thus far. Yes, its really annoying trying to re-zero off of a marked location but it does work. If anything, it helps you think things through to prevent it from happening again :P

swarfer, 2016-10-16:
@jahnj0584    are you sure?   I have UGS 1.0.8 here and it has 'Reset axis' buttons on the machine control page.  when clicked they give commands such as
G10 P0 L20 Y0
which sets the work coordinates for G54 by default but if you give a manual G55..59 is will change those offsets quite happily.  (note no G92 is used)

Do you mean it has no buttons for changing from G54 to G55 etc?    Sorry, if you choose to use UGS then you are choosing to learn manual commands  (-:

109JB, 2016-10-17:
I am not sure about UGS, because I haven't tried it in quite some time, but I have tried bCNC, Chilipepper, and a few others recently and all of them had simple ways to access the G54-G59 work coordinates.  Even so, just typing 3 characters in the MDI doesn't seem too daunting to me.

109JB, 2016-10-17:
@chamnit :  I disagree with this statement:

" I'd like to mention again that work coordinates are ineffective when you don't have a homing reference point. In this case, you're mostly ok using G92."

Work coordinates are not ineffective without home/limit switches.  Lets say you are making those dice I mentioned and are using G54-G59.  For some reason, you have to reset, but since you don't have home switches are you screwed?  No.  This is a limited case where I would say G92 would be appropriately used.  Simply pick up the zero point on the G54 part, invoke G92X0Y0Z0 and all the work offsets will now be correct because G92 is applied globally across all of the G54-G59 work offsets.

I just think the widespread arbitrary use of G92 as if it were a work offset is a practice that is neither safe nor in line with industry practice.  Lets say the hobby CNC user that gets in the habit of using G92 in this way really likes the CNC stuff and gets a lerger machine and then a larger machine and before too long is doing production in his garage on a small commercial vertical machining center. There are many that have done this. If that person uses G92 as a work offset and carries this through to his VMC, disastrous results can happen.  I think it is best to start by using the best industry practices to begin with, whether you are a hobbyist or not.

My job now is no longer in machining, but in aviation, and specifically in investigating aircraft accidents. I see the results of not applying standard operating procedures on a regular basis and it is never pretty.  I aviation, and in my opinion machining, there are accepted procedures and practices for a simple reason, safety.  A private pilot is not held to the same standards as a commercial pilot, but he is also not taught things that are contrary to standard operating procedures for higher levels of certification. This is why I feel strongly that even hobbyist CNCers should, as closely as possible, use the same procedures as proven in industry.

chamnit, 2016-10-17:
@109JB : I disagree with your disagreement. :) 

I fully understand your viewpoint, as a former professional in the field. However, most hobbyists are not doing production. They are mostly one-offs involving small projects, typically made of soft, cheap materials. There is also the consideration that not all machines are subtractive milling machines, where tool forces are substantial. 3d printers, water jets, laser cutters, pen plotters, and drag knife cutters are only a few example of near-zero tool forces. In these cases, work coordinate systems are not as important. They are helpful to setup jobs but not critical to safety or preventing machine damage. 

I'm of the opinion that people SHOULD use work coordinate systems and have homing switches. That is a given. If you don't have switches or don't have an application that is as potential dangerous as milling or lathing, then you should be **mostly** ok using G92. (Note the mostly in my prior comment.) 

As for having no homing switch and work coordinates, it's not guaranteed that your machine position will be retained after a reset either. Depending on the machine or if you bumped something, you can be off from a few thousandths to a tens of thousandths. With a small cutter, this can easily mean a broken tool. It's preferable to not depend on the work coordinate systems without homing to restart a job. There are a lot of things that can go wrong with that.

pitzerwm, 2016-10-17:
You guys got off my subject, I still need to know where to enter the plate thickness in the Bcnc probe page/tab.

winder, 2016-10-17:
@pitzerwm Since it is a bCNC specific question, you should ask your question here: https://github.com/vlachoudis/bCNC/issues

109JB, 2016-10-17:
@chamnit : You said:

"As for having no homing switch and work coordinates, it's not guaranteed that your machine position will be retained after a reset either. " 

And I believe you missed the jist of my post where I said:

"This is a limited case where I would say G92 would be appropriately used. Simply pick up the zero point on the G54 part, invoke G92X0Y0Z0 and all the work offsets will now be correct because G92 is applied globally across all of the G54-G59 work offsets."

The G92 in this case would be OK in my opinion if using multiple work offsets, because  the G92 would be offsetting all of the work offsets, effectively resetting all of them at once. Since G92 is applied globally, it would "fix" all of the G54-G59 work offsets.

Whether a user knows it or not they are always using a work coordinate, system, probably G54.  If only using one work coordinate system then I see no benefit whatsoever to using G92. If using only one work coordinate system you may as well just zero the WCS rather than use G92 and avoid the non-standard use of G92 in the first place. Most GUI's use G10L2... to zero WCS by a button push anyway.

jahnj0584, 2016-10-17:
This thread was very informative. Thank you all.

Now, tell me how to change my probe macros to reset the G54 and not the G92
and we can all get along :)

chamnit, 2016-10-17:
@109JB : I think my point is that, if you have no homing switches and no reference point, it doesn't matter what you use. G92 is arguably less confusing because it resets. Hear me out on this.

For example, you start your machine without homing. Grbl starts up and initializes at 0,0,0. You jog your machine to your starting point of your job. Set the G92 to part origin. If you reset, Grbl starts over at 0,0,0. It's very clear on what Grbl is doing. You can easily command a motion from 0,0,0 with little risk of user error.

If you have some saved work coordinate offset that is non-zero, Grbl will startup and initialize again at 0,0,0 but the work coordinates will be offset to some strange, non-intuitive location that changes every time the work coordinate offsets do. It can be very inconsistent and confusing. If you try to move your machine in this state, you have to explicitly move in incremental mode or take this offset into account every time. Overall, it takes more know-how to manage a work coordinate offset without a consistent home, and introduces a lot more chance for user error. 

I think the positives of work coordinates and negatives of G92 essentially cancel out this in case. That said, I'm not disagreeing with you that it's more important for users to learn and use work coordinate systems. I think there still is a use case for G92 outside of traditional machining and when you don't have homing switches, because it resets automatically and is more intuitive.

109JB, 2016-10-17:
@jahnj0584 ;
From an MDI standpoint, you would use the G10 command in combination with the L20 and P# switches as follows:

G10 L20 P# X(value) Y(value) Z(value)  

To explain each of these switches:

L20: Tells grbl that you want to define the current position to be equal to the values denoted by the X, Y, and Z words. Any X, Y, or Z word omitted will not be reset.

P# tells GRBL which coordinate system to apply the settings to. P0 sets the currently active work coordinate system. If you want to set a coordinate system other than the current one you would use the P word as follows:

P0 = Current active work coordinate system
P1 = G54 work coordinate system
P2 = G55
P3 = G56
P4 = G57
P5 = G58
P6 = G59

I believe you mentioned that you use UGS for your GUI. From the UGS standpoint, the "Reset zero" button zeros the current position in the current work coordinate system. It is simply sending a G10L20P0X0Y0Z0 command. So if you are in G54, it will reset the G54 coordinates. I don't see a way to change work coordinate systems using a button in UGS, but in the MDI command line you can simply type the G54 - G59 command to change to that coordinate system. and then use the reset zero button if you would like.

Comparing the G10 command to G92, the equivalent would be:

G92 X(value) Y(value) Z(value)
G10 L20 P0 X(value) Y(value) Z(value)

jahnj0584, 2016-10-17:
Oh, so i can totally still use it to probe (using a plate and edge finder).
I'll give it a test next time I'm cutting