grbl/grbl #1113: Changed baud rate to 9600 from 115200 via config.h, trouble connecting

Sophi, 2016-10-18:
Hi- I want to run at 9600 instead of 115200. When I compiled the new hex, the communications via G-code sender stopped working.
Steps:
-Downloaded latest GRBL from grbl/grbl
-Edited config.h here: #define BAUD_RATE 9600
-Recompiled the file in examples->grblUpload-->grblUPload.ino
-Loaded hex into Atmega328
What am I missing?
Thanks!

chamnit, 2016-10-18:
@Sophi : That's the intended way to alter the serial baud rate, but it hasn't been 9600 baud for a long time now. So, I can't say how it'll run. Also, make sure that you are editing the config.h file inside the Arduino library. Usually in the ~/ or My Documents/Arduino/libraries folder.

Sophi, 2016-10-18:
Thanks for the response!
Yes, that file path My Documents/Arduino/libraries is where I am editing it. The issue is that it doesn't connect. Will remake and try again :/

chamnit, 2016-10-18:
@Sophi : Just did a quick check to see if 9600 baud works. It compiles and connects to a serial terminal ok. I would check if it connects to a basic serial terminal to rule out if the GUI is having issues connecting at 9600 baud.

Sophi, 2016-10-18:
I'm compiling the hex using Arduino IDE, when loading the hex with Xloader, it doesn't communicate. Testing comm with serial monitor and $$
-Getting hex via Export compiled Binary, which provides 2 files, one with bootloader (grblUpload.ino.with_bootloader.mega.hex) and one without.
When I send the file directly via Arduino IDE, the communication works. $$ provides received data.