grbl/grbl #1115: 2 wire motor + grbl = what shield to use

pajamac, 2016-10-18:
Hey guys,Im trying to make very basic 3 axis machine.I use cd rom motors.I have used h bridge to control my motors which have only 2 pins and all worked well.Then I got informed that grbl dont support h bridge.
Im looking for shield that can step controll 3x 2 pins motors.
Could I use motor driver A3967?There is problem that i have 2 wire motor and this is for 4 wired motors?
Can someone help me please ?

whitetd, 2016-10-18:
Sounds like your trying to use plain DC motors. Not suitable for cnc. Needs
stepper motors or servos to give position  control.

pajamac, 2016-10-18:
Thank you for reply :)
Well Im getting into controlling motors(Even tho I made robot that was driving around) and I dont know how to know what that motor is :( Any kind of arduino code that would work only if its stepper motor ?

whitetd, 2016-10-18:
It needs to be a stepper so it can be controlled precisely in position or
be a servo that has feedback to control position.  A DC motors just turns
continues as long as power is applied. No position control. Do some
research on stepper motors.

pajamac, 2016-10-18:
Alright now I understand that I need stepper,is there any way I can test if that motor is stepper ? I dont have cd rom number or anything and motors are clean too :(

whitetd, 2016-10-18:
Look for 4 plus wires to motor. Old floppy drives used them for head
position. Cd roms possibly.

splitn2, 2016-10-18:
a 2 wire motor won;t be a stepper sorry to tell you. If it had 4 or 6 or 8 it might be a stepper and then you need to use identifying marks and wiring colors etc and/or a multimeter to work out what is the correct wiring. This 2 wire motor sounds like a simple free running dc motor. You do need to do some googling about "stepper motor vs dc motor" which will help define the differences

https://www.modmypi.com/blog/whats-the-difference-between-dc-servo-stepper-motors is a good example

hope this helps

cheers
Richard

pajamac, 2016-10-18:
Well thats negative.Only 2 wires here.
Ok RIP me.
But why does then Stepper.h work with this kind of motor ?

pajamac, 2016-10-18:
I have this code 
`#include <Stepper.h>

const int stepsRevolution = 200;  
Stepper myStepper(stepsRevolution, 6, 3);

int stepCount = 0;  // number of steps the motor has taken

void setup() {
  Serial.begin(9600);
  // nothing to do inside the setup
}

void loop() {
  int motorSpeed = 1;
  if (motorSpeed > 0) {
    Serial.println(stepCount);
    myStepper.setSpeed(motorSpeed);
    myStepper.step(stepsRevolution / 100);
    stepCount = stepCount + 1;
  }
}`
Would this work on normal dc motors ?

whitetd, 2016-10-18:
DC motors will not work.

pajamac, 2016-10-18:
Then I have stepper motor :D

whitetd, 2016-10-18:
A stepper will only rotate a small amount when you apply power to a set of
Leads, then sequence that switching to different  set of Leads, to continue
rotating. This gives control to how far it moves. DC motors will not work
that way.

chamnit, 2016-10-18:
This issue isn't Grbl-related so I'm closing this thread. @pajamac: There are lots of online resources explaining how stepper motors work. Please take the time to read these. Thanks.

closed 2016-10-18 by chamnit