grbl/grbl #1116: Others commands while streaming G-Code

Breman33, 2016-10-21:
Hello everybody,

First of all I'm new here and I don't speak a perfect english so don't hesitate to tell me if you don't understand a part of my issue. That issue is - IMO - linked to the USB connection, but I cannot be sure at 100%. The problem is that GRBL seems to freeze at "random" positions (begin or end of a motion, never during a motion) I will explain my issue as clearly as possible : 

1) **Context:** I am an engineer apprentice and incharge of a CNC project since 2015. So I developed for 1 year a G-Code-streaming application based on Processing (Java environment). It works fine with the use of "ok" response protocol described on the GRBL home page. Meaning that no new command is sent until we receive an 'ok' from GRBL.

_**WIRING:**_ We have one Arduino MEGA (contains the GRBL), one Arduino UNO (to command 3 motors which respectively command, to resume quickly : 
- the"Blade Rise/Descent" (to Rise when a G0 is performed, and to Descent when others command)
- the "Cutting-head Rotation" (to follow the cutting route)
- the "Blade On" (in order to cut when G1/G2/G3 command)

2) **Current behavioral** The streaming application has finished 15 attempts on 15 when I'm only waiting for the 'ok' from GRBL. But when i uncomment the part when i'm waiting for a new angle (received by the Processing app from the Arduino Uno), it seems lead to random freezes (on 15 attemps, only 6 or 7 have finished). I am disposed to give you all codes you need to help me, of course, do not hesitate. I really want this project to work.

3) **Issue possible causes** I searched on Google for so many times that I am currently blocked. I am just wondering some things like : 
- The GRBL seems to freeze JUST AFTER i sent a new command. The 'ok' never comes, whatever in case of G0, G1, G2 or G3. Totally random... I know about the electromagnetic interferences, electrical noises etc... 
BUT I really don't think the issue comes from that, because of the change of behavorial when I just add the rotation feature.

- How can I 'cleanly'  read the GRBL position feedback during a file streaming?, isn't there any communication conflicts ? In fact, to add the feedback reading causes TOO the random freezes...

- I noticed that if I replace the "New angle waiting from UNO" by a delay() or an empty while{} in order to make a temporization, that causes the freezes too. 

To resume, as soon as I need to wait for something before continue streaming, the application OR the GRBL freezes. I am not sure of the real cause...

I really need to add features to my streaming program, or the GRBL solution could not be good for us.... implying that my work during the year wasn't useful :(

Please help me !

Strooom, 2016-10-21:
Is it possible that by not waiting for an 'ok' response, you overflow the receive buffer in GRBL ?
Overflowing this buffer will likely cause a crash of the sw.
Several mechanisms are explained to prevent overflow, and one of them is waiting for an 'ok' before sending a new command. But if you don't wait for an 'ok', you should apply another strategy for managing the receive buffer.

Breman33, 2016-10-21:
I understand, but I am already using the "wait for ok" protocole before sending another command.  Maybe I was not clear, sorry.

During the streaming, i can see on the Processing console (thanks to some println() that I put on the code) that a response 'ok' is received after every command, until the blocking command (random). That's especially why I can't find the cause, I think i'm doing correctly the 'ok' waiting...

The issue of freezing just happens when I add a delay between the rotation calculations (sense of rotation, number of steps, frequency of rotating) and the command sending.

But you're right about the receive buffer in GRBL. I think it does not clear itself at some "random" times... but the cause is, for now, impossible to detect...

I know that my explanations are not perfect, so I will try to develop you the algorithm of my Processing streaming : 

 WHILE (line < file.length)
{
    detectCommand(); // determines if the command is a motion (G0 G1 G2 G3) or not
    IF (G0 || G1 || G2 ||G3)
    {
       SavePoints(); // Current point, Target point and eventually Center point are saved
       computeAngle(); // Do the calculations for the transition rotation to do
       IF (G0 || G1) // Only the transition rotation need to be sent in this case
        {
                 send1stRotationSteps(); // Send the transition rotation parameters to Arduino UNO
                 waitForAngle(); // Waits the angle response from ARduino UNO
                 sendGCommand(); // Send the G0/G1 command to Arudino MEGA (GRBL)
                 waitForOK(); // Waits the 'ok' reponse from GRBL
        }
       ELSEIF (G2 || G3) // 2 rotations need to be sent : the transition and during the arc motion
        {
                 send1stRotationSteps();
                 waitForAngle(); // Waits the angle response from ARduino UNO
                 sendGCommand(); // Send the G2/G3 command to Arduino MEGA which contains GRBL
                 send2stRotationSteps(); // Send the rotation to make the blade orientation follow the arc
                 waitForOK(); // Waits the 'ok' reponse from GRBL
                 waitForAngle(); // Waits the 2nd angle response from Uno
        }
    }
    ELSE // if the command is not a motion
    {
     sendGCommand();
     waitForOK();
     }
line++
}

This is, very resumed, my streaming algorithm. I am pretty sure that I make a bad thing on USB communication, but I can't see where.

109JB, 2016-10-21:
Are you using import Arduinos that have the CH340g USB to serial converter chip?

This sounds as if it could be related to an error I found that occurs with clone Arduino boards that use the CH340g chip.  What happens is that occasionally the chip will truncate some of the data sent to GRBL. The truncated data is usually from somewhere in the middle of one line to the middle of the next line.  This results in GRBL not getting the end of line character and could result in GRBL sending an error back.

Does your streaming code have provision for responding to an error generated by GRBL to stop the motion?

You can read more about this error on the "Known bugs" page in the wiki, kinked here:

chamnit, 2016-10-21:
@Breman33 : Try enabling the ECHO config.h option. This will cause Grbl to send back the pre-parsed command (capitalized and no spaces/comments) right before it executes it. If it's not correct, then you have a communication problem. If it looks correct, then it's possible that it's a Grbl problem or a USB-serial problem like @109JB mentioned.

Breman33, 2016-10-24:
Thanks you very much for your answers, I'll try this today.

Breman33, 2016-10-24:
Okay so great news !

Because i did not succeed to reflash my GRBL program to my Mega board (because of libraries conflicts), I decided to re-install everything : Arduino IDE + GRBL and to delete all previous GRBL versions that were on my computer. After that i tried to put the last version of GRBL (0.9j) but it worked on UNO but not on my MEGA!

So finally i found the 0.9f version for Mega only, and since that change my G-Code files are entirely streamed wih success !! 
Usually it crashes after 2 or 3 attempts maximum. I just succeeded 6 attempts without any error, so i'm confident, I come back here if the problem comes back. I really want to thank you for the replies, even if you didn't directly find the cause but it helped me a lot to keep calm in front of that bug!