grbl/grbl #1117: App scaling

CreateiveRobotics, 2016-10-23:
This is a feature request, and it may be impossible or far to difficult to implement but ... I've just started using this on my touchscreen laptop. Being able to interact without using the mouse or touchpad when machining is great but the buttons are a little small. Is there any way to scale the entire app to fit an HD screen (1280x720 or 1920x1080) so that all the controls, in particular the radio buttons, are slightly larger ... ?

gerritv, 2016-10-23:
You will need to post the request on the site for your GUI. Grbl itself does not provide a ui.

CreateiveRobotics, 2016-10-23:
LoL - My fault for having GRBL and GRBL Panel open in my browser at the same time ...

closed 2016-10-23 by CreateiveRobotics