grbl/grbl #1119: Homing issues

matthewhertel, 2016-10-24:
Hi Sonny,

I am having issues with my homing cycle.  When I try and home the machine Z travels negative at the homing feed rate as if it is trying to back off a limit switch.   I am using hall sensors with a 1k pull up ( when the hall is triggered, the drops low, I have checked the signal at Arduino and it is working correctly).  My settings are as follows,
#define DISABLE_LIMIT_PIN_PULL_UP
$5=0
$22=1
$23=1 ( x homes to the - position)
$24=25
$25=100
$26=25
$27=1

If I change setting $5 to 1, Z will travel onto and past the limit switch (causing it to drop low) 

Thanks for the help!
-Matt

109JB, 2016-10-25:
Set $5=1

matthewhertel, 2016-10-25:
Hi 109JB,

I did try that, the $5=1 setting causes z to travel onto the limit switch but when the limit switch drops low, the machine keeps going.

Thanks,
-Matt

Encarnacao, 2016-10-25:
dont think u need to enable the #define DISABLE_LIMIT_PIN_PULL_UP, you can keep the internal pull ups along with the external pullup and use the $5=1 for NC switch

matthewhertel, 2016-10-25:
The problem is the hall sensors are wired to 3.3 and the internal pull ups are for 5v.  I will get a reverse voltage to 3.3 if I enable the pull ups.  

I did have this working... but I managed to cook the arduino I was using  and now I cannot see what settings I used (poor documentation) 

Thanks for helping,
-Matt

Encarnacao, 2016-10-25:
are you using command $H to go home?

matthewhertel, 2016-10-25:
I am using $H to home the machine.

lmmeng, 2016-10-28:
If I right understood, your machine detects a "switch press" and doesn't detect a "switch release".
Maybe it is a problem with the input levels?
Shouldn't be (in the data sheet the "high thresholds" are ~2.6 V), but you can try to pull the hall sensor's output to 5 V instead of 3.3 V (remove the 1k resistor and activate the internal pull-ups).
As long as you need a pull-up on the hall sensor's output, I suppose it is an open collector/open drain type and pulling up to 5V instead of 3.3 V shouldn't cause any problem.