grbl/grbl #1120: Machine coordinates location?

RandyG-G, 2016-10-26:
I have a Nomad 883 "Classic" (original version) that homed positive in Z, Y and X.  The tool sensor is at the left side of the machine, and I moved the X home switch to the left side and set $23=1 to make X home negative, and end up near the tool sensor.

What I don't understand is, the machine coordinates did not change.  Machine coordinates of the tool sensor, tool change position and table center are the same as when the machine homed in +X.

How are the machine coordinates being defined, where are they stored?  HOMING_FORCE_ORIGIN is not set, so somehow the controller is remembering where the machine origin is.

The Omron home switches in the Nomad are very accurate.  I can power down overnight with a workpiece in place, power up and home the next day, and the machine is within a thousandth or so of where it was the night before.  I don't need to re-indicate the object.

But I am curious as to where machine origin is being stored, becuase it isn't being set at the home switches.

Thanks,

Randy

chamnit, 2016-10-26:
@RandyG-G : Grbl's machine origin is placed in the top-right-back corner of the machine. THat's where the Nomad moves to by default. When you altered the direction of homing to a different switch on the opposite side of axis travel, Grbl takes that into account to keep the origin in that location. It uses the max travel settings to do so. 

It's more than likely that your machine origin isn't exactly the same. Though, it's probably close. It depends on how accurate your max travel setting is to the actual distance between the two limit switches.

closed 2016-10-26 by chamnit

chamnit, 2016-10-26:
FWIW, I have an original Nomad too. I didn't move my limit switch or modify my machine, but I remember doing something to speed up the homing and probing cycles. There's a lot of wasted motions with the probe on the opposite side of the machine (they moved it to the right place in the Nomad Pro). I can't remember exactly what I did, but I think I altered a couple of their stored EEPROM parameters to keep it from moving all the way back to home. 

Be careful here, because the Nomad stores the probe machine location and other important offsets in the G55-G59 and G28/30 parameters. I think there are just two locations. One is the probe and the other is the "move to after homing or probe". I altered the latter to make the extra motions tolerable. 

**Just make sure you write them down before you alter any of them.** Otherwise you can make your machine incompatible with Carbide's software. I'm also not guaranteeing that it'll work at all for you. So don't do anything until you are certain of what you are doing and know how to change it back.

RandyG-G, 2016-10-26:
Hi Sonny,

Thank you for the quick reply.  I've reflashed the Nomad with grbl 0.9j and
I've set up bCNC for the Nomad, but have only done homing and tool
measurement so far but not run a gcode yet.  I'm intrigued with the
possibility of using the Shuttle Express with bCNC--I've used one on my
Mach3 Tormach for years and also use one with my Mach3 lathe.

Randy