grbl/grbl #1121: unsupported statement G40, G41, G42, G43

Martinlipo, 2016-10-27:
Hello my name is Martin Vargas and work on cnc machines three years ago.
I updated my cnc machine with shield v3. It seems to work well.
The problem is that I use Solidcam2016 (on Solidworks) to generate the G-Code and apparently GRBL not recognize many of the codes as G40, G42, G43, and others. This causes problems on my machine.
Someone can help me solve these problems?

This is the photo of the part:
[image]

IIn this picture you can see the displacement out of the line;
[image]

Here I leave a code generated by solidcam2016,

%
O1000 (PIEZA1)
N100 (COMPENSATION-WEAR)
N102 (REV-0.70)
N104 (OCT-27-2016-4:23:59PM)

N106 (TOOL 1 - DIA 3.)

N1 G90 G17 G40 G80 G00 
N108 M06 T1 ()
N110 (F-contour)
N112 G00 G54 G90 X30.749 Y-32.5 S3500 M03 
N114 G43 H1 Z120. 
N116 Z25. 
N118 Z2. 
N120 G01 Z-0.5 F50. 
N122 G41 D1 X31. F1500. 
N124 G03 X25. Y-26.5 I-6. J0. 
N126 G01 X-25. 
N128 G02 X-26.5 Y-25. I0. J1.5 
N130 G01 Y25. 
N132 G02 X-25. Y26.5 I1.5 J0. 
N134 G01 X25. 
N136 G02 X26.5 Y25. I0. J-1.5 
N138 G01 Y-25. 
N140 G02 X25. Y-26.5 I-1.5 J0. 
N142 G03 X19. Y-32.5 I0. J-6. 
N144 G01 G40 X19.251 
N146 Z-1. F50. 
N148 G42 D1 X19. F1500. 
N150 G02 X25. Y-26.5 I6. J0. 
N152 G03 X26.5 Y-25. I0. J1.5 
N154 G01 Y25. 
N156 G03 X25. Y26.5 I-1.5 J0. 
N158 G01 X-25. 
N160 G03 X-26.5 Y25. I0. J-1.5 
N162 G01 Y-25. 
N164 G03 X-25. Y-26.5 I1.5 J0. 
N166 G01 X25. 
N168 G02 X31. Y-32.5 I0. J-6. 
N170 G01 G40 X30.749 
N172 Z-1.5 F50. 
N174 G41 D1 X31. F1500. 
N176 G03 X25. Y-26.5 I-6. J0. 
N178 G01 X-25. 
N180 G02 X-26.5 Y-25. I0. J1.5 
N182 G01 Y25. 
N184 G02 X-25. Y26.5 I1.5 J0. 
N186 G01 X25. 
N188 G02 X26.5 Y25. I0. J-1.5 
N190 G01 Y-25. 
N192 G02 X25. Y-26.5 I-1.5 J0. 
N194 G03 X19. Y-32.5 I0. J-6. 
N196 G01 G40 X19.251 
N198 Z-2. F50. 
N200 G42 D1 X19. F1500. 
N202 G02 X25. Y-26.5 I6. J0. 
N204 G03 X26.5 Y-25. I0. J1.5 
N206 G01 Y25. 
N208 G03 X25. Y26.5 I-1.5 J0. 
N210 G01 X-25. 
N212 G03 X-26.5 Y25. I0. J-1.5 
N214 G01 Y-25. 
N216 G03 X-25. Y-26.5 I1.5 J0. 
N218 G01 X25. 
N220 G02 X31. Y-32.5 I0. J-6. 
N222 G01 G40 X30.749 
N224 Z-2.5 F50. 
N226 G41 D1 X31. F1500. 
N228 G03 X25. Y-26.5 I-6. J0. 
N230 G01 X-25. 
N232 G02 X-26.5 Y-25. I0. J1.5 
N234 G01 Y25. 
N236 G02 X-25. Y26.5 I1.5 J0. 
N238 G01 X25. 
N240 G02 X26.5 Y25. I0. J-1.5 
N242 G01 Y-25. 
N244 G02 X25. Y-26.5 I-1.5 J0. 
N246 G03 X19. Y-32.5 I0. J-6. 
N248 G01 G40 X19.251 
N250 Z-3. F50. 
N252 G42 D1 X19. F1500. 
N254 G02 X25. Y-26.5 I6. J0. 
N256 G03 X26.5 Y-25. I0. J1.5 
N258 G01 Y25. 
N260 G03 X25. Y26.5 I-1.5 J0. 
N262 G01 X-25. 
N264 G03 X-26.5 Y25. I0. J-1.5 
N266 G01 Y-25. 
N268 G03 X-25. Y-26.5 I1.5 J0. 
N270 G01 X25. 
N272 G02 X31. Y-32.5 I0. J-6. 
N274 G01 G40 X30.749 
N276 Z-3.5 F50. 
N278 G41 D1 X31. F1500. 
N280 G03 X25. Y-26.5 I-6. J0. 
N282 G01 X-25. 
N284 G02 X-26.5 Y-25. I0. J1.5 
N286 G01 Y25. 
N288 G02 X-25. Y26.5 I1.5 J0. 
N290 G01 X25. 
N292 G02 X26.5 Y25. I0. J-1.5 
N294 G01 Y-25. 
N296 G02 X25. Y-26.5 I-1.5 J0. 
N298 G03 X19. Y-32.5 I0. J-6. 
N300 G01 G40 X19.251 
N302 Z-4. F50. 
N304 G42 D1 X19. F1500. 
N306 G02 X25. Y-26.5 I6. J0. 
N308 G03 X26.5 Y-25. I0. J1.5 
N310 G01 Y25. 
N312 G03 X25. Y26.5 I-1.5 J0. 
N314 G01 X-25. 
N316 G03 X-26.5 Y25. I0. J-1.5 
N318 G01 Y-25. 
N320 G03 X-25. Y-26.5 I1.5 J0. 
N322 G01 X25. 
N324 G02 X31. Y-32.5 I0. J-6. 
N326 G01 G40 X30.749 
N328 G00 Z25. 
N330 M05 
N332 G00 G28 G91 Z0 
N334 G00 G28 G91 X-15.0 Y0. 
N336 G90 
N338 M06 T1 
N340 M30 
%

If I use the normal driver with MACH3 the part finish okey.

chamnit, 2016-10-27:
@Martinlipo : Please see the [list of supported g-code commands](https://github.com/grbl/grbl/wiki). It's also on the bottom of the main page. You'll need to remove the unsupported g-codes from your programs. This can be done either manually with a text editor or automatically, if you configure your CAM post-processor to export to Grbl. Most popular CAM programs have a Grbl post. You may want to look into it.

closed 2016-10-27 by chamnit

jahnj0584, 2016-10-27:
Just use the GRBL post instead. Where did you get solidcam?

Martinlipo, 2016-10-27:
Thanks Chamnit, Now I can see the post processor here, but I don´t know how to setup because the extension is .GPP and the Grbl library doesn´t have one. 
I´m reading about it. If you know something more, please help me. 
I will fix this, and then I will keep this post open to help other people.

I find an Article. 
[Solidcam Gpptool%281%29.pdf](https://github.com/grbl/grbl/files/557196/Solidcam.Gpptool.281.29.pdf)

this is the post processor files directory for solidcam:
[image]

Martinlipo, 2016-10-27:
Hi jahnj0584, I got the hardware keys from my work here in Argentina.