grbl/grbl #1122: time / feed rate ?

DTLamb, 2016-11-01:
I have noticed that grbl is moving faster than requested - !?

Example: 
X axis is at X=-80
Command to move to position X=-81 (1mm of travel) at a feed rate of 1mm/min:
G90 G21 G01 X-81 F1
Move only take ~54 seconds to complete instead of 60 seconds. ?

I first noticed this for a longer move of 42mm at F1 that was completing several minutes ahead of the instrument that was riding on the x-axis.  Anything that can be done to tighten up the timing?  What parameter/setting am I missing? 

Thanks,

Don

Grbl 0.9i
Arduino Uno
G-shield

$0=10 (step pulse, usec)
$1=255 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=0 (dir port invert mask:00000000)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.020 (junction deviation, mm)
$12=0.001 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=0 (hard limits, bool)
$22=1 (homing cycle, bool)
$23=3 (homing dir invert mask:00000011)
$24=25.000 (homing feed, mm/min)
$25=500.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=1600.000 (x, step/mm)
$101=800.000 (y, step/mm)
$102=20.000 (z, step/mm)
$110=1000.000 (x max rate, mm/min)
$111=1000.000 (y max rate, mm/min)
$112=500.000 (z max rate, mm/min)
$120=50.000 (x accel, mm/sec^2)
$121=50.000 (y accel, mm/sec^2)
$122=10.000 (z accel, mm/sec^2)
$130=91.000 (x max travel, mm)
$131=56.000 (y max travel, mm)
$132=200.000 (z max travel, mm)

chamnit, 2016-11-01:
@DTLamb : The 16-bit timer on the Arduino isn't the most accurate, especially when moving slowly. Grbl v0.9's step smoothing algorithm trades reducing multi-axis vibrations and noise with a little bit of timing accuracy and range. For 99% of expected applications, that is ok, but it sounds like you have a specific application that needs to move slowly and very accurately.

One thing you could try is disabling the step smoothing by commenting out ADAPTIVE_MULTI_AXIS_STEP_SMOOTHING in the config.h file. Re-compile and flash. This will cause Grbl to use the full range of the 16-bit timer. It may help with improving slow motion timing accuracy. Not sure though.

In addition, Grbl can move slower than 1.0mm/min, maybe down to 0.1mm/min, but it depends on the settings whether or not the planner math becomes unstable when set below 1.0. To change it, you'll can alter the MINIMUM_FEED_RATE value in config.h.

DTLamb, 2016-11-02:
By my stopwatch the 1.0 mm/min move takes 1 minute to complete a 1 mm move after commenting out ADAPTIVE_MULTI_AXIS_STEP_SMOOTHING in the config.h file.

MINIMUM_FEED_RATE value set to 0.1 - a 0.1 mm/min move of 0.1 mm is running fast, ~42 seconds to complete.  

For now I hope to stay at 1 mm/min or faster.

Thanks!

chamnit, 2016-11-02:
@DTLamb: Great! Glad to hear it helped. I'm not too surprised that 0.1mm/min has about the same error as the 1.0 mm/min with the step smoothing enabled. But it's good to know. Thanks for the update.

closed 2016-11-02 by chamnit