grbl/grbl #1123: X steps being sent to both X and Y driver, same with Y

kittenkong, 2016-11-02:
I've got an Arduino Uno with an Easydriver and a little 28BYJ-48 motor. When step and dir are plugged into pins 2 & 5 (x-axis) I get beautiful movement when I send gcode for x movement, but I also get movement when I send gcode for y movement. 

I'm getting the same problem when I plug into pins 3 & 6 (y-axis) but 4 & 7 (z-axis) works fine and only responds to z axis commands.

chamnit, 2016-11-02:
@kittenkong : It sounds like you are having electrical noise problems. You probably have a bad ground. Make sure you "star" ground everything. Your stepper drivers, power supply, Arduino, etc.

kittenkong, 2016-11-02:
[This](http://nuclearfirecracker.com/files/plotter.jpg) is my setup at the moment, I've pretty much just copied the wiki. Could it be the breadboard? Should I be soldering the wires directly to get a good result? You'll note I'm using a weird Arduino Uno in the image but I've tried it with a few different types with the same result.

109JB, 2016-11-02:
Looks to me like the step wire is plugged into pin number 1

kittenkong, 2016-11-03:
Hey it does look like that, it's in pin 2 though and I get the same issue when I plug it into pins 3 and 6 instead, movement from both x and y movement commands. The only time I got it to only respond to commands for just one axis was when I had in in 4 and 7, it loves being the z axis.

vMeph, 2016-11-03:
arent you missing to connect also the enable pin for the motors ?

kittenkong, 2016-11-03:
What should I have done with the enable pin? My problem doesn't seem to be that the motor isn't enabled, but that the driver is being sent a signal when it shouldn't be.

I've followed the wiring diagram in the wiki https://dl.dropboxusercontent.com/u/54312401/grbl/GRBL%20EasyDriver_big.png

electrokean, 2016-11-03:
It sounds almost like you're running in CoreXY mode.
What version of grbl are you running?  Where did you get it, and how did you load it?
Did you change any of the default configuration during compile or after upload?

kittenkong, 2016-11-03:
FIXED!

You are a champion and I definitely should have tried getting the latest standard version first. I was using the grbl-servo version and it must be set to CoreXY mode by default or something.

closed 2016-11-03 by kittenkong