grbl/grbl #1124: Limit switch issue

kyjak, 2016-11-03:
I got setup the limit switches as normally closed NC, 
the CPU map definition is below. My problem is when looking on verbose output,
I get clear the limit switch flags as 000, when I push a switch, it validly goes to "1".
However, when running gcode, pushing a switch does not trigger any action.
It was supposed to pause/break execution, but nothing happens, any idea if this is 
a bug in the code?

#define LIMIT_DDR       DDRC
#define LIMIT_PIN       PINC
#define LIMIT_PORT      PORTC
#define X_LIMIT_BIT     6  // mega Digital Pin 31
#define Y_LIMIT_BIT     5  // mega Digital Pin 32
#define Z_LIMIT_BIT     4  // mega Digital Pin 33
#define LIMIT_INT       PCIE0  // Pin change interrupt enable pin
#define LIMIT_INT_vect  PCINT0_vect 
#define LIMIT_PCMSK     PCMSK0 // Pin change interrupt register
#define LIMIT_MASK ((1<<X_LIMIT_BIT)|(1<<Y_LIMIT_BIT)|(1<<Z_LIMIT_BIT)) // All limit bits

electrokean, 2016-11-03:
What do you want it to do?  Have you configured Hard Limits?
See $21 here in the docs https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.9

kyjak, 2016-11-03:
Yes, hard limits are ON, switch gets triggered, however nothing happens when the limit is hit, no execution interruption.

kyjak, 2016-11-03:
Updating with more info - the alarm goes off while homing. However, is this not supposed to work during any operation not to harm the machine?

electrokean, 2016-11-03:
Maybe because you're using NC switches.  I've never tried them.
So I'm not sure if the invert flags work as you may expect with hard limits.
Can you test with NO switches?

vMeph, 2016-11-03:
should work i with NC switches my settings are
$5=1
$21 = 1    
$22 = 1 
i have put from the switch addicional pullup 1k to IOREF
i would guest you have wired correctly the switchs in series and not in parallel

kyjak, 2016-11-04:
I have the switches wired in series GND->SW1->SW2->Limit PIN.
They work, but just for homing, I'd like to have the alarm go off also during any XYZ travel, any idea if it is my fault, or grbl does not have the feature?

vMeph, 2016-11-04:
i have the external pull up with out having removed the internal pull up and a 47 uf cap between switch and gnd, i was having noise issues that caused false alarms and that have solved my problem(didnt disable the config.h)
but for what you describe works they trigger during home $H, just not working during normal XYZ travel mode,kind of weird,grbl have that feature,must be something wrong,

kyjak, 2016-11-04:
vMeph: but your switches are NO,right? I think it's better with the noise issues if you go with NC. Not sure how to debug the issue, would like to help/fix it also for others.

vMeph, 2016-11-04:
My switchs are NC

kyjak, 2016-11-06:
Can you show a sketch of the connection wiring for comparison?

vMeph, 2016-11-06:
i got like this
[image]

kyjak, 2016-11-11:
I did your setup, it works the very same way as mine. I uploaded a video of the problem. As you will see the limit switch will get triggered to "1", but machine moving does not stop, no alert is created.
https://youtu.be/apzkxUgY7uY

vMeph, 2016-11-11:
not expert would be nice some more people to input but for what i see your defenitions dont looks correct, i was looking on the new version of grbl and the defenitions you have there looks diferent
this is the ones on  from grbl 1.1 for mega
` #define LIMIT_DDR       DDRB
  #define LIMIT_PORT      PORTB
  #define LIMIT_PIN       PINB
  #define X_LIMIT_BIT     4 // MEGA2560 Digital Pin 10
  #define Y_LIMIT_BIT     5 // MEGA2560 Digital Pin 11
  #define Z_LIMIT_BIT     6 // MEGA2560 Digital Pin 12
  #define LIMIT_INT       PCIE0  // Pin change interrupt enable pin
  #define LIMIT_INT_vect  PCINT0_vect 
  #define LIMIT_PCMSK     PCMSK0 // Pin change interrupt register
  #define LIMIT_MASK ((1<<X_LIMIT_BIT)|(1<<Y_LIMIT_BIT)|(1<<Z_LIMIT_BIT)) // All limit bits

it looks like you have define 
#define LIMIT_DDR DDRC
#define LIMIT_PIN PINC
#define LIMIT_PORT PORTC
and should be like above
the link where i got info is here
https://github.com/gnea/grbl-Mega/

kyjak, 2016-11-11:
I changed the port mappings so I can use the RAMPS board on top 2560. Pins/ports set according to 2560 specs. Again, the switch gets triggered as you see on video. Only alarm does not go off.

chamnit, 2016-11-11:
@kyjak : I think it would be good to share all modifications you did, otherwise it'd be impossible to help you. The pin interrupt assignment can be tricky if not done exactly right. 

Just for reference, the Mega2560 compile option on v0.9 is purely user-supported. There may be bugs when altering things from the default build. I will provide only limited support the the v1.0+ Mega versions until Grbl v1.1 has gone to master and I have ported the new stuff into the Mega branch.

kyjak, 2016-11-11:
No problem, http://brana.sk/grbl.zip 
I checked out latest GRBL, modified cpu_map.h, added cpu_map for ramps modification. Dont have a repo server, otherwise I would share by git.

chamnit, 2016-11-11:
@kyjak : I took a quick look. You may want to check if your PORTE pins have a pin change interrupt or not. It doesn't look like it, but then again, I just glanced at the data sheet.

kyjak, 2016-11-11:
well, from the specs [not a deep expert], I assumed the "/INTx" indicates an interrupt availability. or?
5	PE3 ( OC3A/AIN1 )	Digital pin 5 (PWM)
6	PE4 ( OC3B/INT4 )	Digital pin 2 (PWM)
7	PE5 ( OC3C/INT5 )	Digital pin 3 (PWM)

109JB, 2016-11-11:
[image]

chamnit, 2016-11-11:
See datasheet excerpt. No pin change interrupts of all pins, except pin 0, on PORT E.

[image]

vMeph, 2016-11-11:
@109JB  reason for not having that cap there?
what wiring you would sugest for NC switches?

109JB, 2016-11-11:
Not needed.

http://reprap.org/wiki/Mechanical_Endstop

vMeph, 2016-11-11:
i would guess wouldnt be need it but sense i had some noise issues in past, i end up making a choise to have a cap there to avoid any flotations, so far never had no false triggers again
but i should try with out it some day sense when i has noise issues i use to have NO switches

kyjak, 2016-11-13:
Update of the problem - I changed to PORT K for the limit switches, using A13,A14,A15, this time checked the datasheet, all are enabled for pin change interrupt. Good news - X limit works as expected, Y,Z work, but without the interrupt. The pin change occurs, but no interruption. Think I still am missing something else ...

// Define step pulse output pins. NOTE: All step bit pins must be on the same port.
#define STEP_DDR      DDRA
#define STEP_PORT     PORTA
#define STEP_PIN      PINA
#define X_STEP_BIT    2 // MEGA2560 Digital Pin 24
#define Y_STEP_BIT    4 // MEGA2560 Digital Pin 26
#define Z_STEP_BIT    6 // MEGA2560 Digital Pin 28
#define STEP_MASK ((1<<X_STEP_BIT)|(1<<Y_STEP_BIT)|(1<<Z_STEP_BIT)) // All step bits

// Define step direction output pins. NOTE: All direction pins must be on the same port.
#define DIRECTION_DDR     DDRC
#define DIRECTION_PORT    PORTC
#define DIRECTION_PIN     PINC
#define X_DIRECTION_BIT   7 // MEGA2560 Digital Pin 30
#define Y_DIRECTION_BIT   1 // MEGA2560 Digital Pin 36
#define Z_DIRECTION_BIT   3 // MEGA2560 Digital Pin 34
#define DIRECTION_MASK ((1<<X_DIRECTION_BIT)|(1<<Y_DIRECTION_BIT)|(1<<Z_DIRECTION_BIT)) // All direction bits

// Define stepper driver enable/disable output pin.
#define STEPPERS_DISABLE_DDR   DDRB
#define STEPPERS_DISABLE_PORT  PORTB
#define STEPPERS_DISABLE_BIT   7 // MEGA2560 Digital Pin 13
#define STEPPERS_DISABLE_MASK (1<<STEPPERS_DISABLE_BIT)

// Define homing/hard limit switch input pins and limit interrupt vectors. 
// NOTE: All limit bit pins must be on the same port
#define LIMIT_DDR       DDRK
#define LIMIT_PORT      PORTK
#define LIMIT_PIN       PINK
#define X_LIMIT_BIT     5 // MEGA2560 Digital Pin 3
#define Y_LIMIT_BIT     6 // MEGA2560 Digital Pin 2
#define Z_LIMIT_BIT     7 // MEGA2560 Digital Pin 5
#define LIMIT_INT       PCIE0  // Pin change interrupt enable pin
#define LIMIT_INT_vect  PCINT0_vect 
#define LIMIT_PCMSK     PCMSK0 // Pin change interrupt register
#define LIMIT_MASK ((1<<X_LIMIT_BIT)|(1<<Y_LIMIT_BIT)|(1<<Z_LIMIT_BIT)) // All limit bits

// Define spindle enable and spindle direction output pins.
#define SPINDLE_ENABLE_DDR      DDRH
#define SPINDLE_ENABLE_PORT     PORTH
#define SPINDLE_ENABLE_BIT      3 // MEGA2560 Digital Pin 6
#define SPINDLE_DIRECTION_DDR   DDRE
#define SPINDLE_DIRECTION_PORT  PORTE
#define SPINDLE_DIRECTION_BIT   3 // MEGA2560 Digital Pin 5

// Define flood and mist coolant enable output pins.
// NOTE: Uno analog pins 4 and 5 are reserved for an i2c interface, and may be installed at
// a later date if flash and memory space allows.
#define COOLANT_FLOOD_DDR     DDRH
#define COOLANT_FLOOD_PORT    PORTH
#define COOLANT_FLOOD_BIT     5 // MEGA2560 Digital Pin 8
#ifdef ENABLE_M7 // Mist coolant disabled by default. See config.h to enable/disable.
#define COOLANT_MIST_DDR    DDRH
#define COOLANT_MIST_PORT   PORTH
#define COOLANT_MIST_BIT    6 // MEGA2560 Digital Pin 9
#endif  

// Define user-control CONTROLs (cycle start, reset, feed hold) input pins.
// NOTE: All CONTROLs pins must be on the same port and not on a port with other input pins (limits).
#define CONTROL_DDR       DDRB
#define CONTROL_PIN       PINB
#define CONTROL_PORT      PORTB
#define RESET_BIT         0  // MEGA2560 D53
#define FEED_HOLD_BIT     1  // MEGA2560 D52
#define CYCLE_START_BIT   2  // MEGA2560 D51
#define SAFETY_DOOR_BIT   3  // MEGA2560 D50
#define CONTROL_INT       PCIE2  // Pin change interrupt enable pin
#define CONTROL_INT_vect  PCINT2_vect
#define CONTROL_PCMSK     PCMSK2 // Pin change interrupt register
#define CONTROL_MASK ((1<<RESET_BIT)|(1<<FEED_HOLD_BIT)|(1<<CYCLE_START_BIT)|(1<<SAFETY_DOOR_BIT))
#define CONTROL_INVERT_MASK 0 // May be re-defined to only invert certain control pins.

// Define probe switch input pin.
#define PROBE_DDR       DDRJ
#define PROBE_PIN       PINJ
#define PROBE_PORT      PORTJ
#define PROBE_BIT       3  // MEGA2560 D15
#define PROBE_MASK      (1<<PROBE_BIT)

// Start of PWM & Stepper Enabled Spindle
#ifdef VARIABLE_SPINDLE
  // Advanced Configuration Below You should not need to touch these variables
  // Set Timer up to use TIMER4B which is attached to Digital Pin 7
  #define PWM_MAX_VALUE       65535.0
  #define TCCRA_REGISTER		TCCR4A
  #define TCCRB_REGISTER		TCCR4B
  #define OCR_REGISTER		OCR4B
  
  #define COMB_BIT			COM4B1
  #define WAVE0_REGISTER		WGM40
  #define WAVE1_REGISTER		WGM41
  #define WAVE2_REGISTER		WGM42
  #define WAVE3_REGISTER		WGM43
  
  #define SPINDLE_PWM_DDR		DDRH
  #define SPINDLE_PWM_PORT    PORTH
  #define SPINDLE_PWM_BIT		4 // MEGA2560 Digital Pin 97
#endif // End of VARIABLE_SPINDLE

closed 2016-11-13 by chamnit