grbl/grbl #1125: LCD on grbl

vMeph, 2016-11-03:
@chamnit i hope you dont mind i have pulled  this post, that isnt a issue,but i tought i would share with those interested in having a LCD on there machine by using a second arduino.

This have been done by this person on shapeoko forum  (jpbbricole)
the link for the project is:

http://www.shapeoko.com/wiki/index.php/LCD_on_GRBL#Full_version_GRBL_1.1

it is done by having a second arduino to recive the data from the GRBL arduino and it have the versions for GRBL 0.9 and also have been updated the version to work with GRBL1.1

works very good the version of grbl0.9 shows machine and work positions at same time, the new version for grbl 1.1 can only show 1 at time  or machine position or work position is done by seeting $10=0 or $10=1

chamnit, 2016-11-03:
@vMeph : That's great! I've been waiting a long time for someone to do this. I'm glad to see that they were able to update it so quickly for v1.1. 

Although the statement that machine position or work position given one at a time in v1.1 is not quite correct. Grbl v1.1 does indeed report only one or the other, but all the GUI has to do is retain the last given `WCO:` work coordinate offset to compute whatever is missing with `WPos = MPos - WCO`. `WCO` doesn't change often. Usually only once prior to setting up a job to a handful of times, if you have tool changes. So, the GUI only needs to monitor for and retain changes with WCO to compute work position, if MPos is only given, or machine position, if WPos is given.

This isn't clearly spelled out in the interface document, so I"ll update it to make sure GUI devs don't mess this up.

tklus, 2016-11-03:
Now we just need to use the LCD display 3D printers use so we can load the gcode via SD card and have a stand alone machine :)

Like this one
https://reprapchampion.com/products/full-graphic-lcd-display-controller-for-ramps-1-4-reprap-3d-printer-prusa-kossel?utm_medium=cpc&utm_source=googlepla&variant=7797778243&gclid=CjwKEAjwwOvABRC08aedoZ_lnTMSJACs_cbuixXZnbkku1oX4YrKgStCn_IRE_Dip3nPmKtRftZTbxoCtiHw_wcB

Great work!

vMeph, 2016-11-03:
i have invited jpbbricole the mate that did the lcd sketch to come here to discuss also, according to what i have spoke with him last time he saied he was going to try to implement sd card also, lets see what happens :)
if you guys see the schematic he got on the wiki he is using  Ic CD4066, but you can use  it out with out the CD4066, just leave the TX of the second arduino disconected, according to jpbbricole the GRBL Rxd signal is crushed by the txD of Mega.
t

jpbbricole, 2016-11-03:
Hello GRBL community.
It's been long since I worked with your great product, Thank you for your work. You must beings indulgent with my English, I'm French Swiss.
Here's what I've done on the basis of this display system (in French):
[Lathe transformation to digital](url=http://www.usinages.com/threads/numerisation-tour-proxxon-pd-230e.90773/).
[Customization of a mini CNC](url=http://www.usinages.com/threads/une-petite-cnc-qui-obeit-au-doigt-et-a-loeil.94573/).

If you have questions about the LCD display, I am at your disposal.

Best regards
jpbbricole

jpbbricole, 2016-11-03:
@chamnit 
I did not understand the exact meaning of WCO. I will correct the program to offer MPOS and WPOS display.

jpbbricole

109JB, 2016-11-04:
I had done this a while ago with a second UNO and a LCD shield.  I was able to implement enough features to be able to Jog, zero work offsets, probe, and run files from a SD card. One problem I had was that the LCD updates slowed the whole thing a little too much for my taste. If I refreshed the work coordinates more than about once every 0.5 seconds, the LCD took too many resources and other tasks, such as the call response protocol were slowed. I was using the LCD libraries supplied by the manufacturer of the LCD, and I researched some stuff and those libraries are apparently known for being kind of slow, so optimizing the libraries is probably needed. I worked around this by changing the LCD update rate if a job was running. I actually ran several test parts, and it worked, but I ultimately decided that for me an old netbook laptop I was given was more useful.  I can see if I still have my Arduino code for the host.  It was wired directly to the TX/RX pins bypassing the CH340G, making the Nano I had usable for the grbl firmware.

jpbbricole, 2016-11-04:
@109JB 
I think all this is too much for a UNO. That's why I took a Mega (under $ 6), which has several hardware communication ports. Does the display is I2c ?. I also tried the PC, with a Excel sheet based DRO.
https://youtu.be/sHiP3trSOs4
The advantage of this solution is a very fun display and the ability to use UGS to send G-code files.

Have a nice day
jpbbricole

109JB, 2016-11-04:
You know what, I misspoke. I did use a mega for mine, but the LCD was not i2c. Glad you got it working.  I may at some time revisit mine, but for now I have too many projects.

jpbbricole, 2016-12-02:
Hello GRBL!
GRBL11universalDRO for GRBL 1.1 is now available. Have a look on the [shapeoko forum](http://www.shapeoko.com/forum/viewtopic.php?f=11&t=1904&p=65833#p65833).

GRBL is great, thanks to you.
jpbbricole

Fosforitofercho, 2016-12-27:
So good, only an arduino that decodes the  serial, is a simple by funtional idea, how cute looks a machine whit own LCD , can use a arduino nano and make an slim version .

vMeph, 2017-01-23:
@chamnit  i dont know if would be a good idea but sense grbl 1.1 using arduino uno is basikly out of space  and LCD on the same uno with grbl is out off option, maybe would be a good idea have on wiki a section for how to adapt a LCD on grbl by using a 2th arduino uno, maybe then @jpbbricole could make a section explaining how it works and how to get it to work, i have been using this option for a long time and its great adiction, and would be great to have it on wiki for those interest in it.

is just a idea...
thanks

tklus, 2017-01-24:
That sounds awesome! Can the second arduino also have an sd card or read programs from USB memory stick?

RunnyBore604, 2017-07-19:
Hi,

I made a GRBL version that can work without a PC, it's for lasers only. It requires LCD 1602, SD card module and only one Arduino Nano (not two, like this do others).

https://github.com/RunnyBore604/standaloneGRBL

closed 2017-09-24 by vMeph

luisluna2, 2018-09-27:
@vMeph hello friend, I have interesting in your project but now the pages not exist, can share with my the Master Files please? my Email is andres.campa_luis@hotmail.com

Thank you!

jpbbricole, 2018-09-27:
Hi Andres

 

I think shapeoko.com disappeared!

I put the file as an attachment and remains at your disposal for more information.

 

Have a nice day

Jean-Pierre

luisluna2, 2018-09-27:
amm I don't undertand, you will put in other comment?