grbl/grbl #1126: Homing: Sometimes Homing starts on the wrong way

toptrapping, 2016-11-04:
Hi,

I have created this post because I have an hasardous problem on my cnc. Sometimes when I start GRBL and when I performe a $h to home it, the Z axis starts in the wrong way. (probably 1 time to 3).
I have solved this problem by adding this code at the startup of GRBL
$X
g0 z-0.01
$H.

When I do that, it's quit better but I have the problem 1 time to 40 too

Someone know this problem?

for information, my home switch are wiring with optocoupler and wire are shielded to ground and so I don't think it's a wiring problem.

my config is:

Grbl 0.9j ['$' for help]
['$H'|'$X' to unlock]
$0=10 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=1 (dir port invert mask:00000001)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.010 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=1 (soft limits, bool)
$21=0 (hard limits, bool)
$22=1 (homing cycle, bool)
$23=3 (homing dir invert mask:00000011)
$24=100.000 (homing feed, mm/min)
$25=500.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=3.000 (homing pull-off, mm)
$100=200.000 (x, step/mm)
$101=500.000 (y, step/mm)
$102=200.000 (z, step/mm)
$110=3000.000 (x max rate, mm/min)
$111=400.000 (y max rate, mm/min)
$112=2000.000 (z max rate, mm/min)
$120=200.000 (x accel, mm/sec^2)
$121=50.000 (y accel, mm/sec^2)
$122=100.000 (z accel, mm/sec^2)
$130=1000.000 (x max travel, mm)
$131=101.000 (y max travel, mm)
$132=149.000 (z max travel, mm)
ok

Can you help me.

best regards

Francois.

PS: sorry for my bad english

vMeph, 2016-11-04:
$23=0

chamnit, 2016-11-04:
@toptrapping : When Grbl starts homing in the wrong direction, how far does it go before stopping? Does it go about 15mm? If so, that means that Grbl detected your homing switch as triggered at the start of the homing cycle and is trying to pull off of it. This would also mean that you have some electrical noise or a wiring issue.

The pull-off motion is 5x your pull-off setting distance, and Grbl's homing search phase at the beginning of homing moves 1.5x your axis travel setting. It would be difficult to confuse the two.