grbl/grbl #1127: strange homing issue

Raymoke, 2016-11-11:
HI guys,

i have a cnc shield v3 on an arduino uno and i want to get homing working on X an Y.
but i have issue that i cant explain. i have the following settings:
enabled homing and hard limits and 
set $23=3
set config.h so it does not Home the Z axis ( works )
uncommented the HOMING_FORCE_ORIGIN option in the config.h file and have the switches negative.
debounce = 100

i allready replaced the Switches but no luck.

if i power on the shield and start Homing. it homes the  X > works ( touches the switch moves 100 away comes back touches the switch and goes 100 away again ), homes the Y > works ( same 100, touch, 100 ) so i should say it works but no.

UGS says X0Y0Z0
Active State: home

and in console i see 
>>>$H
OK
Alarm: hard limit
[ Reset to continue ]

When i check with ? i see LIM000 so no switch is pressed.....

i dont get it :(

chamnit, 2016-11-11:
@Raymoke : Since you get and 'OK' after issuing a '$H' homing command, then homing completed just fine. It looks like you are triggering the hard limits immediately after the homing sequence somehow. You likely have lots of electrical noise in your limit pins. Grbl polls the limit pins, so it's very easy to miss noise in a status report. Also, make sure your Z-limit pin is wired to be buffered. It can trigger the hard limits, even though it's not used in a homing cycle. That's how Grbl was written.

Raymoke, 2016-11-11:
thnx for the swift reply, i thought didnt think about wiring the Z axis as i have the CNC shield v3 with the latest grbl version on it and it doesn not support the homing of Z on v3.

vMeph, 2016-11-11:
you can set the z switch on the cnc shield v3 just bend the pins 11 and 12  on shield and swap them then  you can solder on bottom of the arduino uno the swap you did

Raymoke, 2016-11-11:
i just attached the Z+ axis as i had them allready wired but unplugged and it works now!

thnx!

no i want to make a touchplate with Z probing. where should i connect the cables on the cnc shield?

http://i0.wp.com/blog.protoneer.co.nz/wp-content/uploads/2013/09/Arduino_CNC_Shield_Micro_Stepping_Settings.jpg

chamnit, 2016-11-11:
The probe input pin is A5. Whatever that is wired to.

closed 2016-11-11 by chamnit

vMeph, 2016-11-11:
on cnc shield i dont see probe pin label but like chamnit saied is on A5 on arduino, on the cnc shield it should be on SCL conection that is there

swarfer, 2016-11-14:
why not just attach the Z limit to the 'Spin En' pin labelled on the CNC Shield?