grbl/grbl #1128: Z limit switch not working

JoshuaFM, 2016-11-12:
Ok, so I have wired my x, y and z limit switches to an arduino cnc shield, what happens is that when i press the switch from x or y it detects it and stops, but when I press Z, nothing happens. 

I checked my wiring and everything is okay, I used a multimeter on the Pins and the X and Y have 5Volts, while the Z has 0 volts. I know they have 5v because they have an internal pull up resistor, but what is happening with my Z? Why is it not working?

darkfibre-nl, 2016-11-12:
from the wiki:
https://github.com/grbl/grbl/wiki/Connecting-Grbl

For Grbl v0.9 with variable spindle PWM ENABLED: (NOTE: The Z-limit and the
spindle enable pin are swapped, because we had to access the hardware PWM
on D11 for variable spindle PWM output to work.) We are still updating this
pin configuration at the moment by weighing future options. We'd like to
only change the pins once. Stay tuned!

eociar, 2017-01-29:
Hello JoshuaFM!

I encountered the same problem (limit switch for z-axis not working).
In addition to your problem, my whole arduino (UNO-clone) crashed, when the limit switch closed.
My assumption was, that the effect is caused by a short circuit on the limit pin.
I went through the sourcecode and found the follwing:
The function of the pin 12 (input for z-axis limit or output for variable spindle) depends on the define _VARIABLE_SPINDLE_ in include file "config.h". 
This include file is missing in the CPU mapping files "cpu_map_atmega328p.h" and "cpu_map_atmega2560.h".
So depending on the order how the arduino-compiler is compiling the sourcecode, parts of the code might be compiled without the right settings.
I adde the include file "config.h" in "cpu_map_atmega328p.h" like this:
`
.......
#ifndef cpu_map_h
#define cpu_map_h

**#include "config.h"**

#ifdef CPU_MAP_ATMEGA328P // (Arduino Uno) Officially supported by Grbl.

  // Define serial port pins and interrupt vectors.
  #define SERIAL_RX     USART_RX_vect
  #define SERIAL_UDRE   USART_UDRE_vect
......
`
After a recompile and download to the arduino everything was working as expected:
The function of the z-limit switch is ok and the variable spindle on pin 11 also working.

Maybe you can try this for your project, too.

All in all, the whole gbrl project is great!! Thanks to  the author and all contributors!

SteBuTOS, 2017-03-02:
Hi all,

that really looked to be the solution to my problem, but the sourcecode modifications (#include "config.h") did not help.

I installed limit switches for all 3 axis (min and max). All NO connected.

I've set $21 and $22 to 1 and left $5 to 0;
When issuing $H, z-axsis starts to move, but when the z-limit switch becomes triggered, it is not recognized. Instead z-continues moving, crashing into mechanical end.

What I did for troubleshooting so far:
- used multimeter to assure wiring works -> ok
- hooked on DSO to check for noises/ Interference -> nice clean signal -> ok
- set status report flag to enable limit switch indication ($10=19) and testet a each of the 6 switches: every sitch is indicated correctly by the statusreport, according if its triggered or not.

Right now I'm clueless and desperated.
Any help is appreciated!

BTW: I compiled with config.h enabled #define VARIBLE_SPINDLE and therefore connected z-limit switch to PIN 12 and the spindle enable signal to PIN 11 (tested the variable "spindle speed" and ok)

SteBuTOS, 2017-03-02:
one more strange thing: I tried to change the order of homing axis in config.h
#define HOMING_CYCLE_1 (1<<Z_AXIS)    
#define HOMING_CYCLE_0 ((1<<X_AXIS)|(1<<Y_AXIS))
...so that it tries first to go for x and y, but: everything compiled without an issue in arduino IDE, also uploaded without any error or warning: no effect at all

It seems, like entries within config.h take no effect when it comes to homing, but will be considered in other areas (e.g. status report)

109JB, 2017-03-02:
How are you uploading to the Arduino? If using the Arduino IDE, do you only have one instance of the grbl firmware in the library folder.  If you have more than one instance, it is hit or miss which file is used in the upload/ compile.

SteBuTOS, 2017-03-03:
First I compiled and uploaded with arduino.
But then I also uploaded the precompiled hex-files directly with ArduinoBuilder (grbl 9f): same behaviour
What makes things strange is, that the status report recognizes the switch status of all switches correctly. Therefore I would say that it can not be a hardware issue.

electrokean, 2017-03-04:
@SteBuTOS First, you should probably move to grbl 1.1 at https://github.com/gnea/grbl/ as grbl 0.9 is no longer recommended for new installs or requesting support.

Second, if you changed the HOMING_CYCLE definitions order and recompiled/uploaded the change, and it made no difference to homing - then you aren't modifying the right copy of grbl.  Make sure the only copy you have and are editing is under the Arduino libraries folder and called grbl.  There must be no other copies even if you've renamed them.

Third, you probably know this, but it is always best to home Z first so that the XY homing can avoid obstacles.

electrokean, 2017-03-04:
@eociar your comment re adding `#include config.h` to the cpu mapping includes doesn't make much sense.  The cpu mapping include files are only used by grbl.h, which has already included config.h.
So unless you were using a non standard procedure to compile, the only way this could have happened is if you had an old version which didn't do the includes in this order, but that seems unlikely as that area of code hasn't really changed in a long time.

closed 2017-03-05 by chamnit

ditrik, 2017-12-06:
Hello to all
I have the problem that grbl does not recognize the limit switch of the Z axis and I found the answer by rereading the characteristics of the various versions of grbl. In the 0.9x versions, the D9 (X) D10 (Y) D12 (Z) pins are assigned to the limit switches, while the cnc shield uses the D9 (X) D10 (Y) D11 (Z) pins. perfectly interpreted by the 0.8c version of grbl, I reloaded the 0.8c firmware on Arduino UNO and now everything works correctly.

vMeph, 2017-12-06:
You can easy adapt the cnc shield to the new versions of grbl, you bend the cnc shield pins that needs to be swaped the you can solder the 2 wires on the bottom of arduino and conect to the bend pins

ditrik, 2017-12-07:
Excellent idea and solution, thank you very much.

arjun1299, 2020-08-11:
In grbl 1.1 just comment out the VARIABLE_SPINDLE line in config.h to get it working if you don't need hardware PWM output on for variable spindle output voltages.

MikePelton, 2020-11-02:
Wanted to say thanks to arjun1299 - commenting out VARIABLE_SPINDLE fixed the issue for me - there's a detailed explanation of what it's all about in config.h itself. You'll need to delete previous Arduino libraries, recompile and re-upload. My Z limit switches work now - thanks!!

IvanGutowskiSmith, 2021-03-27:
The above steps answer the question, however for anyone unsure on the exact steps to do this:

Firstly to test if what is and is not working. Connect to the arduino using something like universal Gcode sender, (Baud rate may have to be 115200)

Changing the $10=   <<numbers here
will change the types of reporting you get. 
sending $$ will show you what settings your arduino has

For me entering $10=17
then sending: ?
Gave me a result ending in 
Lim:000
1=limit switch triggered
Order those 3 zeros refer to is: X > Y > Z

Holding down endstops and sending ? allows you to check the status. 

On Github I then downloaded the grbl 1.1 file. Extracted it (with e.g. winrar) and using notepad ++ opened the file:
grbl-master\grbl\config.h

Pressed CTRL + F to search for: _VARIABLE_SPINDLE_
for me this was line 247 - I then put two slashes infront so it read:
//#define VARIABLE_SPINDLE // Default enabled. Comment to disable.

I then saved the file.

Using the arduino official computer program:
Sketch>include library>add .ZIP library

When navigating to the file you need to select the grbl folder, within the grbl-master folder.

Then ensure you're connected to your arduino: Tools>port> _your com port number here_
also ensure under tools>board, you've chosen the right arduino board e.g. 'uno'

Then, sketch>include library>grbl (at the bottom)
then sketch>verify/compile

Then press the right hand arrow icon to upload it. 

Close the arduino program.
On univeral gcode sender connect to your board. You may ahve to play around with the numbers on the $10=
But for me now, entering $10=17
and then ?
Shows the first number on the limit field as 1, when I press my z limit switch.
Lim:100

If you see that, it should work. You can also invert the limit switches if needed, if you want a normally closed vs normally open set of limit switches. Reverse by setting $5=    to 1

I've spent so many hours reading helpful articles, and even though this one gave the right answer, it was a faff to know how what to comment out and compile. I've never posted, only consumed help articles.. so hopefully in like 5 years time this will help someone.

Have a wonderful day and hope this saves you a few hours of troubleshooting o7

Wonko-the-Sane100, 2023-11-27:
Thanks for this...battling with same issue. Hoping to resolve using your suggestions

IvanGutowskiSmith, 2023-11-27:
> Thanks for this...battling with same issue. Hoping to resolve using your suggestions

So glad this helped, from memory my board died and only when getting a second did I realize my code was fine but hardware broken.

I'm also very surprised at my past self for the comment I added, I usually just consume content, never write up a fix and move on. However glad I was able to help, it will encourage me to contribute more solutions in the future.

Best of luck