grbl/grbl #1130: Creeping Axis Errors

PeterHunt848, 2016-11-13:
Hi Guys,

I am running a grbl converted 6040 using this board from 3Dtek http://3dtek.xyz/products/grbl-aio 
The initial setup and calibration was not a major issue, but after a two weeks of head scratching I have one major issue. 
The machine creeps. Always Y positive, and seemingly at a constant rate. 
I have switched the X and Y directions on the machine to eliminate the possibility of it being a mechanical issue and found the error persisted according to the electrical signal, i.e. it swapped on the machine. 
I have slowed the max feed rate to 1200 mm/min, in thinking it could be overloading the poor little 328p, but no cigar. Is there anything further I can test in this regard?
 
The 3Dtek board has a motor driver for a tandem motor, so I have utilized this in place of the Y driver, attempting to isolate the error to the board. This has caused (what could be an altogether separate issue) the Z axis to creep negative during a cut (turning a spiral into a helix for example).

It might be worth mentioning that the errors outlined are within a single operation with no homing or funny business in between and Fusion 360 was used to create the code.

Attached is the G-code for a test I have been doing. You will see that it does a pocket in the small hole, then does all the operations in the large hole, then comes back to do the through cut on the small hole. It is this through cut that is always 1 mm (+/- 0.2) non-concentric to the initial pocket.

[Non-Concentric Test 1.1.txt](https://github.com/grbl/grbl/files/587546/Non-Concentric.Test.1.1.txt)

Any help on this would be hugely appreciated.

chamnit, 2016-11-13:
@PeterHunt848 : Make sure your step idle delay setting is set to 255. This will keep the steppers from disabling during a job when there is some buffer starvation. When you have several consecutive G2/3 arcs, there can be instances that your GUI streamer may have enough latency to not sent the next G2/3 arc before Grbl finishes the last one. In this case, if your step idle delay is less 255, your steppers will disable momentarily. This will lead to some drifting that is hard to see.

Second a spiral helix plunge operation is very hard on the z-axis. Typically z-drift is associated to a z-max acceleration that is set too high. Lower your z max acceleration and it should go away.

chamnit, 2016-11-13:
For the record, the 328p is just as fast as a Mega2560, which runs the vast majority of 3d printers with Marlin. It's more than capable for a large majority of CNC applications.

PeterHunt848, 2016-11-13:
Thanks chamnit

I checked the settings and the step idle delay is set at 255. Also I slowed the acceleration of all the axes to 100 mm/sec/sec with no change. 

I do not know if this is related, but in a previous trouble shoot I found electrical noise generated from the spindle cooling pump was causing a board reset during jobs. I fixed this with a simple ferite bead on the pump line. Is there any way for electrical disturbance, created elsewhere in the system, to cause an error in the Y steps? 

Sorry If I am asking the wrong questions, but it isn't obvious to me what could be the issue.

chamnit, 2016-11-13:
Can you paste you Grbl $ settings please?

chamnit, 2016-11-13:
And are you using an Arduino with a CH380 USB-serial driver? This has been known to slightly truncate serial streams in certain situations.

PeterHunt848, 2016-11-13:
[image]

I don't know which serial driver I am using. It is the 3Dtek board (not Arduino). http://3dtek.xyz/pages/grblaio

Thanks a million for putting your thinking hat on for me.

electrokean, 2016-11-13:
You might want to check your current setting pot on the problem axis isn't set lower than the others

chamnit, 2016-11-13:
Change your junction deviation setting to 0.01. 0.05 is way too high and can lead to lost steps.

PeterHunt848, 2016-11-13:
Thanks team, we did it!

I just got my first successful job, absolutely flawless. 

I checked the pots as per electrobean's suggestion and they were set correctly, so it looks like it was the Junction Deviation. 

Just a quick ups to the whole platform, I couldn't expect better customer service from even the biggest of companies, so this quality coming from an open source platform is simply astounding! Here I was thinking I was in the dark, on my own because I went with free ware, now I know I should have just asked sooner.

Happy CNCin' Everyone!

chamnit, 2016-11-13:
Thanks for the update and glad to help.

closed 2016-11-13 by chamnit