grbl/grbl #1131: All direction pins must be on the same port?

markwbrown, 2016-11-14:
I'm wading through the code to see what is needed to get grbl running on mega 2560 with grove shield. I've got external stepper drivers, and the 4 pin grove connector looked to be a handy option for wiring the arduino to the stepper drivers (5v, ground, and two digital pins), but I came across "All direction pins must be on the same port." The same type of note is made for step pins as well. Now, I'm thinking the stock grove shield may not be what I wanted.

I'm not really sure what is meant by "on the same port" and am looking to get some clarification. Is this to mean on the same port register? i.e. for a 4 stepper system (2x, 1y, 1z motor) pins 15-18 (on register H) would be appropriate for step, and 23-26 (on register B) would be appropriate for direction?

swarfer, 2016-11-14:
Exactly that, on the same port like all on H or all on B as you postulated.

Why not start with the actual Mega port at 
https://github.com/gnea/grbl-Mega/
?

markwbrown, 2016-11-14:
Thanks, @swarfer. I hadn't seen the gnea port of this.

closed 2016-11-14 by markwbrown