grbl/grbl #1133: Timeout waiting for GRBL to Idle during cancel, cancel incomplete.

russellandsusan, 2016-11-16:
Hello All,

I'm not new to Arduino but am new to working with GRBL. I made myself a very simple file in Fusion 360, CAM'd the paths and made myself some G Code. I have been able to connect to my GRBL and operate all functions within the software. I have NOT yet connected motors - I just wanted to make sure I could get everything to talk correctly.

Note: I have the latest HEX file uploaded and used XLoader to do it. My new (non genuine) Arduino UNO had some problems but I found them to be because of JAVA permissions - This appears to be completely corrected. I have also run some other programs I have built in the past to 'stress' the Arduino UNO and all of those work fine. I can not give specific makes/models of the GRBL shield at the moment as it is at work however here is the link where I got it from and the items I received are identical to what is pictured, albeit short one driver board :(

Purchase link: http://www.ebay.com.au/itm/CNC-Shield-V3-0-UNO-R3-Board-A4988-Driver-Heatsink-Kits-for-Arduino-CNC-Kit-/191959361357

The only problem I am coming across is when I try to Cancel an operation once G Code is running. Pressing cancel returns the following error:

`
>>> X65.267Y52.165
ok
>>> X65.266Y52.161
ok
>>> X64.795Y50.865
**** Pausing file transfer. ****
ok
**** Canceling file transfer. ****
Timeout waiting for GRBL to Idle during cancel, cancel incomplete.>>> G90 G0 X0 Y0
**** Canceling file transfer. ****
Timeout waiting for GRBL to Idle during cancel, cancel incomplete.
`

I have spent an hour or so trying to search this error but have had no luck yet.

If anyone can help, it would be much appreciated.

electrokean, 2016-11-17:
That sounds like a GUI/protocol compatibility issue

For starters, I'd suggest trying a different GUI - maybe bCNC - to see where the problem lies
See https://github.com/vlachoudis/bCNC/

You may also want to try the latest (beta) release of grbl which supersedes this one
See https://github.com/gnea/grbl/

russellandsusan, 2016-11-18:
Thanks mate.

I'll give it a go tonight and let you know how things pan out - Thanks electrokean.

bCNC looks good but I know nothing of Python and I'm running a windows 10 environment so it's out of my league (level of caring is probably more precise...)

electrokean, 2016-11-18:
OK, another great option for Windows is https://github.com/gerritv/Grbl-Panel/

winder, 2016-11-18:
@russellandsusan are you using a version of UGS from before November? This issue should be fixed in the latest nightly build as of November 1.

nsiatras, 2016-11-24:
I have build this GCode Sender https://github.com/nsiatras/sourcerabbit-gcode-sender/
It is fully asynchronous and has an **event drive UI** which informs you in real time about the machine and work status.

I have also implement tools for Touch Probe and Hole Center finding and at the moment I am implementing an Edge Finder tool. Give it a shot, I am sure you will like it.