grbl/grbl #1134: how to add custom command in grbl perser

shamsiqbal, 2016-11-19:
hi 
i want to add some new parameters to control some external hardware and also want to store those parameter in internal eeprom so it can retrieved during power up same like as $100 etc format 
so plz help me out that how to add these parameters in serial parser where i do not alter existing parameter functionality and compatibility 
here i also want to add some report massage also where it shows additional hardware functional parameters 
waiting for quick help
regards

frett27, 2016-11-20:
I've made that in this GRBL fork, http://gihub.com/frett27/grblpunch seeing the modified elements from the given grbl trunk. Hope this help

frett27, 2016-11-20:
We added a M100 new command

shamsiqbal, 2016-11-20:
dear
thanks for input can you tell me which file of grbl you added these commands i want to do same where i want to add spi interface of my motors drives so there parameter can be adjusted with same grbl gui 
regards

frett27, 2016-12-02:
you can have a detailed view with diffs from the forked, here is a pointer : https://github.com/frett27/grblPunch/commit/26c23ce3ccd608f8714f83509379fd04ab345a3c

closed 2017-01-26 by chamnit