grbl/grbl #1135: No (complete) boot on older PC

JohanTL, 2016-11-20:
I succeed in set everything up: Atmega328 (with GRBL 8) - GRBL shield V5 (al powered from USB port).
Everything is working smooth. I can send g-code to the motors.

I want to use an older laptop for my project, but I get no good connection with the atmega328. When I'm plugging in the USB cable, the led on the atmega flashes once (on my other computer it flashes several times during boot). I am sure that the setup on both computers is equal.
I can't send G-code (it says that GRBL boot is not finished). The arduino ide is also not successfull.
I don't find a reason why it is not working on the older laptop. 
Any suggestions to get it to work on the older laptop?

winder, 2016-11-20:
It sounds like you might not have the FTDI drivers installed. You can go through the steps here to make sure everything is installed: https://www.arduino.cc/en/Guide/HomePage

JohanTL, 2016-11-20:
Yessss! Thank you!
Weird that I have to install it separately on that machine... On the other computers: no problem.

closed 2016-11-20 by winder