grbl/grbl #1136: Feed rate changes ignored

ChristianRenaux, 2016-11-22:
Hi,

I'am using the latest grbl version 0.9j and everything works fine except that changing the feed rate is ineffective on any of my stepper motors ( three different types ) through DV8825 drivers.  I have used so far the universal gcode sender ( tried two versions) and also tried the simple_stream.py python script and the behaviour is always the same. 

I have connected a scope to one of the motor step signal on the arduino Uno and clearly it does not change with different feed rate values.  Therefore the problem should be related to the grbl or my settings, which are: 

Grbl 0.9j ['$' for help]
>>> $$
$0 = 10    (step pulse, usec)
$1 = 25    (step idle delay, msec)
$2 = 0    (step port invert mask:00000000)
$3 = 0    (dir port invert mask:00000000)
$4 = 0    (step enable invert, bool)
$5 = 0    (limit pins invert, bool)
$6 = 0    (probe pin invert, bool)
$10 = 3    (status report mask:00000011)
$11 = 0.010    (junction deviation, mm)
$12 = 0.002    (arc tolerance, mm)
$13 = 0    (report inches, bool)
$20 = 0    (soft limits, bool)
$21 = 1    (hard limits, bool)
$22 = 0    (homing cycle, bool)
$23 = 0    (homing dir invert mask:00000000)
$24 = 25.000    (homing feed, mm/min)
$25 = 500.000    (homing seek, mm/min)
$26 = 250    (homing debounce, msec)
$27 = 1.000    (homing pull-off, mm)
$100 = 100.000    (x, step/mm)
$101 = 50.000    (y, step/mm)
$102 = 1.000    (z, step/mm)
$110 = 50.000    (x max rate, mm/min)
$111 = 500.000    (y max rate, mm/min)
$112 = 500.000    (z max rate, mm/min)
$120 = 500.000    (x accel, mm/sec^2)
$121 = 10.000    (y accel, mm/sec^2)
$122 = 10.000    (z accel, mm/sec^2)
$130 = 200.000    (x max travel, mm)
$131 = 200.000    (y max travel, mm)
$132 = 200.000    (z max travel, mm)

Any idea ?

Best regards,
Christian.

swarfer, 2016-11-22:
Your X max step rate is 50mm/min which is very slow.  Are you sure this is the correct value?

Any move that involves X will not move faster than X can move, ie all X & Y moves will be restricted to 50mm/min.

and Z is set to 1 step/mm.  are you sure?

109JB, 2016-11-22:
Are you using G0 or G1 to move?

ChristianRenaux, 2016-11-23:
Thank you for your help, you were both right.  At first the X max step rate that has been changed by error and I did not noticed it.  After correction, the X motor travel at full speed only and could not reduce it, and that where the G0, G1 commands come into action.  Because it was the first trials I was using the jog buttons and it is a G0 command.  And for ease, I was just copy and paste that command, changing only the speed rate.
  
So it now works correctly with the G1 command instead.

The Z 1 step/mm is correct as it is in fact microns and not mm in my case, so that it is easier to program.  I am upgrading a 25 years old semiconductor wafer dicing saw to control all the motors.

Best regards,
Christian.

closed 2016-11-25 by ChristianRenaux