grbl/grbl #1137: Bresenham style arc algorithm in GRBL?

usbcnc, 2016-11-23:
@chamnit, have you thought of putting Bresenham style arc algorithm in GRBL? 
There is plenty information in computer graphics that explains how it is done, it can maximize the machine accuracy with very little calculation. So far I have not found CNC world is using it. Google "midpoint circle algorithm" should give you plenty of information on how this is done in theory.

Using this algorithm not only increase the accuracy but also use less memory and CPU power, I did small experiment and the code on STM32 target and the code size is a little less than the current method. The most saving is by eliminate the function of atan2.  I have not yet do experiment on Ardurino but if there is enough interest I can do certainly do it.

In order to keep the constant feed rate on arc, we need to change the timer for interrupt for each step, this also lead the need of per-step-acceleration. That means we could not use AMASS. With per step acceleration, the code saving is about additional 2KB-2.5KB.

Strooom, 2016-11-23:
I read this paper http://www.wseas.us/e-library/conferences/cancun2004/papers/485-304.pdf
and I think it would be difficult on the Arduino as it requires at least 3 multiplications per step.
As you mention, to keep a constant velocity through the arc is also not simple...

Maybe it can be done by reducing max feedrate in arcs G2/G3 so the timer ISR has a bigger time budget.
For G0 it is always a straight line, so these can be faster.
I would stick to AMASS, but again only when you have time for it, so when the feedrate is lower.

I am currently writing all the above in a grbl port for ARM (Teensy 3.2 = Cortex M4), so let me know if you are interested, I'll post a repository on Github.

usbcnc, 2016-11-23:
First, The Bresenham algorithm is moved out of the ISR line or arc, so the issue of the time budget is not an issue. 
Second, I have improved version of the arc calculation and no multiply is used in calculation the points. However to calculate the time (keep constant speed), I need to do some division.

So far, the combination of the resolution resulting several deviation in terms of the float point or integer calculation.

Case 1. Both X and Y are of the same step/mm, float point or not, this can lead to only integer plus and minus.

Case 2. X and Y are of different step/mm, both are integer, then only integer plus and minus is used. This is ellipse not a circle.

Case 3. X and Y are of different step/mm and both are floating point, then there is about 1/3 of the float point plus and minus needed. 2/3 are still integer calculation.

I am experimenting on cheap STM32 (cortex M3) board widely available at about USD$1.6 shipped. On aliexpress and search "stm32f103c8t6 board".
 It has 64KB published flash but in fact is 128KB flash. 20KB of ram. It is far more capable to hold GRBL. Since it does not have EEPROM, I use 1K of ram and 1K of flash to simulate the EEPROM. It has built in USB to work as serial port. 
The initial port of GRBL was on github already. The Bresenham style of arc is not since it is not complete yet. Also I might consider to publish my algorithm before make the source code available.

I will start to port it back to Ardurino to see if it works to my expectation.

chamnit, 2016-11-23:
@usbcnc : I've looked into this about 5 years ago. The stepper implementation isn't a problem, but it breaks the planner. You'd have to modify it to allow for accelerations and decelerations throughout the arc and account for transitions into and out of the arc. It's not that you couldn't, but it's a lot easier to deal with lines only.

closed 2016-11-23 by chamnit