grbl/grbl #1138: Y Axis not working

FlyingPyroDoge, 2016-12-02:
More specifically, the pin 3 is not working on my arduino for some reason. It was working earlier but it isn't now. When I plug the easy driver into the x axis (pin 2) It works. I tested with my multi meter and the other pins go up to .05v when the steeper moves, but it stays at 0v on the y axis. I've tried to build a new viersion of grbl with different pin outs but it isn't working for me. Does anyone have a different version of grbl with different pin outs or can I make it so it combines the Z axis (I'm not using the Z axis) With the Y axis?

109JB, 2016-12-02:
So if I have this right, you are only using X and Y axis, but pin 3 (Y step) is dead.  In that case you can swao the Z axis and Y axis pins by editing the cpu_map.h file.  You will want to change these lines

#define Y_STEP_BIT      3  // Uno Digital Pin 3
#define Z_STEP_BIT      4  // Uno Digital Pin 4

to 

#define Y_STEP_BIT      4  // Uno Digital Pin 4
#define Z_STEP_BIT      3  // Uno Digital Pin 3

Then you would use pin 4 for the Y-axis step pin and pin 3 for the Z-axis step pin. All others remain unchanged.

FlyingPyroDoge, 2016-12-02:
You are correct I'm only using the X and Y axis. I tried that, but when I compiled it and uploaded it to my arduino with arduino IDE Grbl Controller can't connect to it. The only version of Grbl that has worked on my arduino so far is .8c hex file. I'm new to this so I don't know what else to do. Is there a way to mix the Y and Z axis together so I can use the output of the Z for Y?

FlyingPyroDoge, 2016-12-02:
I don't know what I did, but it is working with the same pin now, Thanks 109JB for the help anyways!

closed 2016-12-02 by FlyingPyroDoge