grbl/grbl #1139: need to always enable motors

vladislav-konoplyev, 2016-12-02:
Hello! I need to use grbl in wall plotter project. In this case i need that drivers should be always enable. When i do DEFAULT_STEPPER_IDLE_LOCK_TIME = 255 in defaults_generic.h , nothing changes.
How should i solve this problem, tell me please?
Best regards

chamnit, 2016-12-02:
@vladislav-konoplyev : You need to either write that value in manually with a $ command or restore your EEPROM with a `$RST=*`, which will wipe everything and reload all defaults.

closed 2016-12-02 by chamnit

vladislav-konoplyev, 2016-12-02:
Thanks a lot!

vladislav-konoplyev, 2016-12-03:
i try to send $RST=* and try to clean EEPROM than recompile grbl and upload, but nothing help.
grbl answer after $RST=* - "restore defaults ok" but nothing changes in behavior of plotter. drivers doesn't always enable. What i do wrong?
Best regards

jahnj0584, 2016-12-03:
If you changed the grbl code you have to load it into the Arduino IDE with
a different name. Otherwise it'll ignore it if there's already the same
name in the library.
You can also just do $1=255 in your control software

<https://github.com/grbl/grbl/issues/1139#issuecomment-264639589>, or mute
the thread

vladislav-konoplyev, 2016-12-03:
Thanks, now it's works for me. In parallel it became clear that it was also in a weak holding torque setting of the motors