grbl/grbl #1140: Occational incomplete status report

milesingrams, 2016-12-06:
Hello!

I have a GUI that polls status reports for an IDLE status to determine when a motion is done and the next motion can begin, however a couple of times I've encountered an incomplete status report out of the blue that can cause problems for my parser. Usually it missed the X and Y coordinates from the report, and reports everything else. This may have been fixed in v 1.x but in case it isn't I thought you should know. I have fixed the bug on my end by ignoring incomplete reports. heres a screenshot of my log of the grbl feedback.

[image]

Thanks!

chamnit, 2016-12-06:
@milesingrams : Not sure what's going on there. I've never had this issue reported by other existing GUI devs. So, I would like to think it's something particular to your setup or GUI. If you have an Arduino with a CH380 USB-serial chip, this is well known to be problematic when there is a lot of serial data. It frequently has buffer overruns and loses data.

milesingrams, 2016-12-06:
I'm using the official Arduino Uno Rev 3 so I don't think it has a CH380 USB-Serial chip. The bug has only happened twice in a couple of months of constant running, so perhaps it is just a really rare issue. Also when I poll using the '?' command I am doing it at a rate of 10 times a second, which might be too fast, I should probably slow that down to 5 times a second.

chamnit, 2016-12-06:
@milesingrams : Ok. That rules out the CH380. Does it seem to drop entire values like you have shown? Are there any other patterns you have noticed with this? 

It seems strange that you look to be moving the z-axis up and the Z position shown is the same as one of the prior ones, but lower than the last. Maybe this is a clue.

chamnit, 2016-12-06:
Can you post your $I info string and settings? Also, what software or libraries are you using for your GUI?

milesingrams, 2016-12-06:
$I Info String
[0.9j.20160317:]

$$ Settings
$0=10 (step pulse, usec)
$1=250 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=0 (dir port invert mask:00000000)
$4=0 (step enable invert, bool)
$5=1 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=1 (status report mask:00000001)
$11=0.020 (junction deviation, mm)
$12=0.010 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=1 (soft limits, bool)
$21=0 (hard limits, bool)
$22=1 (homing cycle, bool)
$23=7 (homing dir invert mask:00000111)
$24=120.000 (homing feed, mm/min)
$25=3000.000 (homing seek, mm/min)
$26=100 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=20.000 (x, step/mm)
$101=20.000 (y, step/mm)
$102=200.000 (z, step/mm)
$110=12000.000 (x max rate, mm/min)
$111=12000.000 (y max rate, mm/min)
$112=3600.000 (z max rate, mm/min)
$120=800.000 (x accel, mm/sec^2)
$121=800.000 (y accel, mm/sec^2)
$122=160.000 (z accel, mm/sec^2)
$130=310.000 (x max travel, mm)
$131=310.000 (y max travel, mm)
$132=135.000 (z max travel, mm)

And I'm using Node.js (https://nodejs.org/en/)  running a library called node-serialport (https://github.com/EmergingTechnologyAdvisors/node-serialport) to interface with the grbl via serial.

milesingrams, 2016-12-06:
It is curious that the shortened status report looks like a copy of the status report 2 lines previous except missing the X and Y values. If I remember correctly the last time it happened the pattern was the same: The X and Y were missing and the status had a Z value and line number identical to the status a few lines previous.