grbl/grbl #1141: Sanguinololu bad pin map.

Cuningan, 2016-12-07:
hello, i have tested grbl on a sanguinololu with 1284p Atmega and found that the limit switches do not work nothing at all.
After checking all that can be checked i started looking to software issue and found that the pin assignement is not correct according to Sanguinololu Schema.

In the original cpu_map.h the pins was assigned as in original.png file.
[image]
But looking to the Sanguinololu schematic the pins must be assigned as in modificado.png
[image]
I have changed the pin bit in the cpu_map.h and all work as expected now.
[image]

Also i have cloned this git and modified to support ATmega1284

https://github.com/Cuningan/Grbl-with-Sanguinololu-Atmega1284

At least for me is working now, maybe it could help someone else.

djecom1, 2024-02-25:
@Cuningan : 
If you still have your grbl code fore Sanguinololu 1284p, could you please share it again ?
Thanks,
Jerome.

xiaoshidecaihonghai, 2024-02-25:
没有了

Nchip1001, 2025-03-09:
If you still have your grbl code fore Sanguinololu 1284p, could you please share it again ?  I flash the board via Arduino IDE. Arduino gives an error eeprom. But the upper level does not see the device.