grbl/grbl #1143: Homing problem with 2 Axis laser engaving

jimthan, 2016-12-10:
I have read all the post about homing ,so the motors not working wen i type ~ or $H.
My settings is 

> 

 Grbl 0.9j ['$' for help]
['$H'|'$X' to unlock]
$0=10 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=0 (dir port invert mask:00000000)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.010 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=1 (hard limits, bool)
$22=1 (homing cycle, bool)
$23=3 (homing dir invert mask:00000011)
$24=50.000 (homing feed, mm/min)
$25=500.000 (homing seek, mm/min)
$26=244 (homing debounce, msec)
$27=3.000 (homing pull-off, mm)
$100=133.333 (x, step/mm)
$101=133.333 (y, step/mm)
$102=1.000 (z, step/mm)
$110=500.000 (x max rate, mm/min)
$111=500.000 (y max rate, mm/min)
$112=1.000 (z max rate, mm/min)
$120=10.000 (x accel, mm/sec^2)
$121=10.000 (y accel, mm/sec^2)
$122=0.000 (z accel, mm/sec^2)
$130=500.000 (x max travel, mm)
$131=440.000 (y max travel, mm)
$132=1.000 (z max travel, mm)
ok

> 

and config ..

> 75- #define HOMING_CYCLE_0 ((1<<X_AXIS)|(1<<Y_AXIS))    // REQUIRED: First move Z to clear workspace.
76- #define HOMING_CYCLE_0 ((1<<X_AXIS)|(1<<Y_AXIS))  // OPTIONAL: Then move X,Y at the same time.
77- #define HOMING_LOCATE_CYCLE ((1<<X_AXIS)|(1<<Y_AXIS))  // OPTIONAL: Uncomment and add axes mask to enable

> 
i have the same problem with  limit switch  enable or  disable  nothing moves.
i have 2 buttons for eath axis and  they are connected in parallel 

what am I doing wrong
jim

jahnj0584, 2016-12-10:
What limit switches are you using

jimthan, 2016-12-10:
No
[[image]
](url)

jimthan, 2016-12-10:
when I press them manually , i have the limit alarm on my board

jimthan, 2016-12-10:
maybe i have problem with config file ? 
i edit with WinARV and i compile him with IDE 1.7.10  Sketch/add file/upload

chamnit, 2016-12-10:
@jimthan : You likely edited the wrong config.h file. You need to edit the one that was copied to your Arduino library folder. 

Change only this in the config.h file and comment out HOMING_CYCLE_1.
#define HOMING_CYCLE_0 ((1<<X_AXIS)|(1<<Y_AXIS)) // REQUIRED: First move Z to clear workspace.

jimthan, 2016-12-10:
@chamnit 
> You likely edited the wrong config.h file. You need to edit the one that was copied to your Arduino library folder.> 
i do not I understand what you mean adout the one file that copied to Arduino.
I choose the file from  C:\--- Grbl_master/grbl/config--- It is right what I do ?

I have made the change that you tell me and i will make a test.

jimthan, 2016-12-11:
unfortunately still I have the same problem. programs freezing check with both programs. universalGCODE SENDER  & Grbl controller.
I think for all this blame the config file I have checked all the circuits with MULTIMETER and working properly.
@chamnit Is it easy to send me a file ready config.h

chamnit, 2016-12-11:
@jimthan : I've updated the [compiling Grbl](https://github.com/grbl/grbl/wiki/Compiling-Grbl) wiki page. It contain information on how to alter the config.h and setup your Arduino IDE to flash Grbl properly. Please follow those directions.

closed 2016-12-11 by chamnit