grbl/grbl #1144: z axis is not mvoing correctly

sieuwe1, 2016-12-13:
Hello GRBL

I have a problem with my z axis. When i want to for example move the axis 1mm from the 0 position i type g0 z1. But when i do this it is not going up 1mm. It only turns the stepper 1/15 of a turn. 

I know this has to do with the steps/mm setting in GRBL but i have tried a lot of different values. When using your formula of the steps/mm i came to the number 2048/1= 2048. But when i use this number the motor turns 1/8 of a turn and then stops but starts making a very high pitch noise and then a second later the sounds stops and the motor turns a bit like 1/8 of a turn. 

I know that i should also use the micro steps in my calculation but i think the motor is already trying to step to fast so i do not think that using the micro steps what will make the steps/mm even higher will make it better. So my question is what value do i need to use for my steps/mm?

Hope you can help.

Greetings Sieuwe

jahnj0584, 2016-12-13:
What are the current settings? Try 2000 mm/min Max with 50 acceleration

swarfer, 2016-12-14:
Hi
Maybe a calculator will help you?
   http://swarfer.co.za/cnc/grblcalc.html
Just put in your actual settings and it will tell you your correct steps/mm.    (one does not guess this value, this is science not art :-)
In other words, exactly what is your motors steps/rev, your leadscrew pitch, and your microstep setting?

Now, moving, then stopping with noise, then moving again sounds  like the maximum speed is much too high.   Set it to a low value with a low acceleration and make sure you are getting the correct distance, then double the value and test until it fails to reach the speed, then go back one step and do the same with acceleration, double it till it fails then go back to the one that worked.

lalo-uy, 2016-12-14:
Start by lowering the puleses per second on the Z axis . The hi pitch sound
is a clear indication that the motor can't move so fast.

sieuwe1, 2016-12-14:
Hi 

Thanks for all your quick responses. I will try to calculate new values using that site and check if it helps when I have time off from school.

For jahnj0584
I have all the settings back to default state right now.

Greetings Sieuwe

sieuwe1, 2016-12-19:
Hello

I have tried to calculate the steps/mm with that grbl calculator. But when i use all the correct values i get these settings.
------------------------------
$0=10
$3=0
$11=0.02
$12=0.002
$100=26.666666666666668
$101=26.666666666666668
$102=32768
$110=3000
$111=3000
$112=2000
$120=500
$121=500
$122=500
--------------------------------------
When i use these setting the stepper motor does nothing. Probably because the motor needs to spin so fast that it can not do anything. 

Maybe someone can help me calculate the steps?
My stepper the Byj48 = 4096 steps in 8 microstep mode.
The driver a easy driver = 8 microstep setting as default.
My leadscrew = 10 turns of the thread on a 1cm lenght (the leadscrew is metric). Here is a picture of the leadscrew.
[image]

Hope someone can help.

Greetings Sieuwe

pth3k, 2016-12-19:
is that a 28BYJ-48 stepper? If so, the docs say that is a 64 steps per rev motor.
So thats 8microsteps * 64 = 512 steps/mm. Not a good choice for a stepper motor used by GRBL.
You really need to look for a 200 steps/rev or 400 steps/rev motor in the NEMA 23 site form factor.

Here is a data sheet for the 28BYJ-48 motor:
http://cdn.instructables.com/ORIG/FQY/RHST/HMVJ7U24/FQYRHSTHMVJ7U24.pdf <http://cdn.instructables.com/ORIG/FQY/RHST/HMVJ7U24/FQYRHSTHMVJ7U24.pdf>

The lead screw looks like it is most likely an 8mmx1.0mm pitch threaded rod.

Ty Hoeffer 
Palmyra, VA

swarfer, 2016-12-20:
Google tells me that the BYJ48 is a 64 step per rev motor WITH a 64:1 gearbox.
This means it takes 512 steps to turn the output shaft one rev.
If you then multiply that by 8 microsteps you get 4096 steps per rev.
Now you couple that to a 1mm pitch leadscrew and you get
32768 steps per mm
However, you can only set your max speed to 54mm/min.  Any higher and you exceed the GRBL limit of 30000steps per second.

What to do?
a) use a different motor.    Common motors are 200 steps per rev, direct drive to the leadscrew.
another good reason to use a different motor is the backlash in the gearbox.

b) use a different leadscrew, say an 8mm pitch like this http://openbuildspartstore.com/8mm-metric-acme-lead-screw/
Now you can use the settings
$102=4096   steps/mm
$112=439   max mm/min
which is usable, but not really fast enough, 400mm.min is a cutting speed not a rapid speed (-:

c) use the 8mm pitch leadscrew AND reduce the microstepping to 2x.   (1x is usually very rough but you can try it)
now the settings can be
$102=1024   steps/mm
$112=1750   max mm/min
which is quite usable (but that backlash will still get you at some point)

sieuwe1, 2016-12-31:
Hey

So i have gotten a different stepper motor. I am on a budget so i got this one for free from a friend. He did not knew a lot about it but after counting the steps it was a 48steps/rev motor. The motor is nema17 sized. After i calculated the values in the grbl calculator i was ready to drill. But when i tried to lift the drill the stepper motor made a high pitch sound again. So after changing for 2 days with the settings trying and trying lower and higher values for $102 $112 $122 i had still no luck. I came close on one point with $102=567 $112=500.00 $122=10.000. But then it would not spin correctly at the beginning and end because than grbl is accelerating or decreasing the speed. When i was using low values like $102= 200 it looked like the motor is trying to turn but it cant because of to much resistance in the lead screw. so  when i changed it so that it had almost no resistance and i could be very very easily turned by hand it still would not work. I know that i should not just guess numbers but when the calculator gives wrong values that is the only thing i can try.

Hope someone can help me. 

Greetings Sieuwe

chamnit, 2016-12-31:
@sieuwe1 : It sounds like you are not giving your stepper enough power or your steppers are mis-wired. You should not be able to turn the stepper easily by hand when it is enabled.

sieuwe1, 2016-12-31:
Hey @chamnit

I am able to turn the leascrew easy when the stepper motor is turnen off. When I have power to the stepper motor i cant turn the leadscrew easy. I have change de the pot on the easy driver boards to give more current but that didn't help.

Greetings

chamnit, 2017-01-01:
@sieuwe1 : How much voltage are you applying?

swarfer, 2017-01-01:
Hi
48 steps per rev x 8 microsteps  on 1mm pitch gives 384 steps/mm, that is the $102 you MUST use in order to get the correct distance travelled.  
(the calculator is absolutely correct by the way, just because your gearboxed motor has way too many steps per rev for GRBLs _step rate limit_ does not make the numbers wrong!  12 days ago I gave you several solutions, one of which was using fewer microsteps, did you try that?)

So now we are moving the correct distance, we can set the speed and acceleration (and deal with the power shortage).

low power:
1 - If you are not running the motors on **24 volts** then you should be.  2 PC power supplies can be tied together to do this, just Google for how to because you must make sure to separate the ground connections.
2 laptop power supplies can also work, just make sure you do not exceed your stepper drivers max voltage.

2 - you can reduce the microstepping to 2 or 4.    motor power drops off sharply as microstepping increases.
for 2x microstepping set $102=96
for 4x microstepping set $102=192

3 - The Easy Driver is limited to 750ma current.   Your motors might/probably need more, a Big Easy Driver can do twice that which is twice the power into the motor and roughly twice the torque output.

Your leadscrew looks like a piece of allthread.   You should polish it with a wire brush, I find them to be quite rough and polishing makes for much easier movement.  You can then rub a candle on the threads to provide some lubrication.   (not oil, dust will just stick to it and make a grinding paste)

The only variables left to experiment with are the max speed and acceleration.
For both of these, start low and double till it does not work, then go back to the previous number.
But, you must do them one at a time, they are related but changing both at the same time will get you in trouble quickly.

Set acceleration very low
$122=5  or maybe 10
now set $112=50 and issue a command like 'G91 G0 Z50'  which should move the Z 50mm up and take 1 minute.  
'G91 G0 Z-50' will move it 50mm down again. (you may need to move further to get up to full speed since the acceleration is so low)
now set $112=100 and do the above commands again, if it works correctly, double and it try again, and so on.
when it fails to move correctly, go back to the number that did work.

Now you can do the same process again but increase the acceleration in steps until it fails to move, then back to the last one that did work correctly.

Note that there are many combinations of speed and acceleration that will work together, at each end of the scale a 'high speed + low acceleration' and a 'low speed + high acceleration'.    Which one to choose depends on what the machine does most of the time,  long segments can lean toward the high speed end, many small segments (like 3D carving) can lean toward the high acceleration end.

I would like to see pictures of your whole machine, maybe we can give better advice if we have visuals (-:

PS:
To change microstepping on the Easy Driver you need to link MS1/MS2 to ground.
from the Easy Driver web page: "MS1/MS2 : These digital inputs control the microstepping mode."
Possible settings are (MS1/MS2) : full step (0,0), half step (1,0), 1/4 step (0,1), and 1/8 step (1,1 : default).
           MS1  MS2
8x       high   high      (this is the default)
4x       low    high     
2x       high   low
1x       low    low     (don't use this, stepper motors don't like it)
You only have to connect the pins to ground to get a low value, they default to high because there are resistors on the board for this.  You will have to solder a wire from the MS1 or MS2 pin to the GND pin.

sieuwe1, 2017-01-01:
Hey @swarfer

I had forgotten to say in my last post that i also tried diffrent microstepping modes but that also did not solve the problem. 

First here is a picture of my Z-axis setup.
[image]

I have tried what you said using the settings $102=384 $112=50 and $122=10. Here is a video on what happend. 

https://drive.google.com/file/d/0B3bQi5MSlIFYX0Yyenpsc1AxMzA/view?usp=sharing

 
After that i saw that the motor cant step because it is to slow. After messing with $112 i came with the value $112=600 and keeping $102=384 and $122=10. these values gave the best result turning the motor sometimes. Here is video on the result.

https://drive.google.com/file/d/0B3bQi5MSlIFYQWxVLTdsbU5XbFU/view?usp=sharing

I really do not know what values i should use. Maybe somebody can come on skype or something to give live feedback?

Greetings Sieuwe

swarfer, 2017-01-03:
In that second video I can see that it is getting stuck, so there is definitely a sticky spot in the leadscrew mechanism.

You need to solve the motor coupling so it runs straight, that is also causing extra torque.

In the end your motor just does not have enough power.      Do you have a photocopier repair center near you?   If you can get a dead photocopier you can harvest at least 3 very nice stepper motors in the NEMA23 size.  Otherwise you need to use stepper drivers that can provide more current.
These are cheap even here in Africa
https://www.pololu.com/product/1182

You can go back to the gearboxed motor, and jumper for 2x microstepping.   Your max rate will be limited by the 30khz GRBL step rate, but it should be able to move the mechanism since the gearbox multiplies the motor torque.

closed 2017-01-26 by chamnit