grbl/grbl #1145: Independent laser axis with no acceleration limitations

pknoe3lh, 2016-12-19:
Hi everyone!

I love grbl! Its glorious work in the last detail!!!

I use grbl for my little CNC. I upgraded a Proxxon MF70.
I also develop also my own control software with auto-leveling functionality and much more.

Now i have a laser :-) and i was thinking to use an extra step Pin as pulse input for the laser driver.
=> this should be much nicer then just using pwm. 1) higher resolution 2) more smooth at junctions

I'm a good programmer and I have a lot of experience on C. 
But I must say that grbl planner is to complex for me :-(

I added an extra A axis and that is really easy :-)
but the A axis is underlying the physics calculations and therefore engraving is very slow. 
Here is my implementation till now:
https://github.com/pknoe3lh/grbl-Mega

What i need is that the maximum federate is limited when the laser is pulsing more than 30KHz.
But all other calculations like acceleration and junction can be ignored for A.

I tried my best like 10 different hacks but i seems everything is depending on each other in this implementation 

Do someone know how to add A to the planer in a way that this properties are given?
I'm ready to pay a lot of coffees ;-)

greetings 
Patrick

chamnit, 2016-12-19:
@pknoe3lh : Please try Grbl v1.1. Use the new dynamic laser mode. Enabled by M4, rather than M3. It produces very clean and smooth results. No edge burning.

pknoe3lh, 2016-12-20:
@chamnit : is this working for engraving? Is the S level synced with the G1 execution?

pknoe3lh, 2016-12-20:
OK I see.
Its doing that! nice!
I will try that soon.

closed 2016-12-20 by pknoe3lh