grbl/grbl #1146: circle issue

kabuumsound, 2016-12-26:
Hi, I made a cnc/laser machine and usign grbl 1.1 and would like to ask if anybody had simillar issue with circle drawing. Two parts of circles are drawn perfectly in case when both axis are moving in same direction, but when one axis changes it's direction it's drawing as a line and not a circle. Like that is some "delay" of reversing direction.
This are my settings:

$0=10
$1=1
$2=0
$3=0
$4=0
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=0
$23=0
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=1000
$31=0
$32=0
$100=1280.000
$101=1310.000
$102=250.000
$110=500.000
$111=500.000
$112=100.000
$120=10.000
$121=10.000
$122=10.000
$130=200.000
$131=200.000
$132=200.000

And attached picture of drawing circle with laser:

[image]

Thank you for support!

kabuumsound, 2016-12-26:
Correction: issue happens when y-axis changes direction.

jahnj0584, 2016-12-26:
It is either a lot of backlash in your linear system or a motor is stalling
under the load.

swarfer, 2016-12-27:
looks like backlash, or stickyness

also, $32 should be 1 for laser mode, read the laser documentation!
https://github.com/gnea/grbl/blob/master/doc/markdown/laser_mode.md

closed 2017-01-26 by chamnit