grbl/grbl #1147: Addition of relay (spindle/Vacuum), e-stop, and pause+start buttons

jahnj0584, 2016-12-26:
Curious if anyone here has ever wired these on, or knows if its possible.

I use a router, not a spindle so Im using a separate 5V relay to control my router, and shop-vac for dust collection with the coolant and spindle commands in the GCode. Does the 5V come from the arduino, or a separate PSU? 

Is it possible to control the send/pause commands with a physical switch? We run Haas machines at work so Im not used to relying on a mouse anymore. 

Is there a diagram for the E-stop and relay already?

Thanks, and happy holidays.

chamnit, 2016-12-26:
@jahnj0584 : 5V comes from the Arduino as a control voltage. So, it doesn't provide a lot of current. 

Yes, you can. The feed hold, resume(cycle start), and reset are on the analog pins, but set as digital I/O. I would recommend wiring them with a filtering circuit or IC to prevent false readings. 

For e-stop, it's best to cut power. For grbl, the recommended way is to wire directly into the hard reset pin. If it goes to ground, it will place the Arduino into an indefinite suspend state until the pin is disconnected.

jahnj0584, 2016-12-26:
So I will use this diagram but use pins 12 and A3 instead. 
[image]

What sort of wiring would you do for the feed hold? I don't understand how it'd relay the info of an analog switch to UGS. 

For the e-stop I'll probably just have it interrupt the power going to the power supply, and hope we eventually get a "run from line" option

chamnit, 2016-12-26:
@jahnj0584 : You can't control spindle and coolant with buttons with Grbl as is. It sounded like you asked about the feed hold and resume inputs. 

However, if you aren't afraid of coding, using the spindle stop override (only during feed hold) and coolant override in v1.1 as button inputs would be possible. You would just need to alter the control pin interrupt handler to flag those overrides instead of the soft-reset, feed hold, or resume controls. If UGS monitors for accessory state as I've documented, then it should be able to tell you when those accessories are enabled or disabled.

I'm assuming that "Run from line" means program restart from a line. This is mostly a GUI task, rather than Grbl, because it involves machine-specific tasks to get to the restart position. GUIs will also have to have the last modal state prior to restart, either by retaining it prior to the job cancel, or fetching it from a $C Grbl check-mode stream up to the point of restart.

jahnj0584, 2016-12-26:
Im wiring the relay to open/close the ground wire of the seperate extension
cord that the router or Shop vac plug into, respectively. The machines
themselves will always be left on and the relay will choose when they can
operate with the code.

Yea run from line is all for UGS. I'll ask will about it later.

<https://github.com/grbl/grbl/issues/1147#issuecomment-269243158>, or mute
the thread

109JB, 2016-12-27:
> Im wiring the relay to open/close the ground wire of the seperate extension
cord that the router or Shop vac plug into, respectively. 

I assume what you really mean is you are going to switch the neutral wire, and this is a no-no.  The hot leg is the one that should always be switched.

jahnj0584, 2016-12-27:
Why not? My friend did the ground on his and it hasn't been an issue yet.
All you're doing is interrupting the circuit.

109JB, 2016-12-27:
> Why not? 

Safety. a switch, be it a relay or a regular switch is supposed to take away all potential downstream of the switch. If you switch the neutral there is still a hot wire and with no wire to provide a path to ground, then the path could become the one that goes through your fingers, down your arm through your heart and out your toes.

lakeside53, 2016-12-27:
Not only is this bad  (as well described by 109JB) ... it's also illegal by the NEC.  Switch the hot, or both hot and neutral, but never the ground.
-Andy

jahnj0584, 2016-12-27:
Right. I planned on using the hot to begin with but he made it seem like it
didn't matter.

if I want to bypass the relay (like vacuum while PC is off) do I just add a
toggle switch which is also connected the the hot wire?

Cheers

109JB, 2016-12-27:
> if I want to bypass the relay (like vacuum while PC is off) do I just add a
toggle switch which is also connected the the hot wire?

Yes, you could just place a switch that basically re-connects the ends of  the hot wire at the relay. Make sure if you use a toggle switch that it is rated appropriately.

GinoTheCop, 2017-03-06:
Hi,

I have implemented a similar solution to what you are looking for and you can check it here:
http://www.openbuilds.com/threads/c-beam-gtc.6425/

I would be very interested to know how I can place a code to override the spindle stop and coolant during a feedhold and start them back again when resume is triggered through the button. Ant help is apreciated on implementing that.

Thanks

jahnj0584, 2017-03-06:
Why would you want them running during a feed hold? It's a safety concern

GinoTheCop, 2017-03-06:
Erm OK let me re-phrase it.. I want the opposite. I would like the spindle and coolant to stop on feedhold (M05 and M09 commands to be sent) and restart them on resume (M03 and M08 commends).

Sorry for the confusion

chamnit, 2017-03-06:
It has already been implemented. Look for safety door in config.h.

GinoTheCop, 2017-03-06:
Alright thanks a bunch.. Will do so..

GinoTheCop, 2017-03-06:
// #define ENABLE_SAFETY_DOOR_INPUT_PIN // Default disabled. Uncomment to enable.

Mu understanding is;
When uncommented it will act as a safety door when feed_hold button is pressed and un-power the spindle and coolant, those are powered thru relays. When resume button is triggered spindle and coolant will be powered again in accordance with the delay timing in seconds as defined in;

#define SAFETY_DOOR_SPINDLE_DELAY 4.0 // Float (seconds)
#define SAFETY_DOOR_COOLANT_DELAY 1.0 // Float (seconds)

Do you think 4.0 is a good setting for spindle delay? will it not make the spindle dragged over the workpiece because the axis will be moving during those 4 seconds?

Thanks a bunch for your comments and input.

chamnit, 2017-03-06:
The delay is before movement resumes.

GinoTheCop, 2017-03-06:
OK perfect, thank you

GinoTheCop, 2017-03-11:
I uncommented;
`#define ENABLE_SAFETY_DOOR_INPUT_PIN`

And when I press the momentary switch thats connected to feed_hold the movement stops but spindle keeps turning. When I press abort button spindle stops with everything else.

Please see the video below @ 01:21

[](https://www.youtube.com/watch?v=NVWWfCDukts)

I am on 9j by the way. What might be the problem any ideas?

chamnit, 2017-03-11:
Make sure you are editing the config.h file that was copied to your Arduino library folder.

closed 2017-03-11 by chamnit

jpagliaccio, 2017-05-02:
Hey All - Would anybody know how to enable the coolant pin or unused pin when Z <= 0 and disable it when Z > 0? I am using relays for a pen plotter and would like to enable the pen relay at or below Z = 0. Would someone have a snippet of code for that? Thanks.

lalo-uy, 2017-05-02:
Why not use the correct G-code for that :

To turn mist coolant on, program: M7.

To turn flood coolant on, program: M8.

lalo-uy, 2017-05-02:
To turn all coolant off, program: M9.

2017-05-02 9:45 GMT-03:00 Eduardo Spremolla <e.spremolla@gmail.com>:

jpagliaccio, 2017-05-04:
@lalo-uy - Sorry. I have it working now. Please disregard my coolant question. I switched to Laser Mode and moved my relay from the coolant pin to the PWM pin. In laser mode I am getting the desired z axis motion without having to change any of the gcode. Thanks.