grbl/grbl #1149: GRBL not working

shyetman, 2017-01-02:
was cutting out a design and machine stop, there is nothing in GRBL settings, just blank, there is nothing in my X,Y,Z settings, and my home button is gone.

hope someone can help
thanks
shawn

shyetman, 2017-01-02:
[image]

electrokean, 2017-01-02:
From the limited info provided, this sounds like a communications error.  The USB serial port may have disconnected.  This can be caused by electrical noise (EMI) from the motors/spindle getting back through the USB cable.  Following good grounding practices, along with use of shielded cables and cable ferrites can help.

shyetman, 2017-01-02:
Thank you sir, after going out and moving cable I found the problem.

closed 2017-01-02 by shyetman