grbl/grbl #1150: incorrect movement when GRBL recive g10

MohamadAlsadi, 2017-01-02:
I have problem with GRBL when receive G10 , It moves random and incorrect movement when only receiving this instruction

please help me

swarfer, 2017-01-03:
Well, G10 is not a movement command so it should not move at all!

You should go here https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.9 and read all the places that mention G10

MohamadAlsadi, 2017-01-03:
my friend #swarfer
I understand that. I copy the G code file and paste it more time in editor bcnc. because I want re-draw the same shape once more.  when grbl receive G10 ,the next command don't execute correct. this happened with me more time

X3msnake, 2017-01-03:
what are you trying to do with the G10 command?

can you past the G10 line and the next one?

MohamadAlsadi, 2017-01-03:
`g1 x105 y80
g3 x95 y90 i-10 j0
g1 x33 y90
g3 x23 y80 i0 j-10
g1 x23 y45
g0 z3
G10 L20 P1 Z0
`g1 x105 y80
g3 x95 y90 i-10 j0
g1 x33 y90
g3 x23 y80 i0 j-10
g1 x23 y45
g0 z3
G10 L20 P1 Z0

.`

X3msnake, 2017-01-03:
for what i read in your code you are

X3msnake, 2017-01-03:
for what i read in your code you are offseting G54 by 0 on Z axis only

that is basically not changing anything or am i reading it wrong?

G10 L2

Coordinate System Origin Setting
http://linuxcnc.org/docs/2.6/html/gcode/gcode.html#sec:G10-L2_

G10 L20

Coordinate System Origin Setting Calculated
http://linuxcnc.org/docs/2.6/html/gcode/gcode.html#sec:G10-L20_

https://smithy.com/cnc-reference-info/coordinate-system/sample-program-using-offsets/page/7

2017-01-03 14:37 GMT+00:00 Vinicius Silva <x3msnake@gmail.com>:

> for what i read in your code you are
>
>
>
>> g1 x105 y80 g3 x95 y90 i-10 j0 g1 x33 y90 g3 x23 y80 i0 j-10 g1 x23 y45
>> g0 z3 G10 L20 P1 Z0g1 x105 y80
>> g3 x95 y90 i-10 j0
>> g1 x33 y90
>> g3 x23 y80 i0 j-10
>> g1 x23 y45
>> g0 z3
>> G10 L20 P1 Z0
>>
>> .`
>>
>> <https://github.com/grbl/grbl/issues/1150#issuecomment-270105375>, or mute
>> the thread
>>
>
>
>
> Com os melhores cumprimentos,
> Vinicius Silva

MohamadAlsadi, 2017-01-03:
I don't write G10 line  . It is auto generate when I drow some shape and import to bcnc

swarfer, 2017-01-03:
that is not the correct way to use the G10 command
G10 is for co-ordinate system setup, which you do not do within a Gcode program.

There is also a warning against using it in a program in the GRBL wiki because it writes to EEPROM which causes an execution delay.

Whatever program is generating such Gcode needs to be fixed (what program is it?).

Please read the LinuxCNC documentation that has been linked above by X3msnake so you learn what G10 is supposed to do, and why and when to use it.

MohamadAlsadi, 2017-01-03:
excellent my friend swarfer
the problem is write to EEPROM and happened interupt in the same time. but how I can this problem?

chamnit, 2017-01-03:
@MohamadAlsadi : Insert a `G4 P0.01` dwell prior to a G10 command. This will force Grbl to empty its planner buffer and complete its motions. Grbl v1.1 does automatically for you, but v0.9 does not.

swarfer, 2017-01-03:
IMHO you solve the problem by using G10 correctly *outside* of the Gcode program.
for example in UGS when you press one of the axis 'set 0' buttons it sends a G10 command to set that axis to zero in the current co-ordinate system (G54 by default).   This is the only time a G10 should be issued, during machine **setup**.

MohamadAlsadi, 2017-01-03:
I will try that my friend @chamnit 
and tell you what results. I wish that your notice solve this problem

chamnit, 2017-01-03:
@MohamadAlsadi : This problem is well-documented and discussed in several places in the Wiki. Please make sure you read and understand what's there before posting.

closed 2017-01-03 by chamnit