grbl/grbl #1153: grbl v1.1 vs grbl0.9

MohamadAlsadi, 2017-01-04:
I want to select version from grbl to use it in my machine . so I would ask you that , any version from grbl don't write in eeprom? becouse I will run my machine to long time and execute huge G-code command. so I want don't damage eeprom During its long use

X3msnake, 2017-01-04:
Writing to the eeprom only happens in the cases where you are changing
stored settings like the G53/54/... position

A well written gcode program will not need to rewrite this, so your rom
will be safe

if for some reason your cam programs is sending WCS coordinate changes
(G53/54/...), then you must change the post-processor so that it wont write
to the rom

chamnit, 2017-01-04:
@MohamadAlsadi : EEPROM is rated up to 100,000 cycles. That's far more than life of a CNC machine unless you are constantly writing to it hundreds of times a day. The G10 command writes to EEPROM, but G92 doesn't. G92 can be useful if you have fixed and known offsets from a known work position. They clear when Grbl is reset, but since you know where things are, it's ok to use them in this scenario.

closed 2017-01-04 by chamnit