grbl/grbl #1154: loop by Grbl master 0.9

MohamadAlsadi, 2017-01-04:
how I can make loop by G-code ? and stop loop at selected condition?
 for example: I have the G-code commands

G10 L2 P1 X-10. Y-10. Z-10 
G10 L2 P2 X-10. Y-11. Z-10 
G10 L2 P3 X-10. Y-12. Z-10

I need loop to repeated my code 5 times. note that: I am using Grbl master 0.9

chamnit, 2017-01-04:
@MohamadAlsadi : There something called subroutines or macro programming in g-code which can do program loops and expression conditions. Grbl does not support them.

closed 2017-01-04 by chamnit

MohamadAlsadi, 2017-01-04:
@chamnit 
 there are not any solve to repeat group commands?

X3msnake, 2017-01-05:
use g92 offsets instead

MohamadAlsadi, 2017-01-05:
@X3msnake 
can you give me example?

X3msnake, 2017-01-05:
http://www.cnccookbook.com/CCCNCGCodeG54G92WorkOffsets.htm

*****

*G92: A Programmable Temporary Work Offset*

Suppose you want a programmable temporary work offset. There are a lot of
ways to accomplish this, but one that is tried and true is to use G92. G92
establishes a work offset based on the offset coordinates you provide. So,
if that vise jaw corner we've talked about is located *from the current
tool position* at *offset* X10 Y10 Z0, you could execute the following:

G92 X10 Y10 Z0

Now the coordinates of the vise corner are X0Y0Z0. You've just established
your own work offset using G92.

Let's consider another example where a G92 or one of the other ways of
programming a work offset might be handy. Let's suppose you have a fixture
plate that has a grid of parts on it. You know in advance that the parts
are 4 across and 2 down, so there will be 8 parts. The plate is made so
that the X and Y distance between each part is constant (within tolerances
of course). So, the parts are maybe 5 inches apart in X and 5 inches in Y.

Since G92 is an offset from the current tool position, we move the tool to
part zero on the first part at the beginning of the g-code program. We can
then use G92 to add offsets relative to that position, 5 inches in X and 5
inches in Y for each part as we step through them. Pretty handy, huh?

Note that G92 is available on mills and some lathes, but most lathes use
G50 for this function.

QGB, 2023-06-06:
grbl 不支持的简单 WHILE-ENDWHILE 循环结构  

grbl not support WHILE  "ENDWHILE"